espressif/esp-idf #3747: lwip icmp ping checksum issue - unsure of exact cause.

benpeoples, 2019-07-05:
## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): n/a
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    v4.0-dev-1136-g28f1cdf5e
- Build System:         GNU Make 3.81
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     macOS
- Power Supply:         external 5V

## Problem Description

Ping responses are improperly formed.   ping is responding with [BAD CHECKSUM].  Wiresharking these pings finds that the checksum is being set to 0x0000.

### Expected Behavior

Ping checksums should be correct.

### Actual Behavior

Ping checksums are not correct.

### Steps to repropduce

Of note, we're using this over ethernet, so I don't know if this is an ethernet-specific bug.   I can't post the code as much as I'd like to since we're using proprietary PHY code.

### Workaround 

Adding these two lines: 
```
#define LWIP_CHECKSUM_CTRL_PER_NETIF 1
#define CHECKSUM_GEN_ICMP 0
```

to the file:  `esp-idf/components/lwip/port/esp32/include/lwipopts.h` solved the issue.  

### Strange side effect 

MacOS's `ping` command does not report bad-checksum ping responses, it treats them as non-responses even if the rest of the packet is correct.

negativekelvin, 2019-07-05:
https://github.com/espressif/esp-idf/issues/3715

benpeoples, 2019-07-06:
Yep, that's the issue, thanks!

suda-morris, 2019-07-07:
Hi @benpeoples Our master should have fixed the issue, does the same problem still exist in your case? If not, could you please close this issue?

benpeoples, 2019-07-08:
Current master fixes the issue, thank you =)

closed 2019-07-08 by benpeoples