espressif/esp-idf #3749: Clarification on whether 8MB PSRAM requires additional SPI Host

deltabeard, 2019-07-06:
It is unclear from the comment below whether a separate SPI host is required for 8MB PSRAM as well as the 4MB version. Documentation in the `menuconfig` interface does not clarify either.

https://github.com/espressif/esp-idf/blob/beaefd3359973dcf8bba4e1d4462aa7f5d67be6a/components/esp32/spiram_psram.c#L145

Please may you add whether a separate SPI host is required for 8MB PSRAM too, or if this is just a limitation of the 4MB PSRAM.

My ESP32-WROVER has 8MB PSRAM integrated (as mentioned in the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf)) so it would be nice to know what configuration I should be using in ESP-IDF.

negativekelvin, 2019-07-06:
No, but some wrover do have the 4mb psram, depending on production date

https://github.com/espressif/esp-idf/blob/beaefd3359973dcf8bba4e1d4462aa7f5d67be6a/components/esp32/spiram_psram.c#L723-L756

deltabeard, 2019-07-07:
So it's just an issue with ESP32 modules that use the 4MB (not 4mb?) PSRAM?

krzychb, 2022-05-07:
The requirement for additional SPI host for 4 MB (32 Mb) PSRAM _running at 80 MHz_ is described in ESP-IDF Programming Guide > Project Configuration > [CONFIG_SPIRAM_SPEED](https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32/api-reference/kconfig.html#config-spiram-speed).

The PSRAM chip/size may be selected in Project Configuration > [CONFIG_SPIRAM_TYPE](https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32/api-reference/kconfig.html#config-spiram-type).

 The requirement is related to specific PSRAM chip (IS25WP032) and other currently supported PSRAM chips are not affected.

closed 2022-06-08 by krzychb