espressif/esp-idf #3751: [Doc] Incorrect opcode for ULP : REG_WR

hex007, 2019-07-06:
https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#180

The op code for REG_WR should be `4'd1` and not `4'd2`

[image]

[image]

Esp-Doc, 2019-07-11:
Hi, thank you for pointing this out! The Documentation team has fixed the typo. Updated version will soon be released.

hex007, 2019-07-11:
@Esp-Doc if possible also include what the syntax `4'd1` means. It is not quite apparent to first timers

Esp-Doc, 2019-07-19:
Will do. Thanks!

closed 2019-12-15 by hex007