espressif/esp-idf #3752: In esp32,  Ethernet example

zchaojian, 2019-07-07:
I use the ESP32 kit develop board, make and load the "ethernet" example. I only display some number information to locate the bug, but I don't discover any issue.  
Brothers and sisters, will you help me to solve the problem?

I don't link the nINT/RETCLK, other pins I connected.
Oh, the main board is ESP-WROOM-32D  ESP32-DevKitC , and the PHY board is LAN8720.

So who answer and explain my prolbem? Thank you very much!!!
## Environment

- Development Kit:     [ESP32-DevKitC]
- Kit version (DevKitC): [v1|v2|v3|v4] I don't know detial.
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-771-gc9bf3a3
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

### Expected Behavior
In the ethernet example Readme file displays as follow:

Example Output
bash
I (0) cpu_start: App cpu up.
I (265) heap_init: Initializing. RAM available for dynamic allocation:
I (272) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (278) heap_init: At 3FFB3FF0 len 0002C010 (176 KiB): DRAM
I (284) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (291) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (297) heap_init: At 4008851C len 00017AE4 (94 KiB): IRAM
I (303) cpu_start: Pro cpu start user code
I (322) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (325) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (325) emac: emac reset done
I (335) eth_example: Ethernet Started
I (4335) eth_example: Ethernet Link Up
I (6325) tcpip_adapter: eth ip: 192.168.2.164, mask: 255.255.255.0, gw: 192.168.2.2
I (6325) eth_example: Ethernet Got IP Address
I (6325) eth_example: ~~~~~~~~~~~
I (6325) eth_example: ETHIP:192.168.2.164
I (6335) eth_example: ETHMASK:255.255.255.0
I (6335) eth_example: ETHGW:192.168.2.2
I (6345) eth_example: ~~~~~~~~~~~

### Actual Behavior
After I configrate all platform, the "make monitor" displays as follows:

I (0) cpu_start: App cpu up.
I (268) heap_init: Initializing. RAM available for dynamic allocation:
I (275) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (281) heap_init: At 3FFB4048 len 0002BFB8 (175 KiB): DRAM
I (287) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (294) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (300) heap_init: At 400887D0 len 00017830 (94 KiB): IRAM
I (306) cpu_start: Pro cpu start user code
I (324) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
----------00----------
----------01----------
----------02----------
----------03----------
----------04----------
----------05----------
----------06----------
----------07----------
----------08----------
----------09----------
I (366) eth_example: Power On Ethernet PHY
I (376) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
----------10----------
----------11----------
----------12----------
I (396) emac: emac reset done
I (396) eth_example: Ethernet Started
----------13----------
I (4396) eth_example: Ethernet Link Up

### Steps to repropduce

I want to display some information as follow:

I (6325) tcpip_adapter: eth ip: 192.168.2.164, mask: 255.255.255.0, gw: 192.168.2.2
I (6325) eth_example: Ethernet Got IP Address
I (6325) eth_example: ~~~~~~~~~~~
I (6325) eth_example: ETHIP:192.168.2.164
I (6335) eth_example: ETHMASK:255.255.255.0
I (6335) eth_example: ETHGW:192.168.2.2
I (6345) eth_example: ~~~~~~~~~~~

I should call which function and modify which function? Please help me, Thank you 

## Code to reproduce this issue

```cpp
void app_main()
{
    static tcpip_adapter_ip_info_t esp_ip[TCPIP_ADAPTER_IF_MAX];
   
    printf("----------00----------\n");
    tcpip_adapter_init();
    printf("----------01----------\n");

    IP4_ADDR(&esp_ip[TCPIP_ADAPTER_IF_AP].ip, 192, 168, 5, 5);
    IP4_ADDR(&esp_ip[TCPIP_ADAPTER_IF_AP].gw, 192, 168, 5, 1);
    IP4_ADDR(&esp_ip[TCPIP_ADAPTER_IF_AP].netmask, 255, 255, 255, 0);

    printf("----------02----------\n");

    ESP_ERROR_CHECK(esp_event_loop_create_default());

    printf("----------03----------\n");
    eth_config_t config = DEFAULT_ETHERNET_PHY_CONFIG;
    printf("----------04----------\n");
    config.phy_addr = CONFIG_PHY_ADDRESS;
    printf("----------05----------\n");
    config.gpio_config = eth_gpio_config_rmii;
    printf("----------06----------\n");
    config.tcpip_input = tcpip_adapter_eth_input;
    printf("----------07----------\n");
    config.clock_mode = CONFIG_PHY_CLOCK_MODE;
    printf("----------08----------\n");
\#ifdef CONFIG_PHY_USE_POWER_PIN
    /* Replace the default 'power enable' function with an example-specific one
     that toggles a power GPIO. */
    config.phy_power_enable = phy_device_power_enable_via_gpio;
    printf("----------09----------\n");
\#endif

    ESP_ERROR_CHECK(esp_eth_init(&config));
    printf("----------10----------\n");
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL));
    printf("----------11----------\n");
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler, NULL));
    printf("----------12----------\n");
    ESP_ERROR_CHECK(esp_eth_enable()) ;
    printf("----------13----------\n");
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
```

## Other items if possible

negativekelvin, 2019-07-07:
https://github.com/espressif/esp-idf/issues/2059

zchaojian, 2019-07-07:
> #2059
> #652

In lan8720 hardware board, I don't link the nINT/REFCLK, because the board already have 50 M crystal oscillator. I see some file know, don't use the nINT/REFCLK pin. And I connected the pin to ESP32 GPIO0, but the result is the same .

suda-morris, 2019-07-07:
@zchaojian 
This is the case that ESP32 should get RMII clock from external source (PHY or oscillator), so make sure connect the nINT/REFCLK to GPIO0, and in the menuconfig, select the right clock mode.
From your log, I guess there isn't a valid RMII clock (50MHz) input to ESP32 via GPIO0, can you take an oscilloscope to captue the signal on GPIO0?

zchaojian, 2019-07-09:
I took an oscilloscope to capture the signal of GPIO0 which is a sine wave.

This issue already is resolved. The reason is ethernet example program which doesn't set a static IP of the PHY. So it doesn't appear those information about IP address. Because I used the network interchanger, and I started to use router the ESP32 is normal. I can ping IP address.

Another discover is set menuconfigeration  for GPIO17, and don't link the pin and nINT/REFCLK pin.
the network will work but the signal is unstable. Maybe lost many databases.

Thank you many brothers and sisters help me! Good luck to you all!

zchaojian, 2019-07-09:
@

zchaojian, 2019-07-09:
@suda-morris

suda-morris, 2019-07-09:
@zchaojian Yes, the network must be unstable, because the RMII clock used for ESP32 and LAN8720 are not the same one. For ESP32, you've configured using the internal APLL to generate the RMII clock, but for LAN8720, you're using an oscillator. Although both clocks are 50MHz, but probably there will be difference between their phase.

zchaojian, 2019-07-09:
Thank you very much!@sudo-morris

closed 2019-07-09 by zchaojian