espressif/esp-idf #3755: Random crashes while connecting to wifi

ducalex, 2019-07-08:
## Environment

- Development Kit:      Lolin32
- Kit version 
- Module or chip used:  ESP32-WROOM-32D
- IDF version:     v3.3-beta3-130-ge443467f3
- Build System:         Make
- Compiler version:   crosstool-ng-1.22.0-80-g6c4433a5
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

I'm encountering random crashes that are seemingly related to wifi or lwip. The trace is different most of the time but the fault happens at the same place.

The crashes happen after connecting to the router but before receiving an IP, on a random basis, perhaps once every 30 boots.

I use Arduino as a component, but I have also reproduced the issue using esp_wifi_*.

### Expected Behavior
No crash

### Actual Behavior
Crash

### Steps to repropduce

I was unable to produce steps to replicate, it is random.

### Code to reproduce this issue

```cpp
void loop()
{
    esp_pm_config_esp32_t pm_config;
        pm_config.max_freq_mhz = 160;
        pm_config.min_freq_mhz = 160;
        pm_config.light_sleep_enable = true;
    esp_pm_configure(&pm_config);
    // ...
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    WiFi.setSleep(true);
    
    while (WiFi.status() != WL_CONNECTED && millis() < (STATION_POLL_INTERVAL * 1000)) {
        WiFi.waitStatusBits(STA_HAS_IP_BIT, 500);
        Display.printf(".");
    }
    
    // Doesn't reach
    if (WiFi.status() == WL_CONNECTED) {
        // ...
    }
    else {
        // ...
    }
    WiFi.disconnect(true);

    hibernate();
}
```

## Debug Logs
[Multiple crash sessions here](https://github.com/espressif/esp-idf/files/3370219/failures.txt)

```
I (1951) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2791) wifi: state: init -> auth (b0)
I (2801) wifi: state: auth -> assoc (0)
I (2801) wifi: state: assoc -> run (10)
I (2811) wifi: connected with Skynet_AP, channel 11, bssid = 6e:f4:ab:55:33:11
I (2811) wifi: pm start, type: 1

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40139104: ffffff2a ffffffff ffffffff
0x40139104: dns_check_entry at C:/users/alex/apps/esp/esp-idf/components/lwip/lwip/src/core/dns.c:420

Core 1 register dump:
PC      : 0x40139108  PS      : 0x00060a30  A0      : 0x801396b9  A1      : 0x3ffc8750
0x40139108: dns_check_entry at C:/users/alex/apps/esp/esp-idf/components/lwip/lwip/src/core/dns.c:420

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb00bc  A5      : 0x3ffc8740
A6      : 0x3ffceba4  A7      : 0x3ffcecf4  A8      : 0x8013b08b  A9      : 0x3ffc8730
A10     : 0x3ffbb038  A11     : 0x400831d0  A12     : 0x3ffcee14  A13     : 0x00000000
0x400831d0: _free_r at C:/users/alex/apps/esp/esp-idf/components/newlib/syscalls.c:41

A14     : 0x00000004  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: c809a25255b70b24aa8a182434761f4188318e32412e718901e07721953fb35e

Backtrace: 0x40139108:0x3ffc8750 0x401396b6:0x3ffc8770 0x40139b46:0x3ffc8790 0x40139bdd:0x3ffc87b0 0x40133f53:0x3ffc87e0
0x40139108: dns_check_entry at C:/users/alex/apps/esp/esp-idf/components/lwip/lwip/src/core/dns.c:420

0x401396b6: dns_check_entries at C:/users/alex/apps/esp/esp-idf/components/lwip/lwip/src/core/dns.c:1144
 (inlined by) dns_tmr at C:/users/alex/apps/esp/esp-idf/components/lwip/lwip/src/core/dns.c:442

0x40139b46: cyclic_timer at C:/users/alex/apps/esp/esp-idf/components/lwip/lwip/src/core/timeouts.c:381

0x40139bdd: sys_check_timeouts at C:/users/alex/apps/esp/esp-idf/components/lwip/lwip/src/core/timeouts.c:361
 (inlined by) sys_timeouts_mbox_fetch at C:/users/alex/apps/esp/esp-idf/components/lwip/lwip/src/core/timeouts.c:433

0x40133f53: tcpip_thread at C:/users/alex/apps/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:483

CPU halted.

```

## Other items if possible

- [sdkconfig.txt](https://github.com/espressif/esp-idf/files/3370205/sdkconfig.txt)

ducalex, 2019-07-08:
It seems that if I disable automatic light sleep or call `esp_wifi_set_ps(WIFI_PS_NONE)`/`WiFi.setSleep(false)` or both it fixes the problem (but isn't a solution).

liuzfesp, 2019-07-10:
HI @ducalex could you let me know your IDF commit id?

liuzfesp, 2019-07-11:
HI @ducalex do you run "git submodule update" after you upgrade your IDF to ge443467f3 ?

ducalex, 2019-07-11:
@liuzfesp Yes it was a clean install doing the following:
````
git clone -b release/v3.3 --recursive https://github.com/espressif/esp-idf.git
git submodule update
````

freakyxue, 2019-11-13:
@chinmaychhajed  We are following your instructions to reproduce the problem

freakyxue, 2019-12-13:
@ducalex  
Can you provide your code for me to reproduce the problem

freakyxue, 2019-12-13:
@ducalex I can't see the real reason through log, so I need your project to reproduce again

ducalex, 2019-12-13:
@freakyxue I'm sorry this project is long terminated, I cannot help you further at this point. 
I resolved the problem in my case by disabling light sleep while wifi was in use.

Feel free to close the issue.
Thanks.

Alvin1Zhang, 2020-01-15:
@ducalex Thanks for the updates.

closed 2020-01-15 by Alvin1Zhang