espressif/esp-idf #3756: sntp_getserver blank ip with DNS

jwcannon, 2019-07-09:
## Environment

- Development Kit: **none**
- Module or chip used: **ESP32-WROVER-B**
- IDF version **v4.0-dev-621-g81ca1c013**
- Build System: **Make**
- Compiler version: **xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0**
- Operating System: **Windows**
- Power Supply: external **3.3V**

## Problem Description

When using SNTP with a DNS ("pool.ntp.org"), sntp_getserver(0) returns 0.0.0.0 after SNTP connected and time is set. Set lwIP debug to be on and it appears to be an issue with lwIP never setting the IP while in sntp_request() because it is waiting to get the host. If a timeout occurs and sntp_request() is called again, logs the IP correctly. If using a static IP instead of a DNS IP logs correctly.

### Expected Behavior

sntp_getserver(0) should return valid ip

### Actual Behavior

sntp_getserver(0) returns 0.0.0.0

### Code

```cpp

static void initialize_sntp(void)
{
	ESP_LOGI(WIFI_TAG, "Initializing SNTP");
	sntp_setoperatingmode(SNTP_OPMODE_POLL);
	if(deviceSettings.sntpUseIP == false){
		sntp_setservername(0, deviceSettings.sntpServerName); //pool.ntp.org
	}
	else{
		sntp_setserver(0, &deviceSettings.sntpServerIP);
	}
	sntp_init();
}

static void obtain_time(void)
{
	ESP_ERROR_CHECK( nvs_flash_init() );

	xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT,
						false, true, portMAX_DELAY);
	initialize_sntp();
	// Set timezone to Eastern Standard Time and print local time
	setenv("TZ", deviceSettings.sntpTimezone, 1);
	tzset();
	const time_t currentTime = time(NULL);
	struct tm *currenttm = localtime(&currentTime);
	// wait for time to be set
	time_t now = 0;
	struct tm timeinfo = { 0 };
	int retry = 0;
	const int retry_count = SNTP_RETRY_COUNT;
	while (sntp_get_sync_status() == SNTP_SYNC_STATUS_RESET && ++retry < retry_count) {
		//try to connect
		ESP_LOGI(WIFI_TAG, "Waiting for system time to be set... (%d/%d)", retry, retry_count);
		vTaskDelay(SNTP_RETRY_DELAY / portTICK_PERIOD_MS);
		time(&now);
		localtime_r(&now, &timeinfo);
	}
	ip_addr_t* ip = sntp_getserver(0);
	xEventGroupSetBits(wifi_event_group, SNTP_BIT);
	if(retry < retry_count){
		//connection success
		ESP_LOGI(WIFI_TAG, "Connected to SNTP");
		store_timeStampedData_entry(TS_CONNECTED_SNTP, &ip->u_addr);
		printf("SNTP Server IP: %s\n", ipaddr_ntoa(ip));
	}
       //code continues on but not relevant
        ...
        ...
	obtainTimeHandle = NULL;
	vTaskDelete(NULL);
}

```
## Debug Logs

```
I (2775) wifi.h: Initializing SNTP
I (2785) esp_https_server: Starting server
sntp_request: Waiting for server address to be resolved.
I (2795) wifi.h: Waiting for system time to be set... (1/10)
I (2795) esp_https_server: Server listening on port 80
I (2805) HTTP Server: Registering URI handlers
sntp_dns_found: Server address resolved, sending request
sntp_send_request: Sending request to server
sntp_process: Tue Jul  9 11:10:13 2019
, 158680 ussntp_recv: Scheduled next time request: 3600000 ms
I (4805) wifi.h: Connected to SNTP
I (4805) MemoryStorage.h: store_timeStampedData_event stored TS event 76, at 18076 offset, with 1562685014 time, and 0 threshold, size of: 12
next offset is: 18088
SNTP Server IP: 0.0.0.0
I (4815) wifi.h: The current date/time in New York is: Tue Jul  9 11:10:14 2019
Set Time Tue Jul  9 15:10:14 2019
I (4825) MAIN: Got past wifi init

```

negativekelvin, 2019-07-09:
Add
`sntp_servers[sntp_current_server].addr = ipaddr;`
before
https://github.com/espressif/esp-lwip/blob/18548d1e256f7651fb4da48dbd770556e0eca8ce/src/apps/sntp/sntp.c#L476

jwcannon, 2019-07-09:
Thanks @negativekelvin that fixes the problem, but I was hoping to avoid modifying lwIP. This is a straightforward and easy solution though. One small correction for anyone who wants to modify lwIP is that ipaddr is a pointer so it would need to be:
`sntp_servers[sntp_current_server].addr = *ipaddr;`
Maybe a future update to lwIP will fix this issue, but for now I'm looking into other ideas I have.

negativekelvin, 2019-07-10:
In fact it was fixed in upstream
https://github.com/lwip-tcpip/lwip/commit/e46b9ad4acb5f271d76799226d4c967ffb5b42a0

negativekelvin, 2019-07-16:
Fixed with lwip upgrade https://github.com/espressif/esp-idf/commit/fa02598b5c978402c1013b22ca81919beded97f3

closed 2019-07-22 by jwcannon

sieren, 2019-09-02:
Isn't https://github.com/espressif/esp-idf/commit/fa02598b5c978402c1013b22ca81919beded97f3 supposed to land on the `release/v4.0` branch?