espressif/esp-idf #3758: [Local Control Feature] Client side implementation and examples

enricop, 2019-07-10:
Hi all,
we have seen that "Local Control" feature has been added recently with this [commit](https://github.com/espressif/esp-idf/commit/b75f8b1b20c2b3c470b0b611125891e60c1d0de9)

[Here](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_local_ctrl.html) the first documentation

We would like a more detailed example for client side implementation.
What does it mean "send and receive protobuf messages understood by the esp_local_ctrl service"? We are using protobuf for other projects. Where is a starting .proto for working with Local Control feature?

@anurag-kar How [protocomm](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/provisioning/protocomm.html) sessions should be handled and created on the client side?

We can't find proper protocomm examples also in these [provisioning-tools](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/provisioning/provisioning.html#provisioning-tools)

We are going to use this feature in our project mainly over BLE.
I suppose the major example structure would be the same as HTTPD, but a specific BLE example would be useful too!
https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_local_ctrl/

Thank you for your support

enricop, 2019-07-10:
I've seen that a very basic client is [here](https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_local_ctrl/scripts/esp_local_ctrl.py). It would be usefull to have a clear specification document.

anurag-kar, 2019-07-10:
@enricop Glad to know that you are interested in the local control feature!

> We would like a more detailed example for client side implementation.

We are currently pushing in that direction, so you can expect a detailed client side implementation guide soon. 

For now you can refer to the [protobuf](https://github.com/espressif/esp-idf/blob/b75f8b1b20c2b3c470b0b611125891e60c1d0de9/components/esp_local_ctrl/proto/esp_local_ctrl.proto) file itself, which has all the command and response packets. The corresponding python wrapper [proto.py](https://github.com/espressif/esp-idf/blob/b75f8b1b20c2b3c470b0b611125891e60c1d0de9/examples/protocols/esp_local_ctrl/scripts/proto.py) can be used readily for client side implementation.

The example script [esp_local_ctr.py](https://github.com/espressif/esp-idf/blob/b75f8b1b20c2b3c470b0b611125891e60c1d0de9/examples/protocols/esp_local_ctrl/scripts/esp_local_ctrl.py) uses `proto.py` and readily available modules from `esp_prov`, for BLE and HTTPS transport, but those have nothing specific to protocomm as such, and could be implemented in any other way suitable to the platform running the client application.

Since, protocomm security is not used for `esp_local_ctrl` yet, that simplifies things on the client side, but we do plan to have a sophisticated scheme for secure communication and pairing.

> We can't find proper protocomm examples 

Currently there is only the `custom_config`  example under `examples/provisioning`, which shows how to extend the provisioning example, by adding a custom protocomm endpoint. Since, presently protocomm is being used in the context of provisioning and local_ctrl only, the need for a separate example was overlooked. But now it seems we atleast need a step by step guide to using it independently.

I'll keep this issue open and keep you posted on any progress on our side.

enricop, 2019-07-11:
> We are currently pushing in that direction, so you can expect a detailed client side implementation guide soon.

that would help, along with an example for working with BLE on esp32 side. thanks

> For now you can refer to the protobuf file itself, which has all the command and response packets. The corresponding python wrapper proto.py can be used readily for client side implementation.

thank you, i suppose this .proto file is generic and can be use in every case

> The example script esp_local_ctr.py uses proto.py and readily available modules from esp_prov, for BLE and HTTPS transport, but those have nothing specific to protocomm as such, and could be [...]

I see there are no imports regarding protocomm in that script. For our project we need to use protocomm security level 1 with proof of possession over BLE.

> Currently there is only the custom_config example under examples/provisioning, which shows how to extend the provisioning example, by adding a custom protocomm endpoint. Since, presently protocomm [...]

Interesting. I'll look to that "[custom_config](https://github.com/espressif/esp-idf/blob/master/examples/provisioning/custom_config/main/app_prov.c)" example.

Is it possible to setup local_ctrl with protocomm security on the esp32 side?
I suppose protocomm is only for application transport layer: is it [here ](https://github.com/espressif/esp-idf/tree/master/tools/esp_prov) that I can find an example script of protocomm on client side for BLE?  

Thanks

anurag-kar, 2019-07-11:
> Is it possible to setup local_ctrl with protocomm security on the esp32 side?

Not yet. But that is our next target. We may improve upon the security scheme to provide something more robust.

>  is it here that I can find an example script of protocomm on client side for BLE?

Yes. `esp_prov` is the tool for testing all the provisioning examples, and can serve a sreference implementation. The BLE client is implemented using Bluez and DBus, and thus only works on linux.

We also provide [android](https://github.com/espressif/esp-idf-provisioning-android) / [iOS](https://github.com/espressif/esp-idf-provisioning-ios) apps to perform provisioning.

bbinet, 2020-02-18:
@anurag-kar Any ETA for the protocomm security feature?

Alvin1Zhang, 2020-03-16:
@enricop @bbinet Thanks for reporting and updates, and sorry for the slow turnaround. We have been evaluating this feature, and will share if any updates. Thanks.

Harshal5, 2022-11-04:
Hello @bbinet @enricop !

The `esp_local_ctrl` [example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/esp_local_ctrl) has been updated with support for both protocomm security version 1 and security version 2.
Please refer to this commit https://github.com/espressif/esp-idf/commit/f9db1ec366ba628cdd5a2be261caaef0e5226175.

Thank you!

closed 2022-11-04 by Harshal5

Fahadk7, 2023-11-10:
Hi

Any example code for local control via BLE for the esp32