espressif/esp-idf #3760: esp32 Bluetootn  Client Not Recv data

xiaodingding, 2019-07-10:
1.esp-idf is master branch.

2.toolchain:esp32_win32_msys2_environment_and_toolchain-20190611.zip

3.我目前测试的主要是在Arduino环境下，将蓝牙数据收发都跑通了，但是在idf环境下，数据能正常发送，但是不能接收数据。

4.arduino代码：
static BLEUUID serviceUUID("0003cdd0-0000-1000-8000-00805f9b0131");

static BLEUUID  charNotifyUUID("0003cdd1-0000-1000-8000-00805f9b0131");

static BLEUUID  charWriteUUID("0003cdd2-0000-1000-8000-00805f9b0131");

static BLEUUID NotifyDescUUID("00002902-0000-1000-8000-00805f9b34fb");

    pcharactDesc = pRemoteNotifyCharacteristic->getDescriptor(NotifyDescUUID);
    if( pcharactDesc == nullptr)
    {
      Serial.print("Failed to find our notify characteristic desc UUID: ");
      Serial.println(NotifyDescUUID.toString().c_str());
      return false;
    }
    Serial.println(" - Found our notify characteristic desc");
    pcharactDesc->writeValue(0x1,false);    
    
    往 characteristic-》descriptor，写1，就可以接收数据了，这个是在Arduino下调好了的。

5.在IDF环境下：
static void notify_desc_set_enable(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gattc_cb_param_t *param, uint16_t handle)
{
    uint16_t count = 3, i = 0,notify_en = 0x0101;
    esp_err_t err = 0;
    esp_ble_gattc_cb_param_t *p_data = (esp_ble_gattc_cb_param_t *)param;

    err = esp_ble_gattc_read_char_descr(gattc_if, p_data->search_cmpl.conn_id, handle,ESP_GATT_AUTH_REQ_NONE);
    if(ESP_OK != err)
    {
        ESP_LOGE(GATTC_TAG, "set before read notify flag error");
    }else
    {
        ESP_LOGE(GATTC_TAG, "set before read flag successful");
    }
    ESP_ERROR_CHECK(err);
    
     gl_profile_tab[PROFILE_A_APP_ID].char_handle = handle;
     esp_ble_gattc_register_for_notify (gattc_if, gl_profile_tab[PROFILE_A_APP_ID].remote_bda, handle);
     ESP_LOGI(GATTC_TAG, "ble notify successful, char_handle:%d, device mac:" MAC_FMT " ", handle, 
                                    MAC_PARAM(gl_profile_tab[PROFILE_A_APP_ID].remote_bda));
   
    err = esp_ble_gattc_write_char_descr(gattc_if, p_data->search_cmpl.conn_id, handle, sizeof(notify_en), 
                                                    (uint8_t *)&notify_en, ESP_GATT_WRITE_TYPE_RSP, ESP_GATT_AUTH_REQ_NONE);
    if(ESP_OK != err)
    {
        ESP_LOGE(GATTC_TAG, "notify enable set flag error");
    }else
    {
         ESP_LOGE(GATTC_TAG, "notify enable set flag successful");
    }
    ESP_ERROR_CHECK(err);

    
    err = esp_ble_gattc_read_char_descr(gattc_if, p_data->search_cmpl.conn_id, handle,ESP_GATT_AUTH_REQ_NONE);
    if(ESP_OK != err)
    {
       ESP_LOGE(GATTC_TAG, "set after read notify flag error");
    }else
    {
       ESP_LOGE(GATTC_TAG, "set after read flag successful");
    }
    ESP_ERROR_CHECK(err);
    
}

notify的handle值为15，并且也写入成功了，还是不能正常接收数据。
 打印日志如下：
I (3950) GATTC_DEMO: esp get desc 0 info, char_handle:15,uuid_len:2, uuid16:2902 uuid32:2902,uuid128:0000805f-9b01-3110-000e-000000002902
I (3960) GATTC_DEMO: find enable notify desc uuid:2902
E (3970) GATTC_DEMO: set before read flag successful
I (3970) GATTC_DEMO: ble notify successful, char_handle:15, device mac:d8:b0:4c:cc:6f:5b
E (3980) GATTC_DEMO: notify enable set flag successful
E (3990) GATTC_DEMO: set after read flag successful

读事件内容如下：
I (4070) GATTC_DEMO: read  status:0, conn_id:0,handle:15, value_len:2 value:
I (4080) GATTC_DEMO: 00 00
I (4100) GATTC_DEMO: write descr success
I (4130) GATTC_DEMO: read  status:0, conn_id:0,handle:15, value_len:2 value:
I (4140) GATTC_DEMO: 01 01

写入的值，试了0x0101和0x01，蓝牙都不能正常接收。

Weijian-Espressif, 2019-07-12:
@xiaodingding Please test our gatt_client and gatt_server example, they will automatically connect, send and receive data. Please try, thanks.

xiewenxiang, 2019-07-16:
@xiaodingding  看log是写ccc的值不对，写过来的应该是 0x1，你log中写过来的是  0x0 和 0x0101

Alvin1Zhang, 2019-07-19:
@xiaodingding Hi. Would you help share if any updates for the issue? Thanks.

xiaodingding, 2019-07-21:
I have solved this problem. Thank you。

xiaodingding, 2019-07-21:
I have solved this problem. Thank you

closed 2019-07-21 by xiaodingding