espressif/esp-idf #3761: Xtensa-esp32-elf-gcc weird behavior on linker process

eminarcissus, 2019-07-11:
Hi,

I'm now trying to build a native plugin for ESP32 Micropython. The codebase is this part https://github.com/micropython/micropython/pull/1627/files.

The plugin is quite straight forward, it defines a 16 bytes header and at the 0x10 starts with init function and all labels are other things are placed after the init function.

Currently the problem is after I did the compiling ,linker process will always failed with 
l32r: literal placed after use :mpytext.literal

The Linker script is like this.

  SECTIONS
 {
     .all :
     {
         . = ALIGN(4);
         KEEP(*(.mpyheader))
         KEEP(*(.mpytext))
         *(.text*)
         *(.rodata*)
         . = ALIGN(4);
     }
 }

which I believe is missing literal label somehow, but even if I added mpytext.literal after mpytext, it still have such a problem, so I did assume maybe init function have some variables have dependencies on literal. So I did give it a try here.

After I added a like *(.mpytext.literal*) in before or after mpyheader(right before mpytext which is the init func), it compiles without a problem, but that didn't solve the problem, the original plugin is a fixed formatted binary which generated the 16 bytes header and at 0x10 there must be the init function. which the modified version linker script comes with the problem, is there anyway to workaround with the compiling process it make it working here?(x86/x64/arm gcc all build without any problem though).

projectgus, 2019-07-11:
Hi @eminarcissus ,

For future reference: as this doesn't relate to an issue with ESP-IDF, a better place to ask this question would be the esp32.com forums.

What gcc options do you use? Suggest that passing `-mtext-section-literals` may help with this as it will place the literals into the .text section.

Cribbing from the esptool flasher stub may help, as it's a similar purpose (although that code is not for dynamic loading as such):
https://github.com/espressif/esptool/blob/master/flasher_stub/Makefile#L69

(At minimum you'll probably also need `-mlongcalls` in order to call out to any functions in the MicroPython firmware. I don't suggest adding `-flto`, though.)

eminarcissus, 2019-07-11:
Thanks very much,

I was quite frustrating where to post this issue as well. Will do so next time :)

I did tried -mtext-section-literals before, but it did come with some quite interesting layout after linking.

build-xtensa/modx.elf:     file format elf32-little

SYMBOL TABLE:
00000000 l    d  .all	00000000 .all
00000000 l    d  .comment	00000000 .comment
00000000 l    d  .xtensa.info	00000000 .xtensa.info
00000000 l    d  .xt.lit	00000000 .xt.lit
00000000 l    d  .xt.prop	00000000 .xt.prop
00000000 l    df *ABS*	00000000 modx.c
000000e8 l     F .all	00000014 modx_add1
000000fc l     F .all	00000032 modx_make_list
0000002c g     F .all	000000ba init
00000000 g     O .all	00000010 header

quite weirdly init is put at 0x2c,and between 0x10 and 0x2c there was indeed still the literal headers, I have no idea what's going on beyond the scene here.

Does it have something to do with l32r? Looks like from the manual it can only load constants from a negative offset from the PC. So the value relevant to l32r will eventually put in before the function?

igrr, 2019-07-11:
That looks correct, in Xtensa the literal values should precede the l32r instruction which uses them.
If your binary format requires a fixed entry point at the start of the program, you can arrange the program like this:

```
entry section:
  j _start (single instruction)
literal section:
 (literals)
text:
 _start (and the rest of the code)
```

The unconditional jump instruction has the range of 128kB, so it should probably be enough to jump over the literals.

eminarcissus, 2019-07-11:
Wow, didn't thought about that, that should work, will try it out :)
Thx very much

closed 2019-07-13 by eminarcissus