espressif/esp-idf #3762: wifi_ap_config_t\wifi_sta_config_t : ssid_len support not equal

filimonic, 2019-07-11:
wifi_ap_config_t says:
`uint8_t ssid[32]; /**< SSID of ESP32 soft-AP. If ssid_len field is 0, this must be a Null terminated string. Otherwise, length is set according to ssid_len. */`

wifi_sta_config_t says:
`uint8_t ssid[32];      /**< SSID of target AP. Null terminated string. */`

There is something wrong about different ways to declare SSID:
- STA mode does not support `ssid_len`
- STA mode does not support setting SSID not null-terminated
- STA mode does not support 32-byte length SSIDs ( only 31-byte SSIDs due to null termination byte)

I thing someone need to deprecate `ssid_len` from `wifi_ap_config_t` or add `ssid_len` to `wifi_sta_config_t`

liuzfesp, 2019-07-14:
Thanks @filimonic, we will consider to improve it. @zhangyanjiaoesp

HaydnLiao, 2019-12-02:
I had the same question.32 is the standard length.

zhangyanjiaoesp, 2019-12-02:
@filimonic @HaydnLiao We will modify this, but it may not be that high priority. Maybe in v5.0

zhangyanjiaoesp, 2020-11-02:
@filimonic @HaydnLiao 
I have tested locally with the `station example`, the ESP32 STA can set the SSID to 32 len.

> I (609) wifi:wifi driver task: 3ffc1390, prio:23, stack:3584, core=0
I (609) system_api: Base MAC address is not set
I (609) system_api: read default base MAC address from EFUSE
I (629) wifi:wifi firmware version: d5c1f19
I (629) wifi:wifi certification version: v7.0
I (629) wifi:config NVS flash: enabled
I (629) wifi:config nano formating: disabled
I (639) wifi:Init data frame dynamic rx buffer num: 32
I (639) wifi:Init management frame dynamic rx buffer num: 32
I (649) wifi:Init management short buffer num: 32
I (649) wifi:Init dynamic tx buffer num: 32
I (659) wifi:Init static rx buffer size: 1600
I (659) wifi:Init static rx buffer num: 10
I (659) wifi:Init dynamic rx buffer num: 32
I (669) wifi_init: rx ba win: 6
I (669) wifi_init: tcpip mbox: 32
I (679) wifi_init: udp mbox: 6
I (679) wifi_init: tcp mbox: 6
I (679) wifi_init: tcp tx win: 5744
I (689) wifi_init: tcp rx win: 5744
I (689) wifi_init: tcp mss: 1440
I (689) wifi_init: WiFi IRAM OP enabled
I (699) wifi_init: WiFi RX IRAM OP enabled
I (809) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (809) wifi:mode : sta (cc:50:e3:92:a8:e4)
I (819) wifi:enable tsf
I (819) wifi station: wifi_init_sta finished.
I (939) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (939) wifi:state: init -> auth (b0)
I (949) wifi:state: auth -> assoc (0)
I (959) wifi:state: assoc -> run (10)
I (959) wifi:connected with qwertyuiopasdfghjklzxcvbnmqwerty, aid = 1, channel 1, BW20, bssid = 30:ae:a4:0c:34:ed
I (959) wifi:security: Open Auth, phy: bg, rssi: -11
I (979) wifi:pm start, type: 1
I (1029) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (1599) esp_netif_handlers: sta ip: 192.168.4.3, mask: 255.255.255.0, gw: 192.168.4.1
I (1599) wifi station: got ip:192.168.4.3
I (1599) wifi station: connected to ap SSID:qwertyuiopasdfghjklzxcvbnmqwerty password:

closed 2020-12-13 by espressif-bot

Emill, 2022-06-08:
It seems this design is still broken, since an ssid is allowed to contain embedded null characters (contrary to the password). Now it's impossible to set the ssid containing embedded null characters in `wifi_sta_config_t`. See https://superuser.com/questions/1263595/create-network-with-ssid-containing-null-characters-0.

julian-weinert, 2026-08-19:
It still appears broken to me in IDF 6.0