espressif/esp-idf #3763: SIM800L and roaming

Arthedian, 2019-07-11:
## Environment

- Development Kit:    none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    v3.3-beta3-34-g8c57aa024
- Build System:         CMake
- Compiler version crosstool-ng-1.22.0-80-g6c4433a)
- Operating System: Fedora
- Power Supply: external 5V

## Problem Description

I am using part of the example code for SIM800L and it is working perfectly with homeland simcard. But when i use foreign simcard, it stops working (so simcard in roaming). Is this normal, or the code doesnt support roaming?

### Actual Behavior

part of terminal output:

> E (43437) esp-modem: Frame Error
> E (43487) esp-modem: Frame Error
> E (43537) esp-modem: Frame Error
> E (43637) esp-modem: Frame Error
> E (43687) esp-modem: Frame Error
> E (43737) esp-modem: Frame Error
> E (43787) esp-modem: Frame Error
> E (43837) esp-modem: Frame Error
> W (43887) esp-modem: Rx Break

### Steps to repropduce

use simcard in roaming.

suda-morris, 2019-07-11:
I'm sorry, I also have no idea because we haven't test that case yet. Even I don't know if SIM800L supports roaming. Maybe I think you should ask support from simcom.

Arthedian, 2019-07-11:
SIM800L is able to connect to roaming, but it returns different value for AT+CREG? command.

Arthedian, 2019-07-17:
@suda-morris could you please test it?

suda-morris, 2020-06-01:
Sorry for the late reply, roaming is not a common scenario in IoT application, we don't have any plan to support this feature.

closed 2020-06-01 by suda-morris