espressif/esp-idf #3765: Cache memory disabled but cached memory accessed

sapuz1508, 2019-07-11:
Hi, everyone.
I'm trying to use an esp32 to get some stuff done, at a certain point i need to update the esp OTA.
In order to do this i have a webserver, which holds the update (.bin file), and a database from which the esp reads the filename and the md5 of the update.
The code i use to update is httpFOTA by nhatuan84.
This code just works fine until i insert this in the code below.

main.ino: 
```
            //recieve the infos from the db.
            vTaskDelete(resetTask); //delete the other tasks I run on core 0
            vTaskDelete(wiegandTask); //same as above

            isOtaRunning = true; 

            OTA();
```

OTA func:
```
            char controllo[33]; //get the md5

           String tmp = String(md5ora);
           tmp.toCharArray(controllo, 33); //some shitty casting

           OTAUp OTAUpdate=OTAUp(url,controllo,NULL); //declaring the obj for the update (i actually transformed that library I mentioned before into a oop one. nothing much changed, i think.)
           if(OTAUpdate.startDownload() == 0) //check if the update is successful
           {
                        Serial.println("OTA successful");
           }
  
           Serial.println("undone. whops.");
```

The actual code of OTAUp::startDownload is the code of the httpFOTA library. But as i just said if works fine in other 5 codes i made.

The problem is that when i start downloading the firmware it crashes with the following log:

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40195fe8: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered () 
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1 panic'ed (Double exception)
 ⸮?8⸮?X⸮?x⸮?Error: Core  1⸮@⸮Unhandled debug exception)
Debu⸮@6

```

Dunno what to do. Pls help.

igrr, 2019-07-12:
@sapuz1508 Could you please add the information requested in the issue template, such as IDF version, build system used, etc. Also, could you please use IDF Monitor to decode the addresses in the log output? Finally, would you mind attaching your program's ELF file?

sapuz1508, 2019-07-13:
@igrr
I use the Arduino IDE to run the code.
1.0.2 version
ESP-WROOM-32
External 5V power supply

That one is the only coredump I get, and the decoded didn’t tell me nothing more.
I’d like not to send the elf file.. I hope that’s not a problem :/

If I missed something tell me, I’m quite a newbie into development with this devices and also in such low level programming.

Thx~

negativekelvin, 2019-07-13:
Since you are using Arduino and third party libraries, this is more appropriate to post on esp32.com forum.  This is probably not a bug in esp-idf. But I would suggest do not use weigand task and see if the problem happens, if not probably something with an interrupt in weigand driver.  Also in general don't just delete tasks like that.

sapuz1508, 2019-07-13:
@negativekelvin what if I must use that task? I know it’s probably there the problem, in fact I tried deleting or suspending it before calling the function that causes the crash.
And what you mean with “don’t just delete tasks like that”?

negativekelvin, 2019-07-13:
You should send a signal to a task to clean up its resources first and then delete itself.

sapuz1508, 2019-07-17:
Got it.
Btw, i tried not creating that task at all in order to get OTA Update completed.
But it didnt work. So.. i cant understand what that error is. can y'all help me?

chegewara, 2019-07-17:
`Debug exception reason: Stack canary watchpoint triggered () `
Too small task's stack.

```
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1 panic'ed (Double exception)
```
@me-no-dev said once on gitter that is probably because there is not enough memory to generate exception, so you are most likely completely out of heap.

sapuz1508, 2019-07-17:
How should i calculate the stack? @chegewara

chegewara, 2019-07-17:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/freertos.html#_CPPv427uxTaskGetStackHighWaterMark12TaskHandle_t

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/heap_debug.html

sapuz1508, 2019-07-17:
@chegewara 

```
Free memory = 46940
state = 2 - percent = 0
Free memory = 46940
state = 2 - percent = 0
Free memory = 46940
state = 2 - percent = 0
Free memory = 46940
state = 2 - percent = 0
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered () 
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1 panic'ed (Double exception)
d00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401956bc: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered () 
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1 panic'ed (Double exception)

```

im plenty of free memory, so i cant understand

chegewara, 2019-07-17:
I said probably lack of heap, but this is problem with stack size for sure:
`Debug exception reason: Stack canary watchpoint triggered`

I am using this code to control stack size from time to time:
```
static void _tasksStatus(void* p)
{
    UBaseType_t uxArraySize = uxTaskGetNumberOfTasks();
    TaskStatus_t pxTaskStatusArray[uxArraySize];
    uint32_t pulTotalRunTime = 0;

    uxTaskGetSystemState(&pxTaskStatusArray[0], uxArraySize, &pulTotalRunTime);

    for(int i=0; i<uxArraySize; i++)
    {
        ESP_LOGI(__func__, "[%d]Task %s => stack remaining: %d", i, pxTaskStatusArray[i].pcTaskName, pxTaskStatusArray[i].usStackHighWaterMark);
    }
    heap_caps_print_heap_info(MALLOC_CAP_DEFAULT);
    ESP_LOGI(__func__, "stack remaining: %d", uxTaskGetStackHighWaterMark( NULL ));
    vTaskDelete(NULL);
}

void tasksStatus()
{
    xTaskCreate(_tasksStatus, "stackInfo", 5*612, NULL, 1, NULL);
}
```

negativekelvin, 2019-07-17:
The stack canary isn't triggered until well into the looping exception.  The root cause is still Cache disabled but cached memory region accessed

sapuz1508, 2019-07-17:
@chegewara sorry but i cant still make uxTaskGetSysyemState() working even though i've already put configUSE_TRACE_FACILITY to 1 in config file..

@negativekelvin ya, and im trying to understand why this happens only when i call that function in this code..
it doesnt give me an appropriate stacktrace, so i cant understand where the problem is..

umer-ilyas, 2020-03-02:
my controller stuck in this 

Guru Meditation Error: Core 1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC : 0x15383520 PS : 0x00060c30 A0 : 0x800dfcac A1 : 0x3fff7590
A2 : 0x00000000 A3 : 0x3ffb965c A4 : 0x15383520 A5 : 0x6c000001
A6 : 0x003fff74 A7 : 0x60000000 A8 : 0x800e0c11 A9 : 0x3fff7460
A10 : 0x3fff7590 A11 : 0x312e3738 A12 : 0x00000115 A13 : 0x3fff746c
A14 : 0x00000000 A15 : 0x3fff7560 SAR : 0x00000008 EXCCAUSE: 0x00000014
EXCVADDR: 0x15383520 LBEG : 0x4000c349 LEND : 0x4000c36b LCOUNT : 0x00000000

ELF file SHA256:

nileshkale123, 2024-09-27:
Hello @sapuz1508 

We are closing this issue as it is not related to the IDF OTA components. For similar questions, we recommend posting them on the [ESP32 forum](https://www.esp32.com/viewforum.php?f=2).

closed 2024-09-27 by nileshkale123