espressif/esp-idf #3766: Warning error on current version of master branch

KaeLL, 2019-07-11:
I get the felling dev's don't build esp-idf prior to pushing changes, do they? LoL

esp-idf/components/bt/common/osi/config.c:446:49: error: '%d' directive output may be truncated writing between 1 and 10 bytes into a region of size 7 [-Werror=format-truncation=]
             snprintf(keyname, keyname_bufsz, "%s%d", CONFIG_KEY, i);

Weijian-Espressif, 2019-07-12:
@KaeLL, Sorry for that. we have fixed the bug, but have not update to the github.
[image]

Weijian-Espressif, 2019-07-12:
[image]

Alvin1Zhang, 2019-12-09:
@KaeLL Thanks for reporting, the issue has been fixed and available. Please help try, and feel free to reopen if the issue still happens. Thanks.

closed 2019-12-09 by Alvin1Zhang