espressif/esp-idf #3767: ESP-IDF-TOOL installer crash

Ostpopcorn, 2019-07-11:
Hi,
I just this afternoon tried to update the IDF i thought that i messed somethings up and therefor tried to reinstall everything. To my satisfaction the version 2.0 of the ESP-IDF-TOOLS had been released so i tried to install it on my machine. 
After running for a few minutes it crashed, se log below. 
I'm trying it with the development-brach of the IDF and python 3.7 (both downloaded today 11/7). For building after I'm using Cmake so i can code i Clion.
I also tried to remove esp-idf to let the installer download it but it didn't help either.

It still works to run the idf.py in a powershell windows where first export.bat has wore executed

This is the lines where the error seems to appear.

```
Running virtualenv with interpreter C:\Program Files\Python37\python.exe
                          Creating a new Python environment in C:\Users\adrian\.espressif\python_env\idf4.0_py3.7_env
                          Traceback (most recent call last):
                            File "C:\Users\adrian\esp\esp-idf\tools\idf_tools.py", line 1249, in <module>
                              main(sys.argv[1:])
                            File "C:\Users\adrian\esp\esp-idf\tools\idf_tools.py", line 1245, in main
                              action_func(args)
                            File "C:\Users\adrian\esp\esp-idf\tools\idf_tools.py", line 1069, in action_install_python_env
                              stdout=sys.stdout, stderr=sys.stderr)
                            File "C:\Program Files\Python37\lib\subprocess.py", line 347, in check_call
                              raise CalledProcessError(retcode, cmd)
                          subprocess.CalledProcessError: Command '['C:\\Program Files\\Python37\\python.exe', '-m', 'virtualenv', '--no-site-packages', 'C:\\Users\\adrian\\.espressif\\python_env\\idf4.0_py3.7_env']' returned non-zero exit status 4294967295.

                          
2019-07-11 17:52:55.997   --------
2019-07-11 17:52:55.997   Message box (OK):
                          Installation has failed with exit code 1
2019-07-11 17:53:01.735   User chose OK.
2019-07-11 17:53:01.751   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2019-07-11 17:53:03.578   User chose Yes.
2019-07-11 17:53:03.703   CurStepChanged raised an exception.
```

The log file can be found here 
[Setup Log 2019-07-11 #004.txt](https://github.com/espressif/esp-idf/files/3383046/Setup.Log.2019-07-11.004.txt)

igrr, 2019-07-12:
@Ostpopcorn Could you please try to open a new cmd.exe window, and run there:

```
C:\Program Files\Python37\python.exe -m virtualenv --no-site-packages C:\Users\adrian\.espressif\python_env\idf4.0_py3.7_env
```

Does this cause the same kind of error? If yes, can you please check if virtualenv is able to create environments at all? (run `C:\Program Files\Python37\python.exe -m virtualenv --no-site-packages <any-other-location>`)

Ostpopcorn, 2019-07-12:
Hi,
I just tested in a blank directory on my desktop and it loops the same way. Watching the process in task manager reveals that it allocates more and more ram to itself. It also dose something recursive because it starts 3 processes per second.

Trying now to reinstall python37 and seeing if that dose something.
Ok, after reinstalling the TOOLS one more time I noticed that when it says `Running command: C:\Program Files\Python37\python.exe C:\Users\adrian\esp\esp-idf\tools\idf_tools.py install-python-env` in the installer terminal it build up thousands of processes one after another (recursive?). Seemingly until a point where it runs out of memory and then every one of the processes print `Running virtualenv with interpreter C:\Program Files\Python37\python.exe`. Finally the last process prints the error message in a messagebox.
Hope it help, I will be happy to provide more info if needed :)

igrr, 2019-07-12:
> I just tested in a blank directory on my desktop and it loops the same way.

It seems that the cause of this is https://github.com/pypa/virtualenv/issues/1380. I will check the workarounds mentioned and get back to you.

igrr, 2019-07-12:
@Ostpopcorn can you please try to downgrade Python to 3.7.3? It seems that for 3.7.4 we need to wait for the next virtualenv release to fix this issue.

Ostpopcorn, 2019-07-12:
Hi,
That was the trick! Installer worked fine now and i was able to build a project!
Thanks @igrr

closed 2019-07-12 by Ostpopcorn

isakovM, 2021-06-11:
hello dear supporter. i am trying to install esp-idf on vs code. but at the end of installation some problems appears. please have look at this picture
[image]