espressif/esp-idf #3768: Unclear documentation: Is malloc task/threadsafe?

raldone01, 2019-07-11:
I carefully read through https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/mem_alloc.html but I couldn't find anything regarding task/thread safety.

Do I have to guard malloc and free calls from concurrent access?
I would suggest to update the documentation and add that info.

PerMalmberg, 2019-07-11:
No, you do not. It's guaranteed by the language (C & C++), assuming you're using a modern version of it, (C11 ?). I have a hard time seeing you running into issues if you use a tool chain provided by Espressif though.

Note however that calling malloc and friends from an interrupt is not safe as the interrupt may have interrupted a call to malloc/free etc.

raldone01, 2019-07-11:
Thank you I didn't know that was the case as the memory functions are written by expressif.

negativekelvin, 2019-07-11:
https://github.com/espressif/esp-idf/blob/81ca1c01395f604972fbf141cfbe49764a746023/components/heap/multi_heap_platform.h#L22

raldone01, 2019-07-11:
Thank you. So is malloc safe in ISRs?

PerMalmberg, 2019-07-11:
> https://github.com/espressif/esp-idf/blob/81ca1c01395f604972fbf141cfbe49764a746023/components/heap/multi_heap_platform.h#L22-L35

Ah, seems an ISR is allowed to call malloc too, that's nice.

raldone01, 2019-07-11:
So the documentation is unclear?

projectgus, 2019-07-12:
@raldone01 malloc(), free() and thread safe.

It is technically possible to malloc() or free() from an ISR. It is not recommended to call malloc() or free() from an ISR as they can take al ong time to run, especially if you have PSRAM with a lot of small allocations in it. Better to structure your code in a way that you pre-allocate any buffer that is needed by the ISR.

projectgus, 2019-07-15:
Will reopen as I've made a clarification in the docs about this, but it hasn't pushed to GItHub yet. This issue will be closed once it has.

closed 2019-07-15 by igrr