espressif/esp-idf #3771: Cmake build System Error : Syntax error in cmake code

Yuvaraj4041, 2019-07-12:
I installed **esp-idf-tools-setup-2.0.exe.** Then i am trying to do menuconfig by idf.py menuconfig. But it was not working.

With idf-tools-setup python 3.7, git 2.21.0, ESP-IDF-V4.0 are installed. 
I am using Windows 7 64-bit operating System.

### **Below is my Log**

Using Python in C:\Users\Yuvaraj\AppData\Local\Programs\Python\Python37\
Python 3.7.3
Using Git in C:\Program Files\Git\cmd\
git version 2.21.0.windows.1
Setting IDF_PATH: C:\esp32-cmake\esp-idf

Adding ESP-IDF tools to PATH...
    C:\ESP32-S2\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin
    C:\ESP32-S2\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin
    C:\ESP32-S2\.espressif\tools\cmake\3.13.4\bin
    C:\ESP32-S2\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin
    C:\ESP32-S2\.espressif\tools\mconf\v4.6.0.0-idf-20190628\
    C:\ESP32-S2\.espressif\tools\ninja\1.9.0\
    C:\ESP32-S2\.espressif\tools\idf-exe\1.0\
    C:\ESP32-S2\.espressif\tools\ccache\3.7\
    C:\ESP32-S2\.espressif\python_env\idf4.0_py3.7_env\Scripts

Checking if Python packages are up to date...
Python requirements from C:\esp32-cmake\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run: idf.py build

### **Then i am trying to do menu configuration.** 

**C:\esp32-cmake\esp-idf\examples\get-started\hello_world>
C:\esp32-cmake\esp-idf\examples\get-started\hello_world>idf.py menuconfig**

Checking Python dependencies...
Python requirements from C:\esp32-cmake\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory c:\esp32-cmake\esp-idf\examples\get-started\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninit
ialized -DCCACHE_ENABLE=0 c:\esp32-cmake\esp-idf\examples\get-started\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/ESP32-S2/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/ESP32-S2/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/ESP32-S2/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/ESP32-S2/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/ESP32-S2/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v4.0-dev-1191-g138c941fa
-- Building ESP-IDF components for target esp32
CMake Error at C:/esp32-cmake/esp-idf/tools/cmake/component.cmake:225 (message):
CMake Warning (dev) at build_properties.temp.cmake:2 (set):

    Syntax error in cmake code at C:/esp32-cmake/esp-idf/examples/get-started/hello_world/build/build_properties.temp.cmake:2

    when parsing string C:\ESP32-S2\.espressif\python_env\idf4.0_py3.7_env\Scripts\python.exe

    Invalid escape sequence \E

    Policy CMP0010 is not set: Bad variable reference syntax is an error.  Run
    "cmake --help-policy CMP0010" for policy details.  Use the cmake_policy
    command to set the policy and suppress this warning.

  Call Stack (most recent call first):
    C:/esp32-cmake/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:1 (include)
  This warning is for project developers.  Use -Wno-dev to suppress it.
  CMake Error at C:/esp32-cmake/esp-idf/components/efuse/CMakeLists.txt:4
 (include):
    include could not find load file: /esp32/sources.cmake

  Call Stack (most recent call first):
    C:/esp32-cmake/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:
56 (include)
    C:/esp32-cmake/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:
72 (__component_get_requirements)

Call Stack (most recent call first):
  C:/esp32-cmake/esp-idf/tools/cmake/build.cmake:386 (__component_get_requirements)
  C:/esp32-cmake/esp-idf/tools/cmake/project.cmake:341 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/esp32-cmake/esp-idf/examples/get-started/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

C:\esp32-cmake\esp-idf\examples\get-started\hello_world>

renzbagaporo, 2019-07-12:
HI @Yuvaraj4041, you mentioned using v4.0. Are you on the latest master?

Yuvaraj4041, 2019-07-12:
Hi @renzbagaporo, Thanks.

Yes . While installing esp-idf-tools-setup-2.0.exe, it asked to download ESP-IDF. I choosed master branch.

renzbagaporo, 2019-07-12:
This looks like a regression with a problem that was solved previously. In the meantime, can you try adding the following line:

```
set(COMPONENT_DIR ${CMAKE_CURRENT_LIST_DIR})
```

after 

```
idf_build_get_property(soc_name IDF_TARGET)
```

for both `components/soc/CMakeLists.txt` and `components/efuse/CMakeLists.txt`

Yuvaraj4041, 2019-07-12:
@renzbagaporo 

It Works....! Thank you.

renzbagaporo, 2019-07-12:
Ok. This issue should automatically get closed once the fix gets merged @Yuvaraj4041

closed 2019-07-15 by igrr

neorevx, 2020-03-24:
This problem are back. The slashs shloud be inverted.

```
CMake Warning (dev) at build_properties.temp.cmake:2 (set):

    Syntax error in cmake code at

      D:/OneDrive/ESP32-Projects/vcar/build/build_properties.temp.cmake:2

    when parsing string

      D:\OneDrive\ESP32-3Tools\python_env\idf4.2_py3.7_env\Scripts\python.exe

    Invalid escape sequence \O
```

baptistepattyn, 2020-03-25:
Yes I am currently having the same problem, each time I edit the file it changes it again...

MihirShahLNM, 2020-07-05:
> 
> 
> Yes I am currently having the same problem, each time I edit the file it changes it again...

Me too !

MihirShahLNM, 2020-07-05:
> This problem are back. The slashs shloud be inverted.
>
> ```
> CMake Warning (dev) at build_properties.temp.cmake:2 (set):
>
>     Syntax error in cmake code at [...]

I am having exactly same problem.

nabeelkirmani, 2020-07-27:
> This looks like a regression with a problem that was solved previously. In the meantime, can you try adding the following line:
>
> ```
> set(COMPONENT_DIR ${CMAKE_CURRENT_LIST_DIR})
> ```
>
> after
>
> ``` [...]

I did the same as instructed above but the issue still remains.

Here is output of _ESP-IDF Command Prompt (cmd.exe)_ :
````powershell

Using Python in C:\Program Files\Python38\
Python 3.8.5
Using Git in C:\Program Files\Git\cmd\
git version 2.27.0.windows.1
Setting IDF_PATH: E:\Source\ESP-IDF\esp-idf

Adding ESP-IDF tools to PATH...
    E:\Program Files\ESP-IDF Tools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin
    E:\Program Files\ESP-IDF Tools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    E:\Program Files\ESP-IDF Tools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    E:\Program Files\ESP-IDF Tools\.espressif\tools\idf-exe\1.0.1\
    E:\Program Files\ESP-IDF Tools\.espressif\tools\ccache\3.7\
    E:\Program Files\ESP-IDF Tools\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    E:\Program Files\ESP-IDF Tools\.espressif\python_env\idf4.3_py3.8_env\Scripts

Checking if Python packages are up to date...
Python requirements from E:\Source\ESP-IDF\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

E:\Source\ESP-IDF\esp-idf>cd examples\get-started\hello_world

E:\Source\ESP-IDF\esp-idf\examples\get-started\hello_world>idf.py build
Executing action: all (aliases: build)
Running cmake in directory e:\source\esp-idf\esp-idf\examples\get-started\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=1 e:\source\esp-idf\esp-idf\examples\get-started\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.27.0.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at E:/Source/ESP-IDF/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Error at build_properties.temp.cmake:2 (set):

    Syntax error in cmake code at

      E:/Source/ESP-IDF/esp-idf/examples/get-started/hello_world/build/build_properties.temp.cmake:2

    when parsing string

      E:\Program

    Invalid character escape '\P'.

  Call Stack (most recent call first):

    E:/Source/ESP-IDF/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:2 (include)

  CMake Error at build_properties.temp.cmake:2 (set):

    Syntax error in cmake code at

      E:/Source/ESP-IDF/esp-idf/examples/get-started/hello_world/build/build_properties.temp.cmake:2

    when parsing string

      Files\ESP-IDF

    Invalid character escape '\E'.

  Call Stack (most recent call first):

    E:/Source/ESP-IDF/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:2 (include)

  CMake Error at build_properties.temp.cmake:2 (set):

    Syntax error in cmake code at

      E:/Source/ESP-IDF/esp-idf/examples/get-started/hello_world/build/build_properties.temp.cmake:2

    when parsing string

      Tools\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe

    Invalid character escape '\p'.

  Call Stack (most recent call first):

    E:/Source/ESP-IDF/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:2 (include)

Call Stack (most recent call first):
  E:/Source/ESP-IDF/esp-idf/tools/cmake/build.cmake:405 (__component_get_requirements)
  E:/Source/ESP-IDF/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "E:/Source/ESP-IDF/esp-idf/examples/get-started/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

E:\Source\ESP-IDF\esp-idf\examples\get-started\hello_world>

````

and here is the attached CMakeOutput.log file 
[CMakeOutput.log](https://github.com/espressif/esp-idf/files/4982954/CMakeOutput.log)

mazorvanunu, 2020-09-09:
is this issue resolved??? i have tried with branches 4.0 4.0.1 4.1, 4.2, 4.3
all exhibit the same (faulty) behavior.. nothing in the getting started guide.
clearly, at least to me, somebody from the build team is not aware of how things are done in windows (at the very least in windows, i usually do not work on linux but maybe thats the way to do things there..)