espressif/esp-idf #3773: esp32外接lan8720报错

mulinluo, 2019-07-12:
我使用esp32外接lan8720模块做ethernet。使用esp-idf-release-v3.2的Ethernet示例代码进行编译。clock信号使用的是lan8720模块上的50MHz晶振，nINT/REFCLKO引脚与gpio0连接，模式选择为Ethernet PHY Clock Mode (GPIO0 Input) 。编译后报错E (1301) emac: Reset EMAC Timeout。
具体log如下：
I (0) cpu_start: App cpu up.
I (232) heap_init: Initializing. RAM available for dynamic allocation:
I (239) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (245) heap_init: At 3FFB4090 len 0002BF70 (175 KiB): DRAM
I (251) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (257) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (264) heap_init: At 40089B48 len 000164B8 (89 KiB): IRAM
I (270) cpu_start: Pro cpu start user code
I (288) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (291) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
E (1301) emac: Reset EMAC Timeout

请教这是什么原因导致的，我该怎么做？谢谢！

suda-morris, 2019-07-12:
Hi @mulinluo Have you set the PHY address according to your board?

mulinluo, 2019-07-12:
我设置的PHY address为1

suda-morris, 2019-07-15:
I would like to suggest you to connect an oscilloscope to GPIO0 to check if there exists a valid RMII clock. @mulinluo 
And since this "issue" is regarding to an usage question not a bug report or feature request, you'd better to put it on esp32.com forum to get more experience for other users who have run LAN8720 successfully in their product.

Alvin1Zhang, 2019-10-11:
@mulinluo Thanks for reporting, feel free to reopen if you still have the issue. Thanks.

closed 2019-10-11 by Alvin1Zhang