espressif/esp-idf #3774: mqtt over web socket

nopnop2002, 2019-07-12:
----------------------------- Delete below -----------------------------
My environment:

	ESP-IDF version:v4.0-dev-1186-gf16f4297c

I tried esp-idf/examples/protocols/mqtt/ws.
I modified your code:

	static esp_err_t mqtt_event_handler_cb(esp_mqtt_event_handle_t event)
	{
	    esp_mqtt_client_handle_t client = event->client;
	    int msg_id;
	    // your_context_t *context = event->context;
	    switch (event->event_id) {
	        case MQTT_EVENT_CONNECTED:
	            ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
	            msg_id = esp_mqtt_client_subscribe(client, "/topic/qos0", 0);
	            ESP_LOGI(TAG, "sent subscribe successful, msg_id=%d", msg_id);

	#if 0
	            msg_id = esp_mqtt_client_subscribe(client, "/topic/qos1", 1);
	            ESP_LOGI(TAG, "sent subscribe successful, msg_id=%d", msg_id);

	            msg_id = esp_mqtt_client_unsubscribe(client, "/topic/qos1");
	            ESP_LOGI(TAG, "sent unsubscribe successful, msg_id=%d", msg_id);
	#endif
	            break;
	        case MQTT_EVENT_DISCONNECTED:
	            ESP_LOGI(TAG, "MQTT_EVENT_DISCONNECTED");
	            break;

	        case MQTT_EVENT_SUBSCRIBED:
	            ESP_LOGI(TAG, "MQTT_EVENT_SUBSCRIBED, msg_id=%d", event->msg_id);
	            msg_id = esp_mqtt_client_unsubscribe(client, "/topic/qos0");
	            ESP_LOGI(TAG, "sent unsubscribe successful, msg_id=%d", msg_id);
	#if 0
	            msg_id = esp_mqtt_client_publish(client, "/topic/qos0", "data", 0, 0, 0);
	            ESP_LOGI(TAG, "sent publish successful, msg_id=%d", msg_id);
	#endif
	            break;
	        case MQTT_EVENT_UNSUBSCRIBED:
	            ESP_LOGI(TAG, "MQTT_EVENT_UNSUBSCRIBED, msg_id=%d", event->msg_id);
	            break;
	        case MQTT_EVENT_PUBLISHED:
	            ESP_LOGI(TAG, "MQTT_EVENT_PUBLISHED, msg_id=%d", event->msg_id);
	            break;
	        case MQTT_EVENT_DATA:
	            ESP_LOGI(TAG, "MQTT_EVENT_DATA");
	            printf("TOPIC=%.*s\r\n", event->topic_len, event->topic);
	            printf("DATA=%.*s\r\n", event->data_len, event->data);
	            break;
	        case MQTT_EVENT_ERROR:
	            ESP_LOGI(TAG, "MQTT_EVENT_ERROR");
	            break;
	        default:
	            ESP_LOGI(TAG, "Other event id:%d", event->event_id);
	            break;
	    }
	    return ESP_OK;
	}

This is http server in/out log.
Forwarding is incoming message.
Responding is outgoing message.

	I (322307) web_socket: Forwarding 17 bytes
	I (322317) : a2 0f 18 28 00 0b 2f 74 6f 70 69 63 2f 71 6f 73
	I (322317) : 30

	I (321557) web_socket: Forwarding 26 bytes
	I (321567) : 10 18 00 04 4d 51 54 54 04 02 00 78 00 0c 45 53
	I (321567) : 50 33 32 5f 34 66 43 31 41 30

	I (321597) web_socket: Responding 4 bytes
	I (321597) : 20 02 00 00

	I (321867) web_socket: Forwarding 18 bytes
	I (321877) : 82 10 b8 64 00 0b 2f 74 6f 70 69 63 2f 71 6f 73
	I (321877) : 30 00

	I (321907) web_socket: Responding 5 bytes
	I (321907) : 90 03 b8 64 00

	I (322277) web_socket: Forwarding 18 bytes
	I (322287) : 82 10 b8 64 00 0b 2f 74 6f 70 69 63 2f 71 6f 73
	I (322287) : 30 00

	I (322307) web_socket: Forwarding 17 bytes
	I (322317) : a2 0f 18 28 00 0b 2f 74 6f 70 69 63 2f 71 6f 73
	I (322317) : 30

	I (322337) web_socket: Forwarding 17 bytes
	I (322347) : a2 0f 18 28 00 0b 2f 74 6f 70 69 63 2f 71 6f 73
	I (322347) : 30

	I (322387) web_socket: Responding 5 bytes
	I (322387) : 90 03 b8 64 00

	W (324647) web_socket: Illegal length
	I (324657) : a7 a1 73 7d 05 a5 44 21 6a de 02 36 2a df 04 26
	I (324657) : 35

	W (326777) web_socket: Illegal length
	I (326787) : 08 fa 83 28 aa fe b4 74 c5 85 f2 63 85 84 f4 73
	I (326787) : 9a

	W (328627) web_socket: Illegal length
	I (328637) : ff d1 24 35 5d d5 13 69 32 ae 55 7e 72 af 53 6e
	I (328637) : 6d

	W (330777) web_socket: Illegal length
	I (330787) : a0 a0 1b e8 02 a4 2c b4 6d df 6a a3 2d de 6c b3
	I (330787) : 32

Q1)
I execute esp_mqtt_client_subscribe() only one time.
But there is same subscribe message:
82 10 b8 64 00 0b 2f 74 6f 70 69 63 2f 71 6f 73 30 00

Q2)
There is no Unsubscribe response.
There is many Forwarding message like this:

	I (328637) : ff d1 24 35 5d d5 13 69 32 ae 55 7e 72 af 53 6e
	I (328637) : 6d

david-cermak, 2019-07-15:
Hi @nopnop2002 

Thanks for reporting this, it's a bug in mqtt library (not only ws layer), it was fixed as https://github.com/espressif/esp-mqtt/commit/d9faeb47a341a73ce5a5112ac74cddb798123bab (not yet referenced from master).
Not sure about the other question, but it could be related (depending on the broker reaction to the fisrt issue).
Could you please test to update mqtt submodule to reference https://github.com/espressif/esp-mqtt/commit/0cc4077bd3e10bb93456ff2785309ec9237a5906 and see if both issues are still present?

nopnop2002, 2019-07-15:
I updated to ESP-IDF version:v4.0-dev-1275-gfdab15dc7.
There is no Unsubscribe response.
There is many Forwarding message like this:

	 (41624) web_socket: Forwarding 26 bytes
	I (41634) : 10 18 00 04 4d 51 54 54 04 02 00 78 00 0c 45 53
	I (41634) : 50 33 32 5f 34 37 35 32 36 43

	I (41664) web_socket: Responding 4 bytes
	I (41664) : 20 02 00 00

	I (42034) web_socket: Forwarding 18 bytes
	I (42044) : 82 10 12 aa 00 0b 2f 74 6f 70 69 63 2f 71 6f 73
	I (42044) : 30 00

	I (42074) web_socket: Responding 5 bytes
	I (42074) : 90 03 12 aa 00

	I (42554) web_socket: Forwarding 17 bytes
	I (42564) : a2 0f 5e e7 00 0b 2f 74 6f 70 69 63 2f 71 6f 73
	I (42564) : 30

	I (44394) web_socket: Forwarding 17 bytes
	I (44404) : a2 0f 5e e7 00 0b 2f 74 6f 70 69 63 2f 71 6f 73
	I (44404) : 30

	W (45514) web_socket: Illegal length
	I (45524) : ea bf 3e 77 48 bb 4f e4 27 c0 09 f3 67 c1 0f e3
	I (45524) : 78

	W (47364) web_socket: Illegal length
	I (47374) : 4f 54 e9 07 ed 50 98 94 82 2b de 83 c2 2a d8 93
	I (47374) : dd

	W (48384) web_socket: Illegal length
	I (48394) : 6a c6 65 cc c8 c2 14 5f a7 b9 52 48 e7 b8 54 58
	I (48394) : f8

	W (50534) web_socket: Illegal length
	I (50544) : d5 78 cc 9d 77 7c bd 0e 18 07 fb 19 58 06 fd 09
	I (50554) : 47

david-cermak, 2019-07-16:
Indeed there is an issue in web socket transport layer, which is revealed only after second retransmission if broker does not respond. (the problem is that masked ws transmission modifies in-buffer data).

You can work this around using un-masked ws_write in https://github.com/espressif/esp-idf/blob/master/components/tcp_transport/transport_ws.c#L220
```
    return _ws_write(t, WS_OPCODE_BINARY | WS_FIN, 0, b, len, timeout_ms);
```

Thanks for reporting this issue. Please keep this open, needs to be fixed in IDF.

The other issue is that your broker does not seem to accept unsubscribe message (had to silence the responses from my broker to reproduce the issue); cannot help with this one.

nopnop2002, 2019-07-16:
Please use this broker for your test.
This broker does not accept unsubscribe request.

https://github.com/nopnop2002/esp-idf-mqtt-broker

david-cermak, 2019-07-16:
Wow, you're developing an ESP32 mqtt broker! 👍  

It won't retry forever, there's a limit (about a minute) after which the message is cleared out from the queue.

closed 2019-07-29 by igrr