espressif/esp-idf #3775: A2DP audio glitches due to zero packets in output

steenou, 2019-07-12:
## Environment
- Module or chip used: ESP32-WROVER
- IDF version: v3.2-dev
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: external 3.3V

## Problem Description
[This issue that was closed](https://github.com/espressif/esp-idf/issues/3407) but still occurs. The original issue was thought to have been fixed with I2S PLL, but unfortunately, it isn't. The audio glitches and freezes, mostly at the beginning of a stream, but also during.
Using A2DP sink code examples results in glitches in the output. When putting a logic analyser on the I2S outputs of my ESP32-WROVER, I can see the glitches occur due to zeroes in the I2S data line. There is a pattern of zero packets on Left and Right data, always followed by 5 data packets (always Left Right Left Right Left), followed by zeroes again. These zeroes and data packets reoccur a couple times: 
[image]
Sometimes there is a period of only zeroes after
[image]

The amount of clock cycles of the LRCLK (WS) during zeroes and during data is always the same as `i2s_config.dma_buf_len` which is declared in I2S.cpp. For example if I do `i2s_config.dma_buf_len = 8`, I see 5½ samples of zeroes, 2½ samples of data:
[image]
This is true for **any** value of `i2s_config.dma_buf_len`.  n- 2½ samples of  zeroes, 2½ samples of data, reoccurring.

The moments these glitches occur seem to occur when a Bluetooth A2DP stream has just started and randomly throughout the stream too. The glitches seem to occur both when the ESP32 is the I2S master and when the ESP32 is the I2S slave, although it seems to occur less during the stream when the ESP32 is the I2S master.

When the stream is paused, and resumed quickly after (maybe within a window of 3 seconds), there will be no glitches in the output. When the stream is paused, and you wait until 
`I (80520) BT_AV: A2DP audio state: Suspended` has been written in the log, and then resume the stream (e.g. by pressing play again), you will hear glitches in the output.

### Expected Behavior
Smooth audio when running ESP example code.

### Actual Behavior
Glitchy audio.

### Code to reproduce this issue
The a2dp_gatts_coex example code in the IDF.

## Debug Logs
```
E (63693) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (63693) BT_AV: A2DP connection state: Connecting, [94:65:2d:20:56:8b]
I (63793) BT_AV: A2DP audio stream configuration, codec type 0
I (63803) I2S: PLL_D2: Req RATE: 44100, real rate: 44642.000, BITS: 32, CLKM: 14, BCK: 4, MCLK: 11289966.924, SCLK: 2857088.000000, diva: 64, divb: 11
I (63803) BT_AV: Configure audio player 0x21-0x15-0x2-0x35
I (63813) BT_AV: Audio player configured, sample rate=44100
W (63863) BT_APPL: new conn_srvc id:19, app_id:0
I (63863) BT_AV: A2DP connection state: Connected, [94:65:2d:20:56:8b]
I (63973) BT_AV: AVRC conn_state evt: state 1, [94:65:2d:20:56:8b]
I (63973) BT_AV: AVRC remote features 0x4b
I (63993) BT_AV: AVRC metadata rsp: attribute id 0x1, Red Eyes
I (63993) BT_AV: AVRC metadata rsp: attribute id 0x2, The War On Drugs
I (64003) BT_AV: AVRC metadata rsp: attribute id 0x4, Lost In The Dream
I (64003) BT_AV: AVRC metadata rsp: attribute id 0x20, 
I (70923) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b

W (70933) BT_APPL: new conn_srvc id:19, app_id:1
I (70933) BT_AV: A2DP audio state: Started
I (72553) BT_AV: Audio packet count 100
I (74023) BT_AV: Audio packet count 200
I (75473) BT_AV: Audio packet count 300
I (76933) BT_AV: Audio packet count 400
I (78383) BT_AV: Audio packet count 500
I (79833) BT_AV: Audio packet count 600
I (81283) BT_AV: Audio packet count 700
I (82733) BT_AV: Audio packet count 800
I (84183) BT_AV: Audio packet count 900
I (85243) BT_AV: A2DP audio state: Suspended
I (87853) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b

I (87853) BT_AV: A2DP audio state: Started
I (89483) BT_AV: Audio packet count 100
I (90943) BT_AV: Audio packet count 200
I (92393) BT_AV: Audio packet count 300
I (93853) BT_AV: Audio packet count 400
I (95303) BT_AV: Audio packet count 500
I (96763) BT_AV: Audio packet count 600
I (98213) BT_AV: Audio packet count 700
I (99663) BT_AV: Audio packet count 800
```

steenou, 2019-07-12:
@redchenjs @pad52 @jason-mao @igrr the glitching as previously reported is still happening

redchenjs, 2019-07-12:
Yeah, the previous issue should be reopened, and the commit is just an improvement not a fix.

TianHao-Espressif, 2019-08-05:
@steenou , I want to notice that the test demo is BLE GATT SERVER and A2DP coexistence example. The RF is shared, so the A2DP performance might be decrease according to the BLE connection interval and data bandwidth. I want to know what's your requirement for your application? Thanks.

steenou, 2019-08-07:
@TianHao-Espressif even with the A2DP_sink example (literally only A2DP, no BLE) there is still glitching:
[image]

The glitching doesn't seem to have anything to do with the coexistence.

steenou, 2019-08-07:
Still a lot of zero frames (57.5) followed by 2.5 data frames (RLRLR). This glitching happens at the A2DP beginning, and randomly throughout the stream

steenou, 2019-08-08:
The bt_sink example in the adf (esp-adf/examples/player/pipeline_bt_sink/) seems to behave better, but still with inexplicable long gaps of more than a second. These gaps sometimes aren't even accompanied by `BT_APPL: Pkt dropped` message in the log.

steenou, 2019-09-11:
Is there any response to this?

pad52, 2019-09-11:
Yes, I am interested too and available for testing.
I experience the same glitches,
Clock configuration helps but didn't solve the problem.
The problem seems very hardware dependant,
with a new Xiaomi Redmi 7 the glitches are many, always and constant, audibles also with songs and not just with Sinusoids.

pad52, 2019-09-20:
> Yes, I am interested too and available for testing.
> I experience the same glitches,
> Clock configuration helps but didn't solve the problem.
> The problem seems very hardware dependant,
> with a new [...]

Update:
With last esp-idf master (96b96ae244f360c84b71c8c3670e24a1186e9844)
and a2dp_gatts_coex - tested on OSX, Win7, Android, Xiaomi 

- I'm experiencing clicks only when a new connection occours and this happens:
W (30082) BT_APPL: new conn_srvc id:19, app_id:1
I (30082) BT_AV: A2DP audio state: Started
I have some clicks in the beginning - It seems like the DMA stream is starting with the ring buffer empty.
I'm not experiencing clicks anywhere else - thank you very much @TianHao-Espressif and to your team.

- About the Xiaomi - main problem was that I was expecting sound at 44100 like any BT device connected with SBC - Xiaomi NOT - it will choose the 48000... 

According to WireShark analyzer the ESP in fact is offering 44100 and 48000.

May I suggest to modify
`if (a2d->audio_cfg.mcc.type == ESP_A2D_MCT_SBC) {
            int sample_rate = 16000;
            char oct0 = a2d->audio_cfg.mcc.cie.sbc[0];
            if (oct0 & (0x01 << 6)) {
                sample_rate = 32000;
            } else if (oct0 & (0x01 << 5)) {
                sample_rate = 44100;
            } else if (oct0 & (0x01 << 4)) {
                sample_rate = 48000;
            }
            i2s_set_clk(0, sample_rate, 32, 2);`

accordingly?

- I had to modify some lines in I2S connection mode.
FROM:
.communication_format = I2S_COMM_FORMAT_I2S_MSB,
TO:
.communication_format = I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB,

- I had also to turn on the option in Menuconfig: `BT/BLE will first malloc the memory from the PSRAM`

TianHao-Espressif, 2019-09-20:
@pad52 @steenou , I thinks that sometimes it may be due to the ring buffer fo I2S data cache is not enough in WiFi and Bluetooth coexistence mode. Could you enlarge the ring buffer and make  buffered time longer?

pad52, 2019-09-23:
> @pad52 @steenou , I thinks that sometimes it may be due to the ring buffer fo I2S data cache is not enough in WiFi and Bluetooth coexistence mode. Could you enlarge the ring buffer and make buffered time longer?

No I'm not using Coexistance - I commented the BLE part in the example and not using any WiFi.
I try to resolve the glitches at the beginning of a stream by modifying this line in main/bt_app_core.c:
`if (item_size != 0){
    i2s_write(0, data, item_size, &bytes_written, portMAX_DELAY);`
in 
`if (item_size >= 1024){
    i2s_write(0, data, item_size, &bytes_written, portMAX_DELAY);`
Unfortunately is not working with all the devices.
(OSX Yes, Android no)

I would like to notice that A Xiaomi phone would crash all the system in case the DSP is not capable of change the I2S frequency dinamically.

I would suggest to control how much data goes out of the I2S stream and how much data is going inside the Ring Buffer - or it could go in Overflow and crash everything.

steenou, 2019-10-15:
Hey @pad52, I'm gonna test the esp-idf master (96b96ae) soon and let you know it works for us. Have you tried the v4.0 pre-release yet?

In which line did you make this change:
`if (item_size != 0){ i2s_write(0, data, item_size, &bytes_written, portMAX_DELAY)`

pad52, 2019-10-15:
Hey,

My thought is: "Maybe we are getting clicks at the beginning of a stream - just because we start the I2S stream immediately without filling the ring buffer first."

That's why I tried to modify that line in
a2dp_gatts_coex/main/bt_app_core.c

item_size >= 1024 is just an arbitrary value - I didn't test it nor I know if it's good.

I didn't tried the v4.0 pre-release, let me know how it goes pls.

steenou, 2019-10-17:
I tried your `item_size >= 1024` test but unfortunately it still gives me glitching in the beginning of the stream.
The v4.0 pre-release gives the same result.

I feel like at this point we need to look at the A2DP code example of the ADF for the solution. As I've said before, this code example has no glitching of this sort.

rsm123, 2019-10-26:
Issue is prominent in my IPAD than in android. When I hold the antenna location of Redmi phone and IPAD i can hear constant drops. When I take off my hands, audio becomes smoother. I'm using a2dp sink example.

steenou, 2019-11-06:
Hey @pad52 

> Maybe we are getting clicks at the beginning of a stream - just because we start the I2S stream immediately without filling the ring buffer first

I'm not sure if this makes a lot of sense. I think `item_size` will be larger than zero at all times when A2DP is receiving sound. I think the problem is that the data inside of `data` is a interlaced with a lot of zeroes, which sounds like glitches.

I've been doing some research into if that is really the case, or if it's `i2s_write` that just drops data packets. Not sure why that would result in the glitching pattern that's always the same, zeroes, 5 words of data, zeroes again, etc.

Do you feel like debugging this together in a more cooperative way? Not sure how I can give you my e-mail address here while remaining a somewhat private person :laughing:

pad52, 2019-11-06:
Hey @steenou 

Yeah well I really don't love to play with fast ring-buffer - Do you really see this 
> interlaced with a lot of zeroes, which sounds like glitches.
?
I mean - did you save a PCM kind of stream somewhere to analyze? If not - this is what to begin with to exclude Bluetooth SBC problems.

In case the stream have no zeroes - then it's somewhere between the I2S and the DMA.

I have a DSP that handle that data and it's not easy to perfectly sincronize the two device clocks wich causes also constant glitches: remind that - according to the datasheet the ESP32 - the esp it's mean to be used in Master mode - and I'm using it in Slave mode - you?
Maybe a DSP with ASRC could help...

We can also try with a direct I2S->ADC interface instead? Yes you will got Analog audio but with no problems about I2S sync.

You said that the esp-adf framework have no glitches.
Is it true? On what kind of hardware?
I really wish to try to make a board.h for my development board (based on esp32wroom and not wroover :-/ ) and test the esp-adf in my/your environment - this - if you wish - we can do together at sometime next week! (this week I'm really full sorry)

steenou, 2019-11-06:
Yeah I also don't really want to mess with those DMA buffers, no clue what to do there...

Until now I have scoped the output I2S of the ESP32 (we have a WROVER), and gotten the images I've shown before with zeroes followed by data, followed by data again. Small update on that: I wrote a small Python script that turns raw I2S data in a graph, as seen below:
[image]

I have tried to get the data straight inside `void bt_app_a2d_data_cb(const uint8_t *data, uint32_t len)`, where I `memcpy` the data from `data` into an array, and then print that array with `ESP_LOGI`. That's a bit difficult because it makes the ESP32 lose a lot of packets (since the processor is constantly busy printing logs). This way, I have been able to see the raw data before it goes into `i2s_write`, but have not yet been able to correlate that to the actual I2S at the output. Sooo it's kind of hard for me to say without a doubt that there are no zero packets before the data is offloaded into `i2s_write`. Tomorrow I'll show my code changes to print the data and the results.

> We can also try with a direct I2S->ADC interface instead? Yes you will got Analog audio but with no problems about I2S sync.

This is what my company has. We have the ESP32 as an I2S master, connected into a Microchip PIC microprocessor that has its own DMA. The PIC's DMA output is connected to a Texas Instruments TAS amplifier, which is then made audible directly on speaker drivers. The LRCLK and SCLK of I2S are the same on the DMA input and the DMA output side, so there is no issue with clock asyncronicity.

Next week is fine. We could chat on jit.si? Which timezone are you in?
I'm always checking the ESP32 output (PIC DMA input) as well as the TAS input (PIC DMA output), so I'm 100% sure the glitching isn't occurring inside our PIC DMA.

steenou, 2019-11-07:
So this is what I did to the code:
`void bt_app_a2d_data_cb(const uint8_t *data, uint32_t len) {
	size_t bytes_written;
	nZeroes = 0;
	int index = 0;
	
	// Research audio packet
	if (len > 0) {
		// Flush out printbuffer
		memset(printbuffer, 0, sizeof printbuffer);
		// Copy A2DP data into researchbuffer
		memcpy(researchbuffer, data, len);

		for (uint16_t i=0; i<len; i++) {
			// Check how many zeroes in an audio packet
			if (researchbuffer[i]==0) nZeroes++;
			// Put all data in audio packet in printbuffer
			index += sprintf(&printbuffer[index], "%x ", researchbuffer[i]);
		}

		// Print packet info (packet size, amount of zeroes)
		if (nZeroes!=0) ESP_LOGI(BT_AV_TAG, "Packet: %u, Zeroes: %u", len, nZeroes);
		
		// Print packet contents (raw A2DP data)
		ESP_LOGI(BT_AV_TAG, "%s", printbuffer);
	}

	// // Send to I2S
	// i2s_write_expand(1, data, len, 16, 32, &bytes_written, portMAX_DELAY);
	// if (++m_pkt_cnt % 100 == 0) {
	// 	ESP_LOGI(BT_AV_TAG, "Audio packet count %u", m_pkt_cnt);
	// }
}`

It collects the raw A2DP data into `printbuffer` and prints it as a log. It also collects the length of the packet and counts the amount of zeroes in the packet.
I'm playing a 440 Hz signal with the Simple Tone Generator Android app. I've turned I2S writing off, since the sound is terrible, and it probably puts more pressure on the processor. My audio architecture requires 32 bit words, so instead of `i2s_write`, I us `i2s_write_expand` which just sticks a bunch of zeroes at the end of every word.

The result in logs is this (you see a watchdog trigger occurs due to  in timestamp 22564 because BtA2dSinkT is taking too long):
```I (914) Bluetooth: pairing enabled
I (914) Leader: Init complete
E (9564) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (9564) BT_AV: A2DP connection state: Connecting, [94:65:2d:20:56:8b]
I (9614) BT_AV: A2DP audio stream configuration, codec type 0
I (9614) I2S: DMA Malloc info, datalen=blocksize=240, dma_buf_count=6
I (9624) I2S: APLL: Req RATE: 44100, real rate: 43945.238, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11249981.000, SCLK: 1406247.625000, diva: 1, divb: 0
I (9634) BT_AV: Configure audio player 21-15-2-35
I (9634) BT_AV: Audio player configured, sample rate=44100
W (9684) BT_APPL: new conn_srvc id:19, app_id:0
I (9684) BT_AV: A2DP connection state: Connected, [94:65:2d:20:56:8b]
I (9764) BT_AV: AVRC conn_state evt: state 1, [94:65:2d:20:56:8b]
I (9764) BT_AV: AVRC remote features 4b
I (10324) BT_AV: AVRC metadata rsp: attribute id 0x1, The Trial - 2011 Remastered Version
I (10324) BT_AV: AVRC metadata rsp: attribute id 0x2, Pink Floyd
I (10324) BT_AV: AVRC metadata rsp: attribute id 0x4, The Wall (Remastered)
I (10334) BT_AV: AVRC metadata rsp: attribute id 0x20, 
I (17534) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b

W (17534) BT_APPL: new conn_srvc id:19, app_id:1
I (17534) BT_AV: A2DP audio state: Started
I (17724) BT_AV: Packet: 3072, Zeroes: 3072
I (17724) BT_AV: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  [... cut ...] 1 2 91 2 9b 2 9b 2 a3 2 a3 2 aa 2 aa 2 ac 2 ac 2 ad 2 ad 2 ab 2 ab 2 a7 2 a7 2 9e 2 9e 2 93 2 93 2 88 2 88 2 79 2 79 2 66 2 66 2 52 2 52 2 3b 2 3b 2 23 2 23 2 8 2 8 2 ea 1 ea 1 cc 1 cc 1 aa 1 aa 1 88 1 88 1 64 1 64 1 3f 1 3f 1 18 1 18 1 f0 0 f0 0 c8 0 c8 0 9f 0 9f 0 75 0 75 0 49 0 49 0 1d 0 1d 0 f4 ff f4 ff ca ff ca ff 9f ff 9f ff 75 ff 75 ff 4a ff 4a ff 21 ff 21 ff f9 fe f9 fe d2 fe d2 fe ac fe ac fe 87 fe 87 fe 63 fe 63 fe 41 fe 41 fe 23 fe 23 fe 
I (23454) BT_AV: Packet: 3584, Zeroes: 222
```

pad52, 2019-11-07:
Ok this seem a interesting test - even if debugging on screen is really slow and could cause false-positive - Could be a better test trying to write the stream down on a SDCARD and read it later.

So what I see it's for example that the first I2S packet is empty and the second is almost empty - this is something that could cause glitches.
If it's true that the ESP-ADF framework works better - it coud be interesting to follow the stream and understand where the espressif team filter the empty stream - did you manage to work with the ADF framework on your custom board?

>This is what my company has. We have the ESP32 as an I2S master, connected into a Microchip PIC microprocessor that has its own DMA

yes So - are you sure that the Pic receive the clock correctly? It takes some sort of master clock from ESP32?

We can also test what happens in case we go with the direct DAC from the ESP itself - going out in analog audio.
I mean - it's wrong and ugly - but it could be another test - easy to check with a scope.

It's ok to chat next week - my timezone is GMT+2 - let's write us something on monday afternoon?

steenou, 2019-11-08:
>did you manage to work with the ADF framework on your custom board?
Yes I have managed to make that work at some point. I will set that up again in the next week.
> So - are you sure that the Pic receive the clock correctly? It takes some sort of master clock from ESP32?
Yes it uses the SCLK and LRCLK generated by the ESP32 to sample the data on the input side and that same clock to output the data on the output side.
> We can also test what happens in case we go with the direct DAC from the ESP itself - going out in analog audio.
That's true, but I'm not sure if I can make that work in my setup. Don't we need some sort of PWM output filter for that?

I'm in Amsterdam, so GMT+1. Let's write Monday afternoon. If you add me on linkedin (name is right there on my account), I can then give you my email address there through private chat?

steenou, 2019-11-08:
By the way, we managed to offload the joining of the A2DP data onto the other core, and are only printing after a while - and are halting I2S writing while writing. So hypothetically, the audio playback should not be hampered too much. But who can tell 100% certainly?
We have also thought about writing the raw A2DP data into a file whilst playing, but this was the quickest way of doing it. We've got some datasets - I2S and raw A2DP that we're looking into, to see if the glitches occur in the A2DP service, or in the I2S service. 

Will install the ADF example again early next week.

Wth-Esp, 2020-06-18:
Hi all 

What's the situation of this issue now? From my inspect, it's not an issue of A2DP but I2S.

Wth-Esp, 2020-07-08:
Hi

Maybe you can use this IDF v3.3 and apply this patch. Have a try, please.
[bt_x20_kadun.zip](https://github.com/espressif/esp-idf/files/4889479/bt_x20_kadun.zip)

Thanks
Weitianhua

Alvin1Zhang, 2020-10-29:
Thanks for reporting, will close due to short of feedback, feel free to reopen if the issue still happens. Thanks.

closed 2020-10-29 by Alvin1Zhang