espressif/esp-idf #3777: Windows installer v2.0 (CMake Preview) - option to uncheck downloading of packages

RonAmihai, 2019-07-13:
In the previous version of the Windows installer (v1.2) - there was an option to uncheck downloading / installing of packages. 
This was handy for offline setup of ESP-IDF environment (of course it would require to install the Python required packages, CMake and other stuff manually - which is possible without internet connection). 

Since v2.0 there is no option to uncheck download of packages like Python packages - and that fails installation in offline environments because the installer can't access the internet for downloading packages.

Please consider to bring back "custom installation" to the installer in order to enable selective installation of ESP-IDF.

igrr, 2019-07-13:
@RonAmihai okay, will add an option to skip installing Python packages, or bundle them with the installer so that it can run offline.

Yuvaraj4041, 2019-07-15:
@RonAmihai In your PC go to advance internet options, then in settings Enable Use SSL 2.0, Use TLS 1.1, Use TLS 1.2. By default this are all not selected. 

After enabling Save it, now your installer will get access to download packages from internet.

georgik, 2021-03-11:
@RonAmihai We have an update regarding the installer for Windows.

We introduced new types of installer Online and Offline version. Offline versions contains all dependencies and does not require access to the internet. The Online version is minimalistic and retrieves all data from the internet which is then stored to local cache.

Please try our new installers:
- Online installer: https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
- Offline installer - contains v4.2 and v4.1.1 (810 MB): https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.5.exe

We also have a new beta version which contains integration with Eclipse:
- Online installer: https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.6.exe
- Offline installer - contains v4.2, v4.1.1, Eclipse (1.2 GB): https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.6.exe

The source code and build artifacts are also available on GitHub: https://github.com/espressif/idf-installer

closed 2021-04-16 by tom-borcin