espressif/esp-idf #3779: Bootloader can't be disabled in 3.2.2

Stasikiii, 2019-07-14:
Hi.

I try to disable bootloader log in **bootloader config > bootloader log verbosity > No output**.
In **sdkconfig** i can see section

> CONFIG_LOG_BOOTLOADER_LEVEL_NONE=y

But after flashing i still see detailed log

> .rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)↵
> .configsip: 0, SPIWP:0xee↵
> .clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00↵
> .mode:DIO, clock div:2↵
> .load:0x3fff0018,len:4↵
> .load:0x3fff001c,len:6200↵
> .load:0x40078000,len:10180↵
> .load:0x40080400,len:6660↵
> .entry 0x40080764↵
> ..[0;32mI (29) boot: ESP-IDF v3.2.2-dirty 2nd stage bootloader.[0m↵
> compile time 21:56:35.[0m↵
> G early entropy source....[0m↵
> 32mI (34) boot: SPI Speed      : 40MHz.[0m↵
> ..[0;32mI (39) boot: SPI Mode       : DIO.[0m↵
> ..[0;32mI (43) boot: SPI Flash Size : 4MB.[0m↵
> ..[0;32mI (47) boot: Partition Table:.[0m↵
> ..[0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length.[0m↵
> ..[0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00005000.[0m↵
> ..[0;32mI (65) boot:  1 otadata          OTA data         01 00 0000e000 00002000.[0m↵
> ..[0;32mI (72) boot:  2 app0             OTA app          00 10 00010000 00140000.[0m↵
> ..[0;32mI (80) boot:  3 app1             OTA app          00 11 00150000 00140000.[0m↵
> ..[0;32mI (87) boot:  4 spiffs           Unknown data     01 82 00290000 00170000.[0m↵
> ..[0;32mI (95) boot: End of partition table.[0m↵
> ..[0;32mI (99) boot: No factory image, trying OTA 0.[0m↵
> ..[0;32mI (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x15cdc ( 89308) map.[0m↵
> ... etc

Help! :)

Stasikiii, 2019-07-15:
UPD
Ok, to disabled it, i change option **Component config > Esp32 specific >Uart for console output > none**
But i'm not sure, maby log not showed but anyway take time before app runing.

projectgus, 2019-08-01:
Hi @Stasikiii ,

I'm not able to reproduce this with the IDF v3.2.2 release.

If I build, for example, the "Hello World" example project with a totally fresh sdkconfig - the only modification being to set Bootloader Log Level to None in menuconfig, then I get this outout:

```
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:816
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5852
entry 0x4008069c
I (94) cpu_start: Pro cpu up.
I (94) cpu_start: Starting app cpu, entry point is 0x40080e94
0x40080e94: call_start_cpu1 at /home/gus/esp/32/idf/components/esp32/cpu_start.c:246

I (0) cpu_start: App cpu up.
I (97) heap_init: Initializing. RAM available for dynamic allocation:
I (104) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (110) heap_init: At 3FFB2EE8 len 0002D118 (180 KiB): DRAM
I (116) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (123) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (129) heap_init: At 40089584 len 00016A7C (90 KiB): IRAM
I (135) cpu_start: Pro cpu start user code
I (153) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
Restarting in 10 seconds...
```

(The first 11 lines are printed by the ROM bootloader and can be suppressed with GPIO15. The remaining lines are printed by the app.)

Relevant section in sdkconfig:

```
#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=y
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=0
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V=
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=
CONFIG_BOOTLOADER_APP_TEST=
CONFIG_BOOTLOADER_WDT_ENABLE=y
CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE=
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
```

* Note that the line which is most used in Bootloader is `CONFIG_LOG_BOOTLOADER_LEVEL=0`. This line is generated based on the other values, by the menuconfig internal system.
* Is it possible that the bootloader was not completely rebuilt or reflashed with the "None" log level option?
* I noticed your IDF version is "v3.2.2-dirty". Is it possible some local modification in IDF is changing the behaviour?

Alvin1Zhang, 2020-01-19:
@Stasikiii  Thanks for reporting, would you please help share if any updates for the issue? Does your issue still exist? Thanks.