espressif/esp-idf #3780: Multiple compile time redefinition warnings (LwIP <=> nghttp)

no1seman, 2019-07-15:
## Environment

- Development Kit:  ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used: ESP32-WROVER-B
- IDF version: v4.0-dev-1275-gfdab15dc7
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System: Linux
- Power Supply:  USB|external 5V&Battery

## Problem Description
There are some redifinition warnings during build (see below).

### Expected Behavior
No warnings
### Actual Behavior
Warnings present
### Steps to repropduce

### Code to reproduce this issue

## Debug Logs

In file included from ../main/networks.c:15:
/home/user/esp/esp-mdf/esp-idf/components/nghttp/port/include/config.h:31: warning: "htons" redefined
 #define htons(x) NGHTTP_PLATFORM_HTONS(x)
 
In file included from /home/user/esp/esp-mdf/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:41,
                 from /home/user/esp/esp-mdf/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:24,
                 from /home/user/esp/esp-mdf/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from /home/user/esp/esp-mdf/esp-idf/components/esp_event/include/esp_event_loop.h:1,
                 from ../main/networks.c:4:
/home/user/esp/esp-mdf/esp-idf/components/lwip/lwip/src/include/lwip/def.h:119: note: this is the location of the previous definition
 #define htons(x) lwip_htons(x)
 
In file included from ../main/networks.c:15:
/home/user/esp/esp-mdf/esp-idf/components/nghttp/port/include/config.h:32: warning: "ntohs" redefined
 #define ntohs(x) NGHTTP_PLATFORM_HTONS(x)
 
In file included from /home/user/esp/esp-mdf/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:41,
                 from /home/user/esp/esp-mdf/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:24,
                 from /home/user/esp/esp-mdf/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from /home/user/esp/esp-mdf/esp-idf/components/esp_event/include/esp_event_loop.h:1,
                 from ../main/networks.c:4:
/home/user/esp/esp-mdf/esp-idf/components/lwip/lwip/src/include/lwip/def.h:120: note: this is the location of the previous definition
 #define ntohs(x) lwip_ntohs(x)
 
In file included from ../main/networks.c:15:
/home/user/esp/esp-mdf/esp-idf/components/nghttp/port/include/config.h:33: warning: "htonl" redefined
 #define htonl(x) NGHTTP_PLATFORM_HTONL(x)
 
In file included from /home/user/esp/esp-mdf/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:41,
                 from /home/user/esp/esp-mdf/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:24,
                 from /home/user/esp/esp-mdf/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from /home/user/esp/esp-mdf/esp-idf/components/esp_event/include/esp_event_loop.h:1,
                 from ../main/networks.c:4:
/home/user/esp/esp-mdf/esp-idf/components/lwip/lwip/src/include/lwip/def.h:121: note: this is the location of the previous definition
 #define htonl(x) lwip_htonl(x)
 
In file included from ../main/networks.c:15:
/home/user/esp/esp-mdf/esp-idf/components/nghttp/port/include/config.h:34: warning: "ntohl" redefined
 #define ntohl(x) NGHTTP_PLATFORM_HTONL(x)
 
In file included from /home/user/esp/esp-mdf/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:41,
                 from /home/user/esp/esp-mdf/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:24,
                 from /home/user/esp/esp-mdf/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from /home/user/esp/esp-mdf/esp-idf/components/esp_event/include/esp_event_loop.h:1,
                 from ../main/networks.c:4:
/home/user/esp/esp-mdf/esp-idf/components/lwip/lwip/src/include/lwip/def.h:122: note: this is the location of the previous definition
 #define ntohl(x) lwip_ntohl(x)

no1seman, 2019-08-24:
Found mistake in my code. Closing issue

closed 2019-08-24 by no1seman

trygvis, 2019-09-04:
I have exactly the same issue, how did you solve it?

no1seman, 2019-09-04:
@trygvis,

I had config.h file in my main project dir, but it was conflicted with $IDF_PATH/components/nghttp/port/include/config.h. I renamed it and warnings was gone.

Good luck.

trygvis, 2019-09-04:
Thanks! I for some reason had an explicit `#include <config.h>` which I didn't need, removing that line fixed it for me.