espressif/esp-idf #3782: Crash with latest IDF version

ammaree, 2019-07-15:
## Environment
- Development Kit:      [custom]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-1275-gfdab15dc7
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- Power Supply:         [external 3.3V]

## Problem Description
Very simple simple application, single task running web server and saving values supplied to NVS blob then rebooting. Nothing changed from previous working version, just compiled with later master (as opposed to that of 1 week ago which worked perfectly)

### Expected Behavior
program continues to work as normal

### Actual Behavior
Program crashes with the listed crash log

### Steps to reproduce
No idea yet, just recompiled application

### Code to reproduce this issue
No idea yet, just recompiled application

## Debug Logs
```
abort() was called at PC 0x400db563 on core 1
0x400db563: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

ELF file SHA256: b0722a27d720a0b6e8373e94bf644ed4fdb5247d6d7c043ceb48da8f11d5514c

Backtrace: 0x4008d091:0x3ffce4f0 0x4008d4b5:0x3ffce510 0x400db563:0x3ffce530 0x4010baca:0x3ffce560 0x4010bb38:0x3ffce590 0x4010ba6d:0x3ffce5c0 0x400fb37f:0x3ffce5e0 0x400f85d2:0x3ffce600 0x400f8609:0x3ffce620 0x400fa24a:0x3ffce650 0x400ea9e0:0x3ffce670 0x4000bd4b:0x3ffce690 |<-CORRUPTED

0x4008d091: invoke_abort at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/esp32/panic.c:719
0x4008d4b5: abort at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/esp32/panic.c:719
0x400db563: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
0x4010baca: netconn_drain at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/api/api_msg.c:1293
0x4010bb38: netconn_drain at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/api/api_msg.c:1293
0x4010ba6d: netconn_free at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/api/api_msg.c:1293
0x400fb37f: netconn_delete at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/api/api_lib.c:908
0x400f85d2: free_socket_free_elements at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/api/sockets.c:2757
0x400f8609: free_socket at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/api/sockets.c:2757
0x400fa24a: lwip_close at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/api/sockets.c:2757
0x400ea9e0: esp_vfs_close at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/vfs/vfs.c:1024
```

## Other items if possible

igrr, 2019-07-16:
Probably unrelated to the issue, but please note that IDF commit fdab15dc7 requires different toolchain version than the one you have mentioned (1.22.0-80-g6c4433a5 5.2.0). There should be a compilation warning asking to update the compiler. Consult https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#id1 if you are using CMake or https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/windows-setup.html if you are using GNU Make. To download just the toolchain, you can use the link listed here: https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/windows-setup-scratch.html#alternative-setup-just-download-a-toolchain

ammaree, 2019-07-16:
Have had the compile warning for a while (last couple of weeks) but ignored it since operation remained consistent (including same problems) so practiced maxim "Don't mess with it if it ain't broken..."

Will update the compiler now and provide update.

ammaree, 2019-07-16:
@igrr 

Have update the toolchain, fixed some EXTRA_CPPFLAGS to remove warnings and recompiled without the warnings.

Problem, as predicted by yourself, the crash details remain exactly the same.

Any suggestions?

igrr, 2019-07-16:
This is likely related to the update of LwIP stack from 2.0.3 to 2.1.2. I would suggest trying to isolate the issue you are seeing, so that we could compile your code and reproduce the issue.

ammaree, 2020-07-01:
The problem suddenly disappeared so I am closing this for now.

closed 2020-07-01 by ammaree