espressif/esp-idf #3785: ASSERT crash related to NVS when performing BLE Bonding

0neblock, 2019-07-16:
## Brief

ASSERT in NVS library related to saving BLE Bonds
`assertion "end <= ENTRY_COUNT" failed: file "components/nvs_flash/src/nvs_page.cpp", line 437, function: esp_err_t nvs::Page::copyItems(nvs::Page&)`

## Environment

Key | Value
--- | ---
 Development Kit |  Custom Board
 Module or chip used | ESP32-WROOM-32U
 IDF version | dca5ed2263134024f13f9e58f4964ad1a409834c - tracking release/v3.3
 Build System | Make
 Compiler version | crosstool-ng-1.22.0-80-g6c4433a
 Operating System |   macOS
 Power Supply |   external 3.3V

## Problem Description

I have an application acting as a BLE Master to connect to a peripheral.
I am encrypting the connection after connecting, using `esp_ble_set_encryption(remote_bda, ESP_BLE_SEC_ENCRYPT_MITM);`

I am using OOB keys to authenticate.

Most of the time, there is no issue doing this and the connection succeeds and I can continue to read characteristics and write data no problem.

However, on occasion, Just after receiving the `ESP_GAP_BLE_OOB_REQ_EVT` GAP event, usually after I send the `esp_ble_oob_req_reply` reply, the nvs_flash component which is used by esp-idf to save the bonding data, has an assertion error and crashes.

I have tried multiple workarounds, such as clearing flash or making my nvs partition larger, none of which seem to help.

This happens no matter how many devices bonding information I already have stored.

The fact that it works sometimes and not other times leads me to believe there is a race condition/corruption going on in the NVS portion of it.

Please let me know what further data I can provide.

## Debug Log

```
assertion "end <= ENTRY_COUNT" failed: file "components/nvs_flash/src/nvs_page.cpp", line 437, function: esp_err_t nvs::Page::copyItems(nvs::Page&)
abort() was called at PC 0x400d3f27 on core 0
0x400d3f27: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

ELF file SHA256: 5b920c5f83d0cbf94e834415d4d41772dab53bf304a0ab01f369a4df6592b5fb

Backtrace: 0x4008cdcf:0x3ffeee10 0x4008d075:0x3ffeee30 0x400d3f27:0x3ffeee50 0x40116b8b:0x3ffeee80 0x4011768a:0x3ffeeec0 0x40115ca7:0x3ffeeee0 0x40116139:0x3ffeef60 0x4011598e:0x3ffeefd0 0x40145341:0x3ffef010 0x4012729e:0x3ffef060 0x40126629:0x3ffef080 0x4012684a:0x3ffef0b0 0x401458ca:0x3ffef0f0 0x40127412:0x3ffef1b0
0x4008cdcf: invoke_abort at components/esp32/panic.c:715

0x4008d075: abort at components/esp32/panic.c:715

0x400d3f27: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x40116b8b: nvs::Page::copyItems(nvs::Page&) at components/nvs_flash/src/nvs_page.cpp:327

0x4011768a: nvs::PageManager::requestNewPage() at components/nvs_flash/src/nvs_pagemanager.cpp:236

0x40115ca7: nvs::Storage::writeMultiPageBlob(unsigned char, char const*, void const*, unsigned int, nvs::VerOffset) at components/nvs_flash/src/nvs_storage.cpp:31

0x40116139: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at components/nvs_flash/src/nvs_storage.cpp:31

0x4011598e: nvs_set_blob at components/nvs_flash/src/nvs_api.cpp:555

0x40145341: config_save at components/bt/bluedroid/osi/config.c:436

0x4012729e: btc_config_flush at components/bt/bluedroid/btc/core/btc_config.c:329

0x40126629: _btc_storage_save at components/bt/bluedroid/btc/core/btc_ble_storage.c:602

0x4012684a: _btc_storage_add_ble_bonding_key at components/bt/bluedroid/btc/core/btc_ble_storage.c:602
 (inlined by) btc_storage_add_ble_bonding_key at components/bt/bluedroid/btc/core/btc_ble_storage.c:144
[... 210 lines cut ...]
CONFIG_OSI_TRACE_LEVEL_ERROR=
CONFIG_OSI_TRACE_LEVEL_WARNING=y
CONFIG_OSI_TRACE_LEVEL_API=
CONFIG_OSI_TRACE_LEVEL_EVENT=
CONFIG_OSI_TRACE_LEVEL_DEBUG=
CONFIG_OSI_TRACE_LEVEL_VERBOSE=
CONFIG_OSI_INITIAL_TRACE_LEVEL=2
CONFIG_BLUFI_TRACE_LEVEL_NONE=
CONFIG_BLUFI_TRACE_LEVEL_ERROR=
CONFIG_BLUFI_TRACE_LEVEL_WARNING=y
CONFIG_BLUFI_TRACE_LEVEL_API=
CONFIG_BLUFI_TRACE_LEVEL_EVENT=
CONFIG_BLUFI_TRACE_LEVEL_DEBUG=
CONFIG_BLUFI_TRACE_LEVEL_VERBOSE=
CONFIG_BLUFI_INITIAL_TRACE_LEVEL=2
CONFIG_BT_ACL_CONNECTIONS=4
CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST=
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=
CONFIG_BLE_HOST_QUEUE_CONGESTION_CHECK=
CONFIG_SMP_ENABLE=y
CONFIG_BLE_ACTIVE_SCAN_REPORT_ADV_SCAN_RSP_INDIVIDUALLY=
CONFIG_BLE_ESTABLISH_LINK_CONNECTION_TIMEOUT=5
CONFIG_BT_RESERVE_DRAM=0xdb5c

```

0neblock, 2019-07-16:
I Have Just had this same crash, but during the Initialisation stage of the BLE Library, So I no longer believe it is related to the bonding functions.

The Crash happened during execution of `init` in the following sample code:

```
#define CLEAR_BONDS 0
static esp_err_t initialize_ble_hardware(){
    esp_err_t ret;
    
    ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        nvs_flash_erase();
        ret = nvs_flash_init();
    }
#ifdef CLEAR_BONDS
#if CLEAR_BONDS == 1
    nvs_flash_erase(); 
#endif
#endif
    if(ret != ESP_OK){
        ESP_LOGE(TAG, "%s nvs init failed, error code = %x\n", __func__, ret);
        return ret;
    }

    ret = esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);
    if(ret != ESP_OK){
        ESP_LOGE(TAG, "Failed to release Classic BT Resources,: %d", ret);
        return ret;
    }

    if(!btStart()){
        ESP_LOGE(TAG, "Failed to start BLE");
        return ESP_ERR_INVALID_STATE;
    }
    
    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGE(TAG, "%s init bluetooth failed, error code = %x\n", __func__, ret);
        return ret;
    }

    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(TAG, "%s enable bluetooth failed, error code = %x\n", __func__, ret);
        return ret;
    }

    return ESP_OK;
}

static esp_err_t register_ble_callbacks(){
    esp_err_t ret = esp_ble_gap_register_callback(esp_gap_cb);
    if (ret){
        ESP_LOGE(TAG, "%s gap register failed, error code = %x\n", __func__, ret);
        return ret;
    }

    ret = esp_ble_gattc_register_callback(esp_gattc_cb);
    if(ret){
        ESP_LOGE(TAG, "%s gattc register failed, error code = %x\n", __func__, ret);
        return ret;
    }

    // register GATTC app profiles
    for(int i = 0; i < TOTAL_GATTC_PROFILES; i++){
        ret = esp_ble_gattc_app_register(i);
        if (ret){
            ESP_LOGE(TAG, "%s gattc app %d register failed, error code = %x\n", __func__, i, ret);
        }
    }
[... 65 lines cut ...]
0x40116c03: nvs::Page::copyItems(nvs::Page&) at components/nvs_flash/src/nvs_page.cpp:327

0x40117702: nvs::PageManager::requestNewPage() at components/nvs_flash/src/nvs_pagemanager.cpp:236

0x40115d1f: nvs::Storage::writeMultiPageBlob(unsigned char, char const*, void const*, unsigned int, nvs::VerOffset) at components/nvs_flash/src/nvs_storage.cpp:31

0x401161b1: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at components/nvs_flash/src/nvs_storage.cpp:31

0x40115a06: intrusive_list_node<HandleEntry>::intrusive_list_node() at components/nvs_flash/src/nvs_api.cpp:555
 (inlined by) HandleEntry::HandleEntry() at components/nvs_flash/src/nvs_api.cpp:48
 (inlined by) nvs_set_blob at components/nvs_flash/src/nvs_api.cpp:426

0x401453b9: config_save at components/bt/bluedroid/osi/config.c:435

0x40127316: btc_config_flush at components/bt/bluedroid/btc/core/btc_config.c:329

0x401266a1: _btc_storage_save at components/bt/bluedroid/btc/core/btc_ble_storage.c:602

0x401269e1: _btc_storage_add_ble_local_key at components/bt/bluedroid/btc/core/btc_ble_storage.c:602
 (inlined by) btc_storage_add_ble_local_key at components/bt/bluedroid/btc/core/btc_ble_storage.c:308

0x40145b34: btc_dm_sec_cb_handler at components/bt/bluedroid/btc/core/btc_dm.c:784

0x4012748a: btc_task at components/bt/bluedroid/btc/core/btc_task.c:107
```

igrr, 2019-07-16:
Can you please dump the contents of NVS partition and attach it to this issue? `parttool.py` can be used to read a partition.

https://github.com/espressif/esp-idf/blob/fdab15dc76a3464ed10ce03e9b9cf1a5cd2aa0d1/examples/storage/parttool/parttool_example.sh#L54

0neblock, 2019-07-18:
Hi @igrr Thanks for the reply.

I haven't received a crash for a couple of days, I have a feeling it might be related to these lines:

```
ret = nvs_flash_init();
if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
    nvs_flash_erase();
    ret = nvs_flash_init();
}

#ifdef CLEAR_BONDS
#if CLEAR_BONDS == 1
    nvs_flash_erase(); 
#endif
#endif
```

I had `#define CLEAR_BONDS = 1`, which caused `nvs_flash_erase();` to be after `nvs_flash_init();`
Could this have been the cause of the ASSERT?

igrr, 2019-07-18:
Yes, calling erase after init would most definitely be an issue.

0neblock, 2019-08-05:
Thanks, this seems to have solved it

closed 2019-08-05 by 0neblock