espressif/esp-idf #3786: how can i run the demo of ble_mesh_fast_provision

isDerek, 2019-07-16:
when i run the demo ( i do not modify anything), the client log tell me :
```
BLE_MESH: Device in queue not support PB-GATT
```
loop the log, and try link PB-ADV again and again.

WCCWCC, 2019-07-17:
Hi isDerek,
    1. We have detailed documentation for this demo (in the engineering path) and we have tested it many times.

    2.  You also check the selection:  Componet config--》Bluetooth--》Bluetooth controller--》BLE full scan feature supported.
    3.   You also check the selection:    Component config--"Bluetooth --" BLuetooth controller--"Scan Duplicate Type (Scan Duplicate By Device Address And Advertising Date)

Alvin1Zhang, 2019-07-17:
@isDerek Thanks for reporting the issue. For such question, we recommend you submit a post on our forum at [forum](https://esp32.com), will close this issue now. Thanks.

closed 2019-07-17 by Alvin1Zhang