espressif/esp-idf #3787: rename XX macro in http_parser.h

freedaun, 2019-07-16:
##  Environment

IDF: v4.0-dev-1275-gfdab15dc7, Windows 10

## Problem Description

I got a conflict with http_parser.h, which uses the same macro name:

    #define XX() 

### Code to reproduce this issue

#define XX(x)      something

## Proposed solution

Never use non-descript unqualified macro-names in libs, as these identifier belong to the user and end up breaking their code. So either qualify the define and rename it HTTP_PARSER_XX(), or use the standard X-macro name convention ENTRY().

igrr, 2019-07-16:
@freedaun I would suggest submitting this upstream (https://github.com/nodejs/http-parser). We can then update our copy of http_parser, and also move it into a submodule for easier maintenance in the future.

freedaun, 2019-07-16:
Well, I got the reference from your github, the http_server. But now I see there is an official supported "esp_http_server.h", so I'd better use that. But I'll follow your suggestion anyway. 

I really need to implement websockets. I found https://github.com/Molorius/esp32-websocket and am working backwards from that, substituting the http_server part. Maybe the Arduino stuff would be better but there's a knowledge gap with the arduino world. 

It would seem the ESP32 is REALLY in need of a native websocket implementation (example), as it is relatively simple and the preferred way to connect to web and mobile apps. 

Using native Wifi access has been a complete disaster because of different Android versions (and even manufacturers/models!). 

Thanks.

freedaun, 2019-07-16:
> @freedaun I would suggest submitting this upstream (https://github.com/nodejs/http-parser). We can then update our copy of http_parser, and also move it into a submodule for easier maintenance in the future.

See: https://github.com/nodejs/http-parser/issues/487

mahavirj, 2020-04-20:
@freedaun It seems like `http-parser` issue you created upstream is not yet fixed. But we have `websocket` support in `esp_http_server` now, [config_option](https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/Kconfig#L42) and [example](https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/ws_echo_server/README.md). I will go ahead and close this issue but please free to reopen if this does not satisfy your requirement or you are looking for something else.

closed 2020-04-20 by mahavirj