espressif/esp-idf #3790: Toolchain installer

tatsutaigu, 2019-07-17:
Hi,

I try latest toolchain installer.
I got trouble.
I use proxy to connect to internet.
So, My PC cannot connect to github.

Please add proxy setting to toolchain installer.

msys32 can set proxy and connect to github via proxy.

[https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html)

igrr, 2019-07-17:
Hi @tatsutaigu, the installer uses the proxy settings configured in Windows. Please make sure that your proxy is configured (in Windows 10: Settings -> Network & Internet -> Proxy).
You can also check if these are valid by inspecting the following registry key: `HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Internet Settings`.

tatsutaigu, 2019-07-18:
Hi @igrr,

git.exe must need HTTP_PROXY & HTTPS_PROXY.

So, I typed below commands on cmd.exe before installing toolchain.
`setx HTTP_PROXY “http://yourhost:yourport”`
`setx HTTPS_PROXY “http://yourhost:yourport”`

closed 2019-07-18 by tatsutaigu