espressif/esp-idf #3793: Running export.sh results in errors

tomasmcguinness, 2019-07-17:
When I try to execute the export.sh script on macOS, I get the following error messages:

readlink: illegal option -- f
usage: readlink [-n] [file ...]
usage: dirname path
Adding ESP-IDF tools to PATH...
-bash: /tools/idf_tools.py: No such file or directory

igrr, 2019-07-17:
@tomasmcguinness the fix for this issue is in review at the moment and will be merged soon. For now you can work around the issue by exporting IDF_PATH manually, before running the script.

tomasmcguinness, 2019-07-18:
Thanks for the quick rely, @igrr. I ended up installing greadlink and updating the script.

sjhodgen, 2019-07-22:
I changed the command to `stat`, per the man page on `readlink`, which is really the man page for `stat`.

stephenlstrange2, 2019-07-22:
run
reload(sys)
sys.setdefaultencoding('utf8')

igrr, 2019-07-25:
Fixed in https://github.com/espressif/esp-idf/commit/54b16b7f5a5778c195e26369a3aa8bcb67f52b4b

closed 2019-07-25 by igrr

vmgarcia, 2020-02-15:
@igrr I'm getting this same error again

idf_export_main:33: no such file or directory: $HOME/esp/esp_idf/tools/idf_tools.py

Viranchee, 2020-03-29:
Same issue for me 
`idf_export_main:33: no such file or directory: $HOME/esp/esp_idf/tools/idf_tools.py`

igrr, 2020-03-30:
@Viranchee the issue which @vmgarcia had was due to a typo in the path name (`esp-idf` not `esp_idf`), see the discussion in https://github.com/espressif/esp-idf/issues/4770. Please confirm whether you are setting `IDF_PATH` environment variable correctly.

Viranchee, 2020-03-31:
> @Viranchee the issue which @vmgarcia had was due to a typo in the path name (`esp-idf` not `esp_idf`), see the discussion in #4770. Please confirm whether you are setting `IDF_PATH` environment variable correctly.

Hi @igrr I had the `IDF_PATH` set on my environment correctly. I am setting up esp-idf again.
`esptool` has an issue, after `pip install esptool` I can not call it from Command line directly

Viranchee, 2020-04-01:
@igrr So here are the detailed steps 

> ./install.sh
Installs perfectly, shows I am ready to run `. ./export.sh`
Deprecation warning about Python2.7

> . ./export.sh
Error. Need IDF_PATH

> IDF_PATH="url/to/path"

> . ./export.sh
Adding ESP-IDF tools to PATH...
idf_export_main:33: no such file or directory: /Users/virancheel/.espressif/tools/idf_tools.py

Viranchee, 2020-04-01:
it complained for 2 files, `idf_tools.py` and `tools.json`
So copied them from regular esp-idf folder to where the installation is in

Then got these errors:

> Adding ESP-IDF tools to PATH...
fatal: not a git repository (or any of the parent directories): .git
WARNING: Git describe was unsuccessul: Command '['git', 'describe', '--tags']' returned non-zero exit status 128.
WARNING: Is it not possible to determine the IDF version: [Errno 2] No such file or directory: '/Users/virancheel/.espressif/components/esp_common/include/esp_idf_version.h'
ERROR: IDF version cannot be determined

igrr, 2020-04-01:
Regarding the last error, how are you obtaining IDF copy? Not by `git clone`?

Viranchee, 2020-04-01:
> Regarding the last error, how are you obtaining IDF copy? Not by `git clone`?

By `git clone`
It gives the above errors on Mac, but installs perfectly and `. ./export.sh` works perfectly on raspberry pi after setting `python = python3`. Unfortunately on my mac, default `python = python2`

Viranchee, 2020-04-01:
I checked with my RaspberryPi, it has no such file `~/.espressif/tools/idf_tools.py`
And esp-idf compiles perfectly

While on Macos, 

On running
`. ./export.sh`

>Adding ESP-IDF tools to PATH...
idf_export_main:33: no such file or directory: /Users/virancheel/.espressif//tools/idf_tools.py

igrr, 2020-04-01:
Could you please check what are the values of IDF_PATH and IDF_TOOLS_PATH in the environment?

Viranchee, 2020-04-01:
Correct `IDF_PATH`
No `IDF_TOOLS_PATH` in environment

Viranchee, 2020-04-01:
I researched, it seems like this issue #4770
I'd like to move back to version 4 or whatever which compiles and runs
@igrr Can you state me the steps?

igrr, 2020-04-01:
Sorry, what is the exact value of IDF_PATH?

This path in the error message:

> '/Users/virancheel/.espressif/components/esp_common/include/esp_idf_version.h

makes me think that IDF_PATH is somehow set to /Users/virancheel/.espressif instead of the location where ESP-IDF is cloned.

Suggest unsetting IDF_PATH, deleting $HOME/.espressif, cloning ESP-IDF into $HOME/esp/esp-idf (just to keep it close to the Getting Started guides), then run install.sh and source export.sh. If you meet an error at any step, please post the full output you receive.

Viranchee, 2020-04-02:
After running `./install.sh`
> All done! You can now run:
  . ./export.sh

```
➜  esp-idf git:(master) ✗ echo $IDF_PATH 
/Users/virancheel/Desktop/code/esp-idf
➜  esp-idf git:(master) ✗ echo $IDF_TOOLS_PATH 
/Users/virancheel/.espressif
➜  esp-idf git:(master) ✗ . ./export.sh
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Traceback (most recent call last):
  File "/Users/virancheel/Desktop/code/esp-idf/tools/check_python_dependencies.py", line 42, in <module>
    default_requirements_path = os.path.join(idf_path, 'requirements.txt')
  File "/usr/local/opt/python/bin/../Frameworks/Python.framework/Versions/3.7/lib/python3.7/posixpath.py", line 80, in join
    a = os.fspath(a)
TypeError: expected str, bytes or os.PathLike object, not NoneType
➜  esp-idf git:(master) ✗ 
```

igrr, 2020-04-02:
Thanks for the log! Could you please add some lines of debugging as shown in the diff below, and see what the output becomes?

```diff
diff --git a/export.sh b/export.sh
index 0408a76bb..f4dddab10 100644
--- a/export.sh
+++ b/export.sh
@@ -21,6 +21,8 @@ function idf_export_main() {
         return 1
     fi
 
+    echo "idf_export_main: IDF_PATH at start: ${IDF_PATH}"
+
     if [[ -z "${IDF_PATH}" ]]
     then
         # If using bash, try to guess IDF_PATH from script location
@@ -32,6 +34,7 @@ function idf_export_main() {
                 script_name="$(readlink -f $BASH_SOURCE)"
                 script_dir="$(dirname $script_name)"
             fi
+            echo "idf_export_main: Setting IDF_PATH to ${IDF_PATH}"
             export IDF_PATH="${script_dir}"
         else
             echo "IDF_PATH must be set before sourcing this script"
@@ -49,6 +52,7 @@ function idf_export_main() {
     eval "${idf_exports}"
 
     echo "Checking if Python packages are up to date..."
+    echo "idf_export_main: IDF_PATH before running check_python_dependencies.py: ${IDF_PATH}"
     python ${IDF_PATH}/tools/check_python_dependencies.py || return 1
 
 
diff --git a/tools/check_python_dependencies.py b/tools/check_python_dependencies.py
index bb1d15765..e02b160a7 100755
--- a/tools/check_python_dependencies.py
+++ b/tools/check_python_dependencies.py
@@ -38,6 +38,7 @@ def escape_backslash(path):
 
 if __name__ == "__main__":
     idf_path = os.getenv("IDF_PATH")
+    print('{}: IDF_PATH at start: {}'.format(sys.argv[0], idf_path))
 
     default_requirements_path = os.path.join(idf_path, 'requirements.txt')
 
```

Also, could you please check which shell version you are using? Should be mentioned in the first line if you run `help` in the console.

Viranchee, 2020-04-02:
``` bash
➜ git diff a/export.sh b/export.sh

fatal: ambiguous argument 'a/export.sh': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
```

igrr, 2020-04-02:
Regarding the diff: please apply the changes highlighted in the diff to your files :)
Alternatively you can use `patch` command to apply the diff. But if you are not familiar with `patch`, then might be easier to do it manually.

Viranchee, 2020-04-02:
ah sorry my bad!

Viranchee, 2020-04-02:
``` console
➜  . ./export.sh
idf_export_main: IDF_PATH at start: /Users/virancheel/Desktop/code/esp-idf
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
idf_export_main: IDF_PATH before running check_python_dependencies.py: /Users/virancheel/Desktop/code/esp-idf
/Users/virancheel/Desktop/code/esp-idf/tools/check_python_dependencies.py: IDF_PATH at start: None
Traceback (most recent call last):
  File "/Users/virancheel/Desktop/code/esp-idf/tools/check_python_dependencies.py", line 42, in <module>
    default_requirements_path = os.path.join(idf_path, 'requirements.txt')
  File "/usr/local/opt/python/bin/../Frameworks/Python.framework/Versions/3.7/lib/python3.7/posixpath.py", line 80, in join
    a = os.fspath(a)
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

igrr, 2020-04-02:
I can reproduce this error if I do the following (in a new shell window):

```
$ IDF_PATH=/path/to/idf
$ . ./export.sh
```

instead of
```
$ export IDF_PATH=/path/to/idf
$ . ./export.sh
```

Is there any chance you did the former instead of the latter (forgot `export`)?

Viranchee, 2020-04-02:
Done, solved for me. I used the shell (1st method) instead of the 2nd. Thank you for your quick help :)
This bug should be documented in FAQs or such

Viranchee, 2020-04-03:
Since macos catalina, Apple changed the default shell to `zsh` and not `bash`

Viranchee, 2020-04-03:
I got into another problem now. Cmake not available, although it is installed systemwide

``` console
➜  esp-idf git:(master) cd ../hello_world 
➜  hello_world idf.py set-target esp32
ESP-IDF v4.2-dev-994-gc9f29e0b5
'cmake' must be available on the PATH to use /Users/virancheel/Desktop/code/esp-idf/tools/idf.py
➜  hello_world idf.py menuconfig
ESP-IDF v4.2-dev-994-gc9f29e0b5
'cmake' must be available on the PATH to use /Users/virancheel/Desktop/code/esp-idf/tools/idf.py
```

igrr, 2020-04-03:
> brew install cmake

That's right, is mentioned here in the getting started guide: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/macos-setup.html#install-prerequisites

Viranchee, 2020-04-03:
Running the project in CLion,
I configured environment variables IDF_PATH and IDF_TOOLS_PATH

CLion uses Make files, and here is how the `hello world` example loads up

```
"/Applications/CLion 2020.1 EAP.app/Contents/bin/cmake/mac/bin/cmake" -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /Users/virancheel/Desktop/code/hello_world
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
The following Python requirements are not satisfied:
click>=5.0
future>=0.15.2
cryptography>=2.1.4
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
CMake Error at /Users/virancheel/Desktop/code/esp-idf/tools/cmake/build.cmake:271 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/build.cmake:397 (__build_check_python)
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/virancheel/Desktop/code/hello_world/cmake-build-debug/CMakeFiles/CMakeOutput.log".
See also "/Users/virancheel/Desktop/code/hello_world/cmake-build-debug/CMakeFiles/CMakeError.log".

[Finished]
```

david-cermak, 2020-04-03:
Hi @Viranchee 

CLion also needs to access the tools, so using the `PATH` variable the same way as it's exported by `export.sh`. The above message says, it cannot launch the IDF python environment.
You can solve the issue running `. export.sh` in a console and copying the updated `PATH` to your CLion environment. Preferences -> CMake -> Environment: `PATH=...`

Viranchee, 2020-04-03:
Thank you, I diffed the old `PATH` and PATH by running `. ./export.sh`
Added these env variables

> idf.py
idf_export_main
idf_monitor.py
idf_size.py
idf_tools.py
IDF_PATH
IDF_TOOLS_PATH
IDF_TARGET
IDF_PYTHON_ENV

Now the python requirements are fulfilled
And error message updated to these:

```
"/Applications/CLion 2020.1 EAP.app/Contents/bin/cmake/mac/bin/cmake" -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /Users/virancheel/Desktop/code/hello_world
/usr/local/Cellar/git/2.26.0/libexec/git-core/git-submodule: line 7: basename: command not found
/usr/local/Cellar/git/2.26.0/libexec/git-core/git-submodule: line 7: sed: command not found
/usr/local/Cellar/git/2.26.0/libexec/git-core/git-sh-setup: line 86: basename: command not found
/usr/local/Cellar/git/2.26.0/libexec/git-core/git-sh-setup: line 86: sed: command not found
/usr/local/Cellar/git/2.26.0/libexec/git-core/git-sh-setup: line 333: uname: command not found
/usr/local/Cellar/git/2.26.0/libexec/git-core/git-submodule: line 1181: sed: command not found
/usr/local/Cellar/git/2.26.0/libexec/git-core/git-submodule: line 1181: cmd_: command not found
sh: ps: command not found
sh: ps: command not found
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from /Users/virancheel/Desktop/code/esp-idf/requirements.txt are satisfied.
CMake Error at /Users/virancheel/Desktop/code/esp-idf/components/esp32/project_include.cmake:11 (message):
  Internal error, toolchain has not been set correctly by project (or an
  invalid CMakeCache.txt file has been generated somehow)
Call Stack (most recent call first):
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/build.cmake:306 (include)
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/build.cmake:451 (__build_process_project_includes)
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/virancheel/Desktop/code/hello_world/cmake-build-debug/CMakeFiles/CMakeOutput.log".
See also "/Users/virancheel/Desktop/code/hello_world/cmake-build-debug/CMakeFiles/CMakeError.log".

[Finished]
```

I will run after clearing the cmake generated files again

Viranchee, 2020-04-03:
I set these CLion Environment variables 

``` console
PATH=<long path appended after running . ./export.sh>:$(PATH);
IDF_PATH
IDF_TOOLS_EXPORT_CMD
IDF_PYTHON_ENV_PATH
IDF_TARGET
idf.py
idf_monitor.py
idf_size.py
idf_export_main
```

The output is now

```
"/Applications/CLion 2020.1 EAP.app/Contents/bin/cmake/mac/bin/cmake" -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /Users/virancheel/Desktop/code/hello_world
-- Could NOT find Git (missing: GIT_EXECUTABLE) 
CMake Warning at /Users/virancheel/Desktop/code/esp-idf/tools/cmake/git_submodules.cmake:4 (message):
  Git executable was not found.  Git submodule checks will not be executed.
  If you have any build issues at all, start by adding git executable to the
  PATH and rerun cmake to not see this warning again.
Call Stack (most recent call first):
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/idf.cmake:39 (include)
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

sh: ps: command not found
sh: ps: command not found
CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
-- Configuring incomplete, errors occurred!
See also "/Users/virancheel/Desktop/code/hello_world/cmake-build-debug-esp32/CMakeFiles/CMakeOutput.log".

[Failed to reload]
```

Viranchee, 2020-04-03:
So I set up CLion with these configurations found here: https://www.reddit.com/r/esp32/comments/cejymt/configure_clion_to_use_espidf/

CMake Options:
```
-DIDF_PATH=/Users/virancheel/Desktop/code/esp-idf 
-DPATH=/Users/virancheel/Desktop/code/esp-idf/tools:/Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:$(PATH)
```
Build Options:
`build`
The environment was already set from before

Now, the error message is:

``` console
"/Applications/CLion 2020.1 EAP.app/Contents/bin/cmake/mac/bin/cmake" -DCMAKE_BUILD_TYPE=Debug -DIDF_PATH=/Users/virancheel/Desktop/code/esp-idf -DPATH=/Users/virancheel/Desktop/code/esp-idf/tools:/Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:$(PATH) -G "CodeBlocks - Unix Makefiles" /Users/virancheel/Desktop/code/hello_world
-- Could NOT find Git (missing: GIT_EXECUTABLE) 
CMake Warning at /Users/virancheel/Desktop/code/esp-idf/tools/cmake/git_submodules.cmake:4 (message):
  Git executable was not found.  Git submodule checks will not be executed.
  If you have any build issues at all, start by adding git executable to the
  PATH and rerun cmake to not see this warning again.
Call Stack (most recent call first):
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/idf.cmake:39 (include)
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

sh: ps: command not found
sh: ps: command not found
CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
-- Configuring incomplete, errors occurred!
See also "/Users/virancheel/Desktop/code/hello_world/cmake-build-debug/CMakeFiles/CMakeOutput.log".

[Finished]
```

Viranchee, 2020-04-03:
What are your ESP programming environments? I don't mind any environment / ide. 
It should work! 
Auto completion would be beautiful, but not needed

Viranchee, 2020-04-03:
Update:
Added Toolchains using https://www.esp32.com/viewtopic.php?t=7507

Error logs now:
```
"/Applications/CLion 2020.1 EAP.app/Contents/bin/cmake/mac/bin/cmake" -DCMAKE_BUILD_TYPE=Debug -DCMAKE_C_COMPILER=/Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-cc -DCMAKE_CXX_COMPILER=/Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-c++ -DIDF_PATH=/Users/virancheel/Desktop/code/esp-idf -DPATH=/Users/virancheel/Desktop/code/esp-idf/tools:/Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:$(PATH) -G "CodeBlocks - Unix Makefiles" /Users/virancheel/Desktop/code/hello_world
-- Could NOT find Git (missing: GIT_EXECUTABLE) 
CMake Warning at /Users/virancheel/Desktop/code/esp-idf/tools/cmake/git_submodules.cmake:4 (message):
  Git executable was not found.  Git submodule checks will not be executed.
  If you have any build issues at all, start by adding git executable to the
  PATH and rerun cmake to not see this warning again.
Call Stack (most recent call first):
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/idf.cmake:39 (include)
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

sh: ps: command not found
sh: ps: command not found
CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
-- Configuring incomplete, errors occurred!
See also "/Users/virancheel/Desktop/code/hello_world/cmake-build-debug-xtensa/CMakeFiles/CMakeOutput.log".
You have changed variables that require your cache to be deleted.
Configure will be re-run and you may have to reset some variables.
The following variables have changed:
CMAKE_C_COMPILER= /Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-cc
CMAKE_CXX_COMPILER= /Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-c++

CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_MAKE_PROGRAM
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    PATH

CMake Generate step failed.  Build files cannot be regenerated correctly.

[Finished]
```

Viranchee, 2020-04-03:
So removed `-DPATH` variable from `CMAKE Options`
Error logs updated:

"/Applications/CLion 2020.1 EAP.app/Contents/bin/cmake/mac/bin/cmake" -DCMAKE_BUILD_TYPE=Debug -DCMAKE_C_COMPILER=/Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-cc -DCMAKE_CXX_COMPILER=/Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-c++ -DIDF_PATH=/Users/virancheel/Desktop/code/esp-idf -G "CodeBlocks - Unix Makefiles" /Users/virancheel/Desktop/code/hello_world
-- Could NOT find Git (missing: GIT_EXECUTABLE) 
CMake Warning at /Users/virancheel/Desktop/code/esp-idf/tools/cmake/git_submodules.cmake:4 (message):
  Git executable was not found.  Git submodule checks will not be executed.
  If you have any build issues at all, start by adding git executable to the
  PATH and rerun cmake to not see this warning again.
Call Stack (most recent call first):
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/idf.cmake:39 (include)
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

sh: ps: command not found
sh: ps: command not found
CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
-- Configuring incomplete, errors occurred!
See also "/Users/virancheel/Desktop/code/hello_world/cmake-build-debug-xtensa/CMakeFiles/CMakeOutput.log".
You have changed variables that require your cache to be deleted.
Configure will be re-run and you may have to reset some variables.
The following variables have changed:
CMAKE_C_COMPILER= /Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-cc
CMAKE_CXX_COMPILER= /Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-c++

CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_MAKE_PROGRAM
-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

[Finished]

Viranchee, 2020-04-03:
Update:
Added CMake Options
```
-DCMAKE_MAKE_PROGRAM=/usr/bin/make
-DIDF_PATH=/Users/virancheel/Desktop/code/esp-idf
```
 
CLion / Cmake compiles

Logs:
>
"/Applications/CLion 2020.1 EAP.app/Contents/bin/cmake/mac/bin/cmake" -DCMAKE_BUILD_TYPE=Debug -DCMAKE_C_COMPILER=/Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-cc -DCMAKE_CXX_COMPILER=/Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-c++ -DCMAKE_MAKE_PROGRAM=/usr/bin/make -DIDF_PATH=/Users/virancheel/Desktop/code/esp-idf -G "CodeBlocks - Unix Makefiles" /Users/virancheel/Desktop/code/hello_world
-- Could NOT find Git (missing: GIT_EXECUTABLE) 
CMake Warning at /Users/virancheel/Desktop/code/esp-idf/tools/cmake/git_submodules.cmake:4 (message):
  Git executable was not found.  Git submodule checks will not be executed.
  If you have any build issues at all, start by adding git executable to the
  PATH and rerun cmake to not see this warning again.
Call Stack (most recent call first):
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/idf.cmake:39 (include)
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

sh: ps: command not found
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
sh: ps: command not found
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from /Users/virancheel/Desktop/code/esp-idf/requirements.txt are satisfied.
-- Found PythonInterp: /Users/virancheel/.espressif/python_env/idf4.2_py3.7_env/bin/python (found version "3.7.7") 
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- App "hello-world" version: 1
-- Adding linker script /Users/virancheel/Desktop/code/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /Users/virancheel/Desktop/code/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/virancheel/Desktop/code/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/virancheel/Desktop/code/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /Users/virancheel/Desktop/code/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /Users/virancheel/Desktop/code/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /Users/virancheel/Desktop/code/hello_world/cmake-build-debug-xtensa/esp-idf/esp32/esp32_out.ld
-- Adding linker script /Users/virancheel/Desktop/code/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /Users/virancheel/Desktop/code/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/virancheel/Desktop/code/esp-idf/components/app_trace /Users/virancheel/Desktop/code/esp-idf/components/app_update /Users/virancheel/Desktop/code/esp-idf/components/asio /Users/virancheel/Desktop/code/esp-idf/components/bootloader /Users/virancheel/Desktop/code/esp-idf/components/bootloader_support /Users/virancheel/Desktop/code/esp-idf/components/bt /Users/virancheel/Desktop/code/esp-idf/components/cbor /Users/virancheel/Desktop/code/esp-idf/components/coap /Users/virancheel/Desktop/code/esp-idf/components/console /Users/virancheel/Desktop/code/esp-idf/components/cxx /Users/virancheel/Desktop/code/esp-idf/components/driver /Users/virancheel/Desktop/code/esp-idf/components/efuse /Users/virancheel/Desktop/code/esp-idf/components/esp-tls /Users/virancheel/Desktop/code/esp-idf/components/esp32  [... cut ...] sktop/code/esp-idf/components/unity /Users/virancheel/Desktop/code/esp-idf/components/vfs /Users/virancheel/Desktop/code/esp-idf/components/wear_levelling /Users/virancheel/Desktop/code/esp-idf/components/wifi_provisioning /Users/virancheel/Desktop/code/esp-idf/components/wpa_supplicant /Users/virancheel/Desktop/code/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/virancheel/Desktop/code/hello_world/cmake-build-debug-xtensa

[Finished]

Viranchee, 2020-04-03:
Now I get Code Completion! Hurray!
Thank you @igrr and @david-cermak !

Do you know how to deploy / serial monitor it? or anything else which would make my development a breeze?

Viranchee, 2020-04-03:
Update: 
Changing Cmake variable to `<path>/esp-idf/tools/cmake/toolchain-esp32.cmake` results in an Error:

```
"/Applications/CLion 2020.1 EAP.app/Contents/bin/cmake/mac/bin/cmake" -DCMAKE_BUILD_TYPE=Debug -DCMAKE_C_COMPILER=/Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-cc -DCMAKE_CXX_COMPILER=/Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-c++ -DCMAKE_MAKE_PROGRAM=/Users/virancheel/Desktop/code/esp-idf/tools/cmake/toolchain-esp32.cmake -DIDF_PATH=/Users/virancheel/Desktop/code/esp-idf -DESP_PLATFORM=1 -G "CodeBlocks - Unix Makefiles" /Users/virancheel/Desktop/code/hello_world
-- Could NOT find Git (missing: GIT_EXECUTABLE) 
CMake Warning at /Users/virancheel/Desktop/code/esp-idf/tools/cmake/git_submodules.cmake:4 (message):
  Git executable was not found.  Git submodule checks will not be executed.
  If you have any build issues at all, start by adding git executable to the
  PATH and rerun cmake to not see this warning again.
Call Stack (most recent call first):
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/idf.cmake:39 (include)
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

sh: ps: command not found
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
sh: ps: command not found
CMake Error: Generator: execution of make failed. Make command was: /Users/virancheel/Desktop/code/esp-idf/tools/cmake/toolchain-esp32.cmake cmTC_f68c7/fast && 
-- Check for working C compiler: /Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- broken
CMake Error at /Applications/CLion 2020.1 EAP.app/Contents/bin/cmake/mac/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/Users/virancheel/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/virancheel/Desktop/code/hello_world/cmake-build-debug-xtensa/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/Users/virancheel/Desktop/code/esp-idf/tools/cmake/toolchain-esp32.cmake cmTC_f68c7/fast && Permission denied
    Generator: execution of make failed. Make command was: /Users/virancheel/Desktop/code/esp-idf/tools/cmake/toolchain-esp32.cmake cmTC_f68c7/fast && 
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /Users/virancheel/Desktop/code/esp-idf/tools/cmake/project.cmake:307 (__project)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/virancheel/Desktop/code/hello_world/cmake-build-debug-xtensa/CMakeFiles/CMakeOutput.log".
See also "/Users/virancheel/Desktop/code/hello_world/cmake-build-debug-xtensa/CMakeFiles/CMakeError.log".

[Failed to reload]
```