espressif/esp-idf #3794: NVS key misc::log erased and written every time wifi is init?

negativekelvin, 2019-07-18:
- IDF Version: latest master

This key seems to be forcibly erased and rewritten each time wifi is init, even if the value hasn't changed.  I don't know what it is for, but it seems undesirable.

```
D (1423) nvs: nvs_open_from_partition misc 1
D (1423) nvs: nvs_get_str_or_blob log
D (1433) nvs: nvs_erase_key log
D (1443) nvs: nvs_get_str_or_blob log
D (1443) nvs: nvs_set_blob log 24
```

```
I (1626) nvs misc::log dump:: 02 00 00 00 00 00 00 00 34 e7 fa 3f 44 08 02 00
I (1626) nvs misc::log dump:: 2c 08 02 00 f8 ff fd 3f
```

sagb2015, 2019-07-18:
It is used to store the log level of different WiFi submodules (check esp_wifi_set_debug_log). Can you provide the commit ID where the dump is shown? I am not able to see the dump on latest commit.

negativekelvin, 2019-07-18:
It was happening on ad3b820 but I don't see it anymore after erasing flash so I am going to close this

negativekelvin, 2019-07-18:
Ok I can reproduce this.  If the log blob is not expected format then it will erase and rewrite every time.  For example if the log is set to the 24 byte blob shown above, it will occur.  The key will be erased but instead of writing a new blob with correct format (length = 4?) it will still write a 24 byte blob.  Then on the next boot the same will happen.

sagb2015, 2019-07-19:
@negativekelvin  I am not able to reproduce this at the mentioned commit. Can you be more specific about the steps to reproduce (like maybe changing menuconfig log levels or so)?

negativekelvin, 2019-07-19:
It actually still happens on latest.  Just manually set misc::log to the blob shown one time.  Then whenever wifi stack is init it will erase and rewrite it but it will keep length of 24.  If you erase the key manually then it will write correct blob length 4 and it will not erase it every time.

Alvin1Zhang, 2021-02-02:
Thanks for reporting, the fix is available at https://github.com/espressif/esp-idf/commit/7c6dad2a295526de2c0257ffea40e5c31da291ff, feel free to reopen.

closed 2021-02-02 by Alvin1Zhang