espressif/esp-idf #3796: Wifi connection problem to specific router

rnascunha, 2019-07-18:
Hi, 

I just updated the ESP-IDF (v4.0-dev-1287-gd7e659df2) and my ESP32 stop connecting my router. My router is a ARRIS TG1692A. With older versions of ESP-IDF (v4.0-dev-546-gad3b820e7) it was working perfectly.

The following scenarios was tested:
Router ARRIS / WPA2-PSK / Updated ESP-IDF - Not working;
Router ARRIS / Open  / Updated ESP-IDF - Working;
Router ARRIS / WPA2-PSK / Old ESP-IDF - Working;
Cellphone as Router / WPA2-PSK / Updated ESP-IDF - Working;

So the problem seems to be something incompatible between my router and the new version of ESP-IDF authentication WPA2-PSK. 

Anyone has faced the same problem? Was there any changes to the WPA2-PSK auth (my guess is something at the handshake)? For now I will keep using the old ESP-IDF, but I really would apreciate to update as soons as possible.

Also, I used to use the Make build, and now using the CMake, but everything compiled normally. (tested with the default wifi_station example).

Thank you,
Rafael  

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: ESP32D0WDQ6 (revision 0)
- Module or chip used:  ESP-WROOM-32
- IDF version:  v4.0-dev-1287-gd7e659df2
- Build System:         CMake
- Compiler version: 8.2.0
- Operating System: Linux
- Power Supply: USB

negativekelvin, 2019-07-18:
https://github.com/espressif/esp-idf/issues/3725

rnascunha, 2019-07-18:
@negativekelvin , thanks!

Using ```idf.py erase_flash``` resolves the problem. I will close this issue and keep track of the permant solution at #3725 .

sagb2015, 2019-07-19:
The fix for the same is merged internally and should be available soon. As mentioned on #3725, you can use [this](https://github.com/espressif/esp-idf/issues/3726#issuecomment-508789844) workaround until then.

ShaqlainGarana, 2022-05-16:
I am having same issue in idf v4.4 please reopen the issue
i have tried erase_flash as well but still the issue exists in specific routers

I (1037) wifi:wifi firmware version: 7679c42
I (1037) wifi:wifi certification version: v7.0
I (1037) wifi:config NVS flash: enabled
I (1037) wifi:config nano formating: disabled
I (1047) wifi:Init data frame dynamic rx buffer num: 64
I (1047) wifi:Init management frame dynamic rx buffer num: 64
I (1057) wifi:Init management short buffer num: 32
I (1057) wifi:Init dynamic tx buffer num: 32
I (1067) wifi:Init static rx buffer size: 1600
I (1067) wifi:Init static rx buffer num: 10
I (1077) wifi:Init dynamic rx buffer num: 64
I (1077) wifi_init: rx ba win: 6
I (1077) wifi_init: tcpip mbox: 32
I (1087) wifi_init: udp mbox: 6
I (1087) wifi_init: tcp mbox: 6
I (1087) wifi_init: tcp tx win: 5744
I (1097) wifi_init: tcp rx win: 5744
I (1097) wifi_init: tcp mss: 1440
I (1107) wifi_init: WiFi IRAM OP enabled
I (1107) wifi_init: WiFi RX IRAM OP enabled
I (1117) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1237) wifi:mode : sta (8c:4b:14:74:cb:cc)
I (1237) wifi:enable tsf
I (1287) FS: Read from file: 'Testing'
I (1287) FS: Read from file: 'White@123'
I (1287) BLUFI_EXAMPLE: ssid is Testing
I (1287) BLUFI_EXAMPLE: password is White@123
I (1597) wifi:new:<3,1>, old:<1,0>, ap:<255,255>, sta:<3,1>, prof:1
I (2357) wifi:state: init -> auth (b0)
I (3357) wifi:state: auth -> init (200)
I (3357) wifi:new:<3,0>, old:<3,1>, ap:<255,255>, sta:<3,1>, prof:1
I (3357) WiFi: retry to connect to the AP : 1
I (5417) WiFi: retry to connect to the AP : 2
I (5487) wifi:new:<3,1>, old:<3,0>, ap:<255,255>, sta:<3,1>, prof:1
I (5487) wifi:state: init -> auth (b0)
I (6487) wifi:state: auth -> init (200)
I (6487) wifi:new:<3,0>, old:<3,1>, ap:<255,255>, sta:<3,1>, prof:1

kapilkedawat, 2022-05-16:
HI @ShaqlainGarana ,

Seems like we are getting the reason code 200, which is due to beacon timeout. Can you please share the wireless sniffer capture so that we can confirm the same.

Also can you set the esp-idf logs level as debug which will give us more info about the issue.

ShaqlainGarana, 2022-05-16:
which tool i should use for sniffer tool ? and also please guide me whats should be the setting for the same
 i will shortly share the logs.

ShaqlainGarana, 2022-05-16:
I have added few more lines in my code in order to receive the reason for disconnection.

I am getting 2 reason 

reason : 02  (AUTH_EXPIRE )
reason : 205

ShaqlainGarana, 2022-05-16:
@kapilkedawat 

Here are the debug logs:

D (66696) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (66696) wifi:perform scan: ss_state=0x9, chan<12,0>, dur<360,360>
D (66936) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (66936) wifi:perform scan: ss_state=0x9, chan<13,0>, dur<360,360>
D (67176) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (67176) wifi:filter: set rx policy=4
D (67176) wifi:first chan=1
D (67176) wifi:handoff_cb: status=0
D (67176) wifi:clear blacklist
D (67186) wifi:clear rc list
D (67186) wifi:clear blacklist
D (67186) wifi:send disconnect event
D (67196) wifi:connect status 1 -> 3
D (67196) wifi:disable connect timer
D (67196) wifi:clear scan ap list
D (67206) event: running post WIFI_EVENT:5 with handler 0x40126ee8 and context 0x3ffd0100 on loop 0x3ffb7638
0x40126ee8: wifi_default_action_sta_disconnected at F:/esp/esp_v4.4_release/esp-idf/components/esp_wifi/src/wifi_default.c:100

D (67216) esp_netif_handlers: esp_netif action disconnected with netif0x3ffcd674 from event_id=5
D (67216) esp_netif_lwip: check: remote, if=0x3ffcd674 fn=0x40125084
0x40125084: esp_netif_down_api at F:/esp/esp_v4.4_release/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1279

D (67226) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffcd674
D (67236) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffcd674
D (67236) esp_netif_lwip: if0x3ffcd674 start ip lost tmr: no need start because netif=0x3ffcff38 interval=120 ip=0
D (67246) esp_netif_lwip: check: local, if=0x3ffcd674 fn=0x401256d8
0x401256d8: esp_netif_update_default_netif_lwip at F:/esp/esp_v4.4_release/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:188

D (67256) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffcd674
D (67266) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (67266) event: running post WIFI_EVENT:5 with handler 0x400dc880 and context 0x3ffd01d8 on loop 0x3ffb7638
0x400dc880: wifi_event_handler at F:\esp\workplace\template-app\build/../main/blufiApp.c:98

I (67276) BLUFI_EXAMPLE: Disconnect reason : 205
D (67286) wifi:Start wifi connect
D (67286) wifi:connect status 3 -> 0
D (67286) wifi:connect chan=0
D (67296) wifi:first chan=3
D (67296) wifi:connect status 0 -> 1
D (67296) wifi:filter: set rx policy=3
D (67306) wifi:clear scan ap list
D (67306) wifi:start scan: type=0x50f, priority=2, cb=0x40170c80, arg=0x0, ss_state=0x1, time=65023232, index=0
0x40170c80: cnx_start_handoff_cb at ??:?

D (67316) wifi:perform scan: ss_state=0x9, chan<3,0>, dur<0,120>
I (67326) WiFi: retry to connect to the AP : 20
D (67396) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (67396) wifi:profile match: ss_state=0x7
D (67396) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (67396) wifi:find first mathched ssid, scan done
D (67406) wifi:filter: set rx policy=4
D (67406) wifi:first chan=1
D (67406) wifi:handoff_cb: status=0
D (67416) wifi:ap found, mac=04:95:e6:b4:39:d1
D (67416) wifi:new_bss=0x3ffcad50, cur_bss=0x0, new_chan=<3,1>, cur_chan=3
D (67426) wifi:filter: set rx policy=5
I (67426) wifi:new:<3,1>, old:<3,0>, ap:<255,255>, sta:<3,1>, prof:1
D (67436) wifi:connect_op: status=0, auth=5, cipher=3
D (67436) wifi:auth mode is not none
D (67446) wifi:connect_bss: auth=1, reconnect=0
I (67446) wifi:state: init -> auth (b0)
D (67446) wifi:start 1s AUTH timer
D (67456) wifi:clear scan ap list
D (68456) wifi:auth timeout
I (68456) wifi:state: auth -> init (200)
D (68456) wifi:connect status 1 -> 4
D (68456) wifi:stop beacon/connect timer
D (68456) wifi:reason: auth expire(2)
D (68456) wifi:add bssid 04:95:e6:b4:39:d1 to blacklist, cnt=0
D (68466) wifi:stop CSA timer
D (68466) wifi:remove 04:95:e6:b4:39:d1 from rc list
I (68476) wifi:new:<3,0>, old:<3,1>, ap:<255,255>, sta:<3,1>, prof:1
D (68476) wifi:filter: set rx policy=8
D (68486) wifi:Send disconnect event, reason=2, AP number=0
D (68486) event: running post WIFI_EVENT:5 with handler 0x40126ee8 and context 0x3ffd0100 on loop 0x3ffb7638
0x40126ee8: wifi_default_action_sta_disconnected at F:/esp/esp_v4.4_release/esp-idf/components/esp_wifi/src/wifi_default.c:100

D (68496) esp_netif_handlers: esp_netif action disconnected with netif0x3ffcd674 from event_id=5
D (68506) esp_netif_lwip: check: remote, if=0x3ffcd674 fn=0x40125084
0x40125084: esp_netif_down_api at F:/esp/esp_v4.4_release/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1279

D (68516) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffcd674
D (68516) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffcd674
D (68526) esp_netif_lwip: if0x3ffcd674 start ip lost tmr: no need start because netif=0x3ffcff38 interval=120 ip=0
D (68536) esp_netif_lwip: check: local, if=0x3ffcd674 fn=0x401256d8
0x401256d8: esp_netif_update_default_netif_lwip at F:/esp/esp_v4.4_release/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:188

D (68546) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffcd674
D (68546) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (68556) event: running post WIFI_EVENT:5 with handler 0x400dc880 and context 0x3ffd01d8 on loop 0x3ffb7638
0x400dc880: wifi_event_handler at F:\esp\workplace\template-app\build/../main/blufiApp.c:98

I (68566) BLUFI_EXAMPLE: Disconnect reason : 2
D (68566) wifi:Start wifi connect
D (68576) wifi:connect status 4 -> 0
D (68576) wifi:connect chan=0
D (68576) wifi:first chan=3
D (68586) wifi:connect status 0 -> 1
D (68586) wifi:filter: set rx policy=3
D (68586) wifi:clear scan ap list
D (68596) wifi:start scan: type=0x50f, priority=2, cb=0x40170c80, arg=0x0, ss_state=0x1, time=66308961, index=0

nishanth-radja, 2022-05-16:
@ShaqlainGarana If you have a ubuntu laptop, you can use this steps to get the capture.
1.Install the wireshark on laptop
sudo apt install wireshark

2. Use this script by editing the wifi interface and channel

ifconfig <wifi-interface> down
iwconfig <wifi-interface> mode monitor
ifconfig <wifi-interface> up
iwconfig <wifi-interface> channel <channel>
wireshark

ShaqlainGarana, 2022-05-16:
i am using windows laptop

ShaqlainGarana, 2022-05-16:
[image]

ShaqlainGarana, 2022-05-16:
[image]

nishanth-radja, 2022-05-16:
@ShaqlainGarana Do you have any MAC laptop?

ShaqlainGarana, 2022-05-17:
[capture.zip](https://github.com/espressif/esp-idf/files/8705346/capture.zip)
@kapilkedawat @nishanth-radja 
here are the package capture

ShaqlainGarana, 2022-05-17:
[image]
[image]

kapilkedawat, 2022-05-17:
@ShaqlainGarana As you can see from the captures, AP is not replying to the auth packets. Can you please try to connect using any other device(Laptop/mobile) and see if that is able to connect?

ShaqlainGarana, 2022-05-17:
all other device are connecting including esp32 with same router/AP

ShaqlainGarana, 2022-05-17:
with same code also some devices are connecting & same hardware

kapilkedawat, 2022-05-17:
Can you share the AP's RSSI which device is getting?

ShaqlainGarana, 2022-05-17:
its vary from -40 to -63 dBm

ShaqlainGarana, 2022-05-17:
@kapilkedawat in debug log i am continuedly getting AUTH_EXPIRE error as i have added the code to determine the disconnect reason.

ShaqlainGarana, 2022-05-17:
D (97636) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (97636) wifi:profile match: ss_state=0x7
D (97636) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (97636) wifi:find first mathched ssid, scan done
D (97646) wifi:filter: set rx policy=4
D (97646) wifi:first chan=1
D (97646) wifi:handoff_cb: status=0
D (97656) wifi:ap found, mac=04:95:e6:b4:39:d1
D (97656) wifi:new_bss=0x3ffcad50, cur_bss=0x0, new_chan=<2,1>, cur_chan=2
D (97666) wifi:filter: set rx policy=5
I (97666) wifi:new:<2,1>, old:<2,0>, ap:<255,255>, sta:<2,1>, prof:1
D (97676) wifi:connect_op: status=0, auth=5, cipher=3
D (97676) wifi:auth mode is not none
D (97686) wifi:connect_bss: auth=1, reconnect=0
I (97686) wifi:state: init -> auth (b0)
D (97686) wifi:start 1s AUTH timer
D (97696) wifi:clear scan ap list
D (98576) event: running post WIFI_EVENT:5 with handler 0x401157c0 and context 0x3ffda7c0 on loop 0x3ffd1d88
0x401157c0: r_intc_init at ??:?

 wifi:profile match: ss_state=0x7
D (94156) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (94156) wifi:find first mathched ssid, scan done
D (94166) wifi:filter: set rx policy=4
D (94166) wifi:first chan=1
D (94166) wifi:handoff_cb: status=0
D (94176) wifi:ap found, mac=04:95:e6:b4:39:d1
D (94176) wifi:new_bss=0x3ffcad50, cur_bss=0x0, new_chan=<2,1>, cur_chan=2
D (94186) wifi:filter: set rx policy=5
I (94186) wifi:new:<2,1>, old:<2,0>, ap:<255,255>, sta:<2,1>, prof:1
D (94196) wifi:connect_op: status=0, auth=5, cipher=3
D (94196) wifi:auth mode is not none
D (94196) wifi:connect_bss: auth=1, reconnect=0
I (94206) wifi:state: init -> auth (b0)
D (94206) wifi:start 1s AUTH timer
D (94206) wifi:clear scan ap list
D (95096) event: running post WIFI_EVENT:5 with handler 0x401157c0 and context 0x3ffda7c0 on loop 0x3ffd1d88
0x401157c0: r_intc_init at ??:?

D (95096) event: running post WIFI_EVENT:5 with handler 0x400d7100 and context 0x3ffda7ec on loop 0x3ffd1d88
0x400d7100: gpio_ll_pullup_dis at D:/esp/esp-idf/components/hal/esp32/include/hal/gpio_ll.h:66
 (inlined by) gpio_pullup_dis at D:/esp/esp-idf/components/driver/gpio.c:105

D (95206) wifi:auth timeout
I (95206) wifi:state: auth -> init (200)
D (95216) wifi:connect status 1 -> 4
D (95216) wifi:stop beacon/connect timer
D (95216) wifi:reason: auth expire(2)
D (95216) wifi:add bssid 04:95:e6:b4:39:d1 to blacklist, cnt=0
D (95226) wifi:stop CSA timer
D (95226) wifi:remove 04:95:e6:b4:39:d1 from rc list
I (95226) wifi:new:<2,0>, old:<2,1>, ap:<255,255>, sta:<2,1>, prof:1
D (95236) wifi:filter: set rx policy=8
D (95236) wifi:Send disconnect event, reason=2, AP number=0
D (95246) event: running post WIFI_EVENT:5 with handler 0x40126ee8 and context 0x3ffd2f9c on loop 0x3ffd1d88
0x40126ee8: sm_WPA_PTK_PTKCALCNEGOTIATING2_Enter at D:/esp/esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:1609
 (inlined by) sm_WPA_PTK_Step at D:/esp/esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:1940
 (inlined by) wpa_sm_step at D:/esp/esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:2374
 (inlined by) wpa_sm_step at D:/esp/esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:2353

D (95256) esp_netif_handlers: esp_netif action disconnected with netif0x3ffd2cb4 from event_id=5
D (95266) esp_netif_lwip: check: remote, if=0x3ffd2cb4 fn=0x40125084
0x40125084: wpa_sta_cur_pmksa_matches_akm at D:/esp/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:2421

D (95266) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffd2cb4
D (95276) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffd2cb4
D (95286) esp_netif_lwip: if0x3ffd2cb4 start ip lost tmr: no need start because netif=0x3ffd2d14 interval=120 ip=0
D (95296) esp_netif_lwip: check: local, if=0x3ffd2cb4 fn=0x401256d8

kapilkedawat, 2022-05-17:
@ShaqlainGarana this is very odd and seems like AP side issue since it is supposed to reply to auth packets. I don't see any ack for those packets in sniffer capture as well.

Could you share the sniffer capture from other devices to see how they are behaving and working fine? 

Also is the environment is too congested? Is this happening with only some chips and happening consistently with them? Can you try to bring the chip little bit closer to AP and see if that helps?

ShaqlainGarana, 2022-05-17:
@kapilkedawat  
i have tried everything and all other devicees except few of my devices easily connects to the router without any issue
you can check the zip file i have sent entire log files including other devices as well

ShaqlainGarana, 2022-05-17:
also tried arduino code and still it is not connecting with that specific AP

kapilkedawat, 2022-05-17:
@ShaqlainGarana I took a look at both capture.pcapng and capture1.pcapng . Both doesn't contain any association from any device.

Also I see RSSI of 8c:4b:14:74:cb:cc around -50. dbm  and RSSI of 0c:dc:7e:cd:23:a8 around -80 dbm on the sniffer. Can you please reconfirm  the RSSI on the device for AP.

You can use the code from scan example https://github.com/espressif/esp-idf/blob/master/examples/wifi/scan/main/scan.c#L110 to print the RSSIs.

kapilkedawat, 2022-05-17:
Is it possible for you to provide a connection capture when device is successfully able to connect? 

From the current logs it seems to me, nothing is wrong in station side, AP is discarding the auth packets and not even giving hardware acks for those so the connection can't proceed.

kapilkedawat, 2022-05-23:
@ShaqlainGarana Can you please share the success capture between same AP and esp device?

ShaqlainGarana, 2022-05-27:
@kapilkedawat can you please tell me how to associate specific device in packet capture so that i can share you the required data

kapilkedawat, 2022-06-03:
Hi @ShaqlainGarana , its the same process, just try to connect to AP when you are sniffing the packets. Sniffer will capture it.

nishanth-radja, 2022-06-06:
@ShaqlainGarana Also I see that the sniffer is configured on channel 2, while the AP is on channel 3. Can you pls configure the sniffer also on channel 3.
iwconfig wlan0 channel 3
wireshark
And is the sniffer laptop kept close to the station or the AP?
How far is the AP from the client? Can you keep the sniffer device a little further from the DUT and close to AP to find out  on how much RSSI does the AP see the client. 
[image]

AxelLin, 2022-06-15:
@ShaqlainGarana Any update regarding the comment: https://github.com/espressif/esp-idf/issues/3796#issuecomment-1147035215

AxelLin, 2022-07-26:
@ShaqlainGarana Any update?

AxelLin, 2023-05-03:
@ShaqlainGarana @rnascunha 
Do you still hit this issue with recent esp-idf releases?

ShaqlainGarana, 2023-05-03:
Issue is fixed

closed 2023-06-02 by jack0c