espressif/esp-idf #3797: New toolchain required

Reiner1210, 2019-07-18:
Hello,

the new IDF requires a new toolchain but the links in you docu are still the old one
I only found this
https://www.esp32.com/viewtopic.php?t=7400

negativekelvin, 2019-07-19:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-set-up-tools

Reiner1210, 2019-07-24:
@negativekelvin And where will I find the xtensa compiler for Linux - nothing in your docu!

projectgus, 2019-07-24:
Hi @Reiner1210,

If you're using the recent ESP-IDF master branch, the gcc 8 toolchain is needed and we've introduced the IDF Tools Installer which can manage user installation of the correct tools including the toolchain. It's mentioned here:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#id3

As you probably are familiar, if you're using an ESP-IDF v3.x branch or stable release, the gcc 5 toolchain is standard and the download links are provided directly in the docs:
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/linux-setup.html#toolchain-setup

projectgus, 2019-07-24:
Will add some description under the Linux and macOS heading so it's clearer what this does (downloads all of the required tools, including the compiler.)

Reiner1210, 2019-07-26:
Ok. In the meantime I have compiled it from sources :-)

projectgus, 2019-07-29:
Reopening until we have a fix in the docs for this.

igrr, 2020-09-09:
Description of the install scripts has been added in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html#install-scripts, and the links to the toolchain in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html#list-of-idf-tools.

closed 2020-09-09 by igrr