espressif/esp-idf #3800: Detect factory reset after deep sleep

ducalex, 2019-07-19:
## Environment

    Development Kit: Lolin32
    Kit version
    Module or chip used: ESP32-WROOM-32D
    IDF version: v3.3-beta1-849-g630ffde6b
    Build System: Make
    Compiler version: crosstool-ng-1.22.0-80-g6c4433a5
    Operating System: Windows
    Power Supply: USB

## Problem Description

If the factory reset gpio is asserted while booting from deep-sleep then the application is started with a possibly incorrect state because of preserved RTC memory but possibly different boot image and gone nvs.

### Expected Behavior

One of three things:
- The boot loader should do a clean restart after a factory reset during deep-sleep wake up
- Or The factory reset should happen only on POWERON_RESET
- Or The IDF gives us a way to detect that a factory reset just occurred

### Actual Behavior

The application boots thinking it just woke from normal deep sleep and assumes its RTC variables and previous boot image are the same.

### Steps to repropduce

1. Set factory reset gpio to 15
2. Set gpio 15 as a wake up source for deep sleep
3. Go to deep sleep
4. Hold gpio 15

### Code to reproduce this issue

```cpp

```

## Debug Logs

```
```

## Other items if possible

negativekelvin, 2019-07-20:
Good suggestion.  How about something like

```
#ifdef CONFIG_BOOTLOADER_OTA_DATA_ERASE
    RESET_REASON rst_res = rtc_get_reset_reason(0);
    uint32_t is_rev0 = (GET_PERI_REG_BITS2(EFUSE_BLK0_RDATA3_REG, 1, 15) == 0);
    if(rst_res == POWERON_RESET || (is_rev0 && rst_res == RTCWDT_RTC_RESET))
        ota_data_erase = true;
#endif
```

ducalex, 2019-07-21:
@negativekelvin That seems reasonable and the safest option. 

Thank you for your code, I will use it for my current project and I hope that it eventually makes its way in esp-idf!

closed 2019-08-30 by igrr