espressif/esp-idf #3801: SPI_slave.c recvbuf malformed

DrTester0x42, 2019-07-19:
I am using a modified version of https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_slave essentially the master just "listens" and the slave just sends data.

However, on my master side I receive the following data while `make monitor`:

```bash
recvbuf[11269]: ���������������������������������������������������������������������������������������������������������������������������������@H@T@`@t@$@\?@\@h?@t?@p@p�

recvbuf[11270]: ���������������������������������������������������������������������������������������������������������������������������������@H@T@`@t@$@\?@\@h?@t?@p@p�
```

The sender is sending a string: 

>"This is the receiver, sending data for transmission number %04d"

Here is my two code sets:

https://gist.github.com/DrTester0x42/a0303af146a92e4d58f44dd33c77e66a

Why is my data "garbled" and what can I do to fix this?

DrTester0x42, 2019-07-19:
Also I am using the ESP32 Huzzah https://learn.adafruit.com/adafruit-huzzah32-esp32-feather

ginkgm, 2019-10-08:
@DrTester0x42 Could you please use the handshake to make sure the slave sending data is ready before you read?

In order to check whether it is this problem, you can add a quite long delay here and try again: 

https://gist.github.com/DrTester0x42/a0303af146a92e4d58f44dd33c77e66a#file-master-c-L79

``vTaskDelay(100)``

Alvin1Zhang, 2020-01-15:
@DrTester0x42 Thanks for reporting. Would you please help share if any updates for the issue? Thanks.

OmNamasi, 2021-05-17:
I have the similar issue. I am using esp-idf stable version 4.2, 5 wire connection, both master & slave is ESP32 (ESP-WROOM-32 module) & used sampled code from "https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_slave" to send & receive the data. Intermittently I see the master sent messages are not received by the receiver.

So to isolate the issue I limit the master to send the data & slave to receive/read the data (attached the files). And changed the clock frequency to 8.89MHz (i.e. SPI_MASTER_FREQ_9M). The modified code can be found at: https://gist.github.com/OmNamasi/9fa13020c01014d2ff41801fa0b33990

Yet I am seeing some message from master isn't received by the slave & sometimes the full length of the message isn't read by the slave. 
E.g. See the below picture from the slave, shows message "0003", "0006", "0008" e.t.c. is not read correctly or dropped.

[image]

After adding a delay (vTaskDelay(100)) to slave code (line 134), still see the issue

[image]