espressif/esp-idf #3802: SPI Flash: set write protect fails

orcatune, 2019-07-20:
## Environment

- Development Kit:     none
- Module or chip used:  ESP32-PICO-D4
- IDF version: v4.0-dev-1290-g15fc4556a
- Build System:         Make
- Compiler version: 1.22.0-80-g6c4433a5
- Operating System:   Windows
- Power Supply:        external 3.3V

## Problem Description
OTA updates failing at esp_ota_write API call. Flash encryption is enabled. No issue with IDF v3.3.

### Expected Behavior
esp_ota_write returns ESP_OK

### Actual Behavior
esp_ota_write returns ESP_ERR_FLASH_OP_FAIL

Traced the problem to esp_flash_set_chip_write_protect function in esp_flash_api.c:382
`VERIFY_OP(set_chip_write_protect);`

Macro VERIFY_OP definition at esp_flash_api.c:205:
```
// Convenience macro for beginning of all API functions,
// check that the 'chip' parameter is properly initialised
// and supports the operation in question
#define VERIFY_OP(OP) do {                                  \
        if (chip == NULL) {                                 \
            chip = esp_flash_default_chip;                  \
        }                                                   \
        if (chip == NULL || !esp_flash_chip_driver_initialized(chip)) { \
            return ESP_ERR_FLASH_NOT_INITIALISED;               \
        }                                                   \
        if (chip->chip_drv->OP == NULL) {                        \
            return ESP_ERR_FLASH_UNSUPPORTED_CHIP;              \
        }                                                   \
    } while (0)
```

In my case, generic SPI flash driver is used. So chip->chip_drv->set_chip_write_protect is defined in spi_flash_chip_generic.c:385
```
    // TODO: figure out if generic chip-wide protection bits exist across some manufacturers
    .get_chip_write_protect = NULL,
    .set_chip_write_protect = NULL,
```

Since .set_chip_write_protect is always NULL, esp_flash_set_chip_write_protect will always return ESP_ERR_FLASH_UNSUPPORTED_CHIP. Commenting out the relevant if statement got OTA update working again.

negativekelvin, 2019-07-21:
https://github.com/espressif/esp-idf/blob/beaefd3359973dcf8bba4e1d4462aa7f5d67be6a/components/spi_flash/Kconfig#L80-L82

https://github.com/espressif/esp-idf/blob/28f1cdf5ed7149d146ad5019c265c8bc3bfa2ac9/components/nvs_flash/test_nvs_host/sdkconfig.h#L2-L3

Hmm

ginkgm, 2019-09-05:
Hi @revtronix @negativekelvin ,
74c2eb3aff0a4b26b1aa8f9aea419e7e6b450f3e this commit on master should fix the problem

ginkgm, 2019-09-24:
Thread closed, feel free to open it later if you still find issues with this commit

closed 2019-09-25 by ginkgm