espressif/esp-idf #3803: httpd_req_get_hdr_value_str() fails to find a field key after an empty value

yappy, 2019-07-20:
## Environment
- IDF version (run ``git describe --tags`` to find it): 
    v3.2
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    1.22.0-80-g6c4433a
- Operating System:     Windows (WSL)
- Power Supply:         USB

## Problem Description
When there is an empty value entry in HTTP header, httpd_req_get_hdr_value_str() returns ESP_ERR_NOT_FOUND.
```
Content-Type:<space><cr><lf>
X-FILE-NAME:<space>index.lua<cr><lf>

CANNOT find X-FILE-NAME
If Content-Type is not empty (e.g. image/jpeg) , it succeeds.
```

### Expected Behavior
httpd_req_get_hdr_value_str() succeeds and returns its value. (index.lua)

### Actual Behavior
httpd_req_get_hdr_value_str() returns ESP_ERR_NOT_FOUND.

### Steps to repropduce
[image]

### Code to reproduce this issue

```cpp
const uint32_t HTTP_FILE_NAME_MAX = 32;
char name[HTTP_FILE_NAME_MAX];
ret = httpd_req_get_hdr_value_str(req, "X-FILE-NAME", name, sizeof(name));
```

anurag-kar, 2019-07-21:
@yappy Thanks for reporting the issue. This [patch](https://github.com/espressif/esp-idf/files/3415000/httpd_parser.patch.txt) should fix the issue. Let me know if it works for you. In the meanwhile we'll try to merge the bugfix as soon as possible.

yappy, 2019-07-22:
Thank you very much!
It works in my app.  It is not urgent for me because there is a workaround (set a non-empty value on the client side) for it.

yappy, 2019-09-10:
Any update on this?