espressif/esp-idf #3808: Why the menuconfig 'store ble mesh node configuration persistently' dose not work?

YuriChen1028, 2019-07-22:
I have enabled the function 'component config -> Ble mesh support -> store ble mesh node configuration persistently' in the menuconfig, but when i turn off the devkit board and turn on again,  the nrf Mesh app remind me provision again.

YuriChen1028, 2019-07-22:
the SDK version is : mesh-v0.6.1

WCCWCC, 2019-08-26:
Hi YuriChen，

1.  nrf mesh app 配置设备入网并进行控制。
2.  设备硬件复位， nrf mesh app 可以直接和设备建立连接。你需要点击右上角的 connect 。如下图所示：
 
[image]

[image]

Alvin1Zhang, 2020-06-23:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2020-06-23 by Alvin1Zhang

arjndr, 2024-02-05:
@Alvin1Zhang this issue is similar to #8942 

Enabling the "Store BLE Mesh configuration persistently" option seems to do nothing, whenever the device restarts I need to reprovision it!

[image]