espressif/esp-idf #3810: What is "Double exception"?

KaeLL, 2019-07-22:
I've been getting this exception after isolating a corner case to trigger it. I tried looking online for information about it, and couldn't find anything but the xtensa ISA pdf file, in which it's said to be a vector of CPU exceptions rather than the exception itself.
What I want to know is: what is this class of exceptions? Why the ESP32 doesn't automatically reboots itself after this (class of) exception(s) is triggered with the "Print registers and reboot" option enabled for FreeRTOS? Is [THIS](https://github.com/espressif/esp-idf/blob/master/components/freertos/xtensa_vectors.S#L528) supposed to be helpful?

igrr, 2019-07-23:
@KaeLL Can you please share the output you get from the monitor when this issue happens? The interesting part is the beginning of the crash, the first couple of panic handler output cycles.

DoubleException happens when an exception happens while handling another exception. So there are two issues here: first is that the panic handler crashes (triggers the DoubleException) and second is that it keeps looping, reentering the panic handler and triggering the exception again.
With information from your monitor output, we can fix the first issue. Also we will modify the DoubleException vector to cause an RTC watchdog reset instead of entering the panic handler. This will provide less information but will be a better safeguard against the device being locked up.

KaeLL, 2019-07-23:
```
Guru Meditation Error: Core  1 panic'ed (Double exception)
 1 1 1 1 1���? 1���? 1���?
```
```
Guru Meditation Error: Core  1 panic'ed (Double except�
```
```
Guru Meditation Error: Core  1 panic'ed (Double exception)
```

Unhelpful, I know. But that's all that it prints. It should be noticed that the app was compiled with Info level log

Here's the complete log of all runs
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6684
load:0x40078000,len:15664
load:0x40080400,len:5696
entry 0x400806a8
I (797) cpu_start: Pro cpu up.
I (797) cpu_start: Application information:
I (797) cpu_start: Project name:     TEST
I (800) cpu_start: App version:      0.2.1-dirty
I (806) cpu_start: Compile time:     Jul  8 2019 18:04:32
I (812) cpu_start: ELF file SHA256:  ae32296154201836...
I (818) cpu_start: ESP-IDF:          v4.0-dev-1136-g28f1cdf-dirty
I (825) cpu_start: Starting app cpu, entry point is 0x40081214
0x40081214: call_start_cpu1 at ~/Documents/VSCode/esp-idf/components/esp32/cpu_start.c:298

W (831) cpu_start: Flash encryption mode is DEVELOPMENT
I (0) cpu_start: App cpu up.
I (841) heap_init: Initializing. RAM available for dynamic allocation:
I (848) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (854) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (860) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (866) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (872) heap_init: At 3FFCB330 len 00014CD0 (83 KiB): DRAM
I (878) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (885) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (891) heap_init: At 40097A44 len 000085BC (33 KiB): IRAM
I (897) cpu_start: Pro cpu start user code
I (936) spi_flash: detected chip: generic
I (936) spi_flash: flash io: dio
I (936) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (984) main: IDF VERSION: v4.0-dev-1136-g28f1cdf-dirty
I (989) main: APPLICATION VERSION: 0.2.1-dirty
I (989) main: PROJECT NAME: TEST
I (994) main: COMPILE DATE: Jul  8 2019 18:04:32
I (999) main: IDF VERSION: v4.0-dev-1136-g28f1cdf-dirty
I (1004) main: esp_register_shutdown_handler: 0
I (1009) softwd: WD OK
I (1014) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1019) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1029) gpio: GPIO[13]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1039) gpio: GPIO[14]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1049) gpio: GPIO[5]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1069) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1269) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
W (6789) mbedtls: ssl_tls.c:5757 x509_verify_cert() returned -9984 (-0x2700)

W (9084) mbedtls: ssl_tls.c:5757 x509_verify_cert() returned -9984 (-0x2700)

W (11204) mbedtls: ssl_tls.c:5757 x509_verify_cert() returned -9984 (-0x2700)

[... 224 lines cut ...]
I (1014) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1019) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1029) gpio: GPIO[13]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1039) gpio: GPIO[14]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1049) gpio: GPIO[5]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1069) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1269) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
[0;33mW (6789) mbedtls: ssl_tls.c:5757 x509_verify_cert() returned -9984 (-0x2700)

W (9394) mbedtls: ssl_tls.c:5757 x509_verify_cert() returned -9984 (-0x2700)

W (11509) mbedtls: ssl_tls.c:5757 x509_verify_cert() returned -9984 (-0x2700)

I (12569) main: initializing NFC... 
I (12569) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (12584) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:1 
I (12584) main: initializing Bluetooth

I (12589) BTDM_INIT: BT controller compile version [7a7769b]
I (12594) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (12689) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (12974) BEACON: Register callback
I (12984) BEACON: Start scanning...
Guru Meditation Error: Core  1 panic'ed (Double except�
```

user897943, 2021-03-18:
Almost 2 years and nobody from Espressif bothered to fix this? I use a Konnected Pro alarm panel that uses this chip and every time the board has to restart for whatever reason, I get:

```
[INFO   1616096023.773201,   169808] /lfs/variables_set.lua:15: Wrote:  device_config.lua
[INFO   1616096024.782533,   170448] /lfs/network.lua:49:  Network connection obtained with IP:  10.0.0.106 Gateway: 10.0.0.1
[INFO   1616096029.799576,   169748] /lfs/network.lua:64: Current Date/Time: 2021-03-18 19:33:49 UTC
[INFO   1616096029.810647,   169856] /lfs/network.lua:28: Booting Konnected application
[INFO   1616096029.853287,   169696] /lfs/server.lua:6: UPnP: Listening for UPnP discovery
[INFO   1616096029.878845,   168880] /lfs/server.lua:24: HTTP:  Starting server at http://10.0.0.106:9123
[INFO   1616096029.894115,   168564] /lfs/network.lua:30: Loaded:  server
[INFO   1616096030.084359,   162344] /lfs/application.lua:17: Initializing sensor zone: 1
[INFO   1616096030.108216,   161980] /lfs/log.lua:92: I (12830) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[INFO   1616096030.127507,   162344] /lfs/application.lua:17: Initializing sensor zone: 2
[INFO   1616096030.146009,   161980] /lfs/log.lua:92: I (12870) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[INFO   1616096030.164845,   162344] /lfs/application.lua:17: Initializing sensor zone: 3
[INFO   1616096030.182925,   161976] /lfs/log.lua:92: I (12910) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[INFO   1616096030.201792,   162344] /lfs/application.lua:17: Initializing sensor zone: 4
[INFO   1616096030.223188,   161976] /lfs/log.lua:92: I (12950) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[INFO   1616096030.247326,   162344] /lfs/application.lua:17: Initializing sensor zone: 5
[INFO   1616096030.265648,   161976] /lfs/log.lua:92: I (12990) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[INFO   1616096030.284518,   162344] /lfs/application.lua:17: Initializing sensor zone: 7
[INFO   1616096030.305640,   161976] /lfs/log.lua:92: I (13030) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[INFO   1616096030.324621,   162344] /lfs/application.lua:17: Initializing sensor zone: 8
[INFO   1616096030.342989,   161976] /lfs/log.lua:92: I (13070) gpio: GPIO[32]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[INFO   1616096030.361846,   162344] /lfs/application.lua:17: Initializing sensor zone: 9
[INFO   1616096030.379955,   161976] /lfs/log.lua:92: I (13100) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[INFO   1616096030.404816,   162300] /lfs/application.lua:24: Initializing actuator Zone: out1 Pin: 23 Trigger: 1 Initial state: -0
[INFO   1616096030.429319,   162048] /lfs/log.lua:92: I (13150) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
[INFO   1616096030.454799,   162280] /lfs/application.lua:24: Initializing actuator Zone: alarm2_out2 Pin: 5 Trigger: 1 Initial state: -0
[INFO   1616096030.474279,   162036] /lfs/log.lua:92: I (13200) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
[INFO   1616096030.498544,   163172] /lfs/network.lua:32: Loaded:  application
[INFO   1616096030.561288,   162912] /lfs/events.lua:17: subscribe to EVENT_STOP
[INFO   1616096030.577729,   162852] /lfs/rest_endpoint.lua:135: REST Endpoint: http://10.0.0.211:8123/api/konnected
[INFO   1616096031.073288,   158236] /lfs/rest_endpoint.lua:14: HTTP Call: 200 state -0
[INFO   1616096031.095704,   158392] /lfs/rest_endpoint.lua:79: Updated actuator Zone: out1 Pin: 23 Trigger: 1 Initial state: -0
[INFO   1616096031.376899,   158152] /lfs/rest_endpoint.lua:14: HTTP Call: 200 state -0
[INFO   1616096031.398407,   158304] /lfs/rest_endpoint.lua:79: Updated actuator Zone: alarm2_out2 Pin: 5 Trigger: 1 Initial state: -0
[INFO   1616096031.712450,   157844] /lfs/rest_endpoint.lua:14: HTTP Call: 200 state 1 zone 1
[INFO   1616096031.996461,   157856] /lfs/rest_endpoint.lua:14: HTTP Call: 200 state 1 zone 2
[ERROR  1616096042.213724,   150220] /lfs/rest_endpoint.lua:29: Timeout connecting to http://10.0.0.211:8123/api/konnected
[INFO   1616096042.229100,   150260] /lfs/events.lua:26: emitting event:EVENT_STOP
[INFO   1616096042.243714,   150296] /lfs/rest_endpoint.lua:129: halting rest endpoint updates
[ERROR  1616096042.259413,   150296] /lfs/rest_endpoint.lua:33: Rebooting now (failsafe) ...
[INFO   1616096044.524849,   150188] /lfs/log.lua:94: I (27250) wifi:
[INFO   1616096044.540727,   150180] /lfs/log.lua:94: state: run -> init (0)
[INFO   1616096044.555614,   150228] /lfs/log.lua:94:
[INFO   1616096044.572481,   150060] /lfs/log.lua:94: I (27290) wifi:
[INFO   1616096044.588700,   149940] /lfs/log.lua:94: pm stop, total sleep time: 18749037 us / 21008518 us
[INFO   1616096044.603553,   150104] /lfs/log.lua:94:
[INFO   1616096044.619982,   150564] /lfs/log.lua:94: I (27340) wifi:
[INFO   1616096044.635593,   150520] /lfs/log.lua:94: new:<11,0>, old:<11,2>, ap:<255,255>, sta:<11,2>, prof:11
[INFO   1616096044.650427,   150600] /lfs/log.lua:94:
[ERROR  1616096044.675216,   155096] /lfs/log.lua:92: E (27390) TRANS_TCP: tcp_poll_read select error 113, errno = Software caused connection abort, fd = 52
[... cut ...]
The vendor confirmed it's because of this issue and they have faith Espressif will eventually fix it.

I do not have faith, nobody should have to have faith, it's a travesty and a disgrace that an issue that can render a device inoperable could be untouched for almost 2 years. Please fix this problem, I'm not the only one affected by this and it's extremely infuriating that the vendor tries everything they can to mitigate but it fails because Espressif can't be bothered. Disgusting, I am beyond angry.

KaeLL, 2021-03-18:
@user897943 are you sure [your issue](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#illegal-instr-msg) is even related to mine?

user897943, 2021-03-18:
The vendor quoted this particular issue directly using the url so I'm going
by what they're saying.

igrr, 2021-03-18:
@user897943 Sorry that you are seeing this issue!
As @KaeLL points out, your issue seems to be unrelated to the original issue report here. IllegalInstruction is a different exception type than DoubleException, and the documentation page linked by @KaeLL describes the situations when it occurs. I am sorry but I don't understand why the vendor of your device thinks this issue is related.

From the crash dump, it looks like the a0 register value, which normally contains the return address of the function, is invalid. The instruction at PC is `00f01d22`, which is a `retw` instruction. So the crash occurs upon a return from a function, when the return address is corrupted.
This might be a but in ESP-IDF, this might be a bug in the application. Without having more details, I'm afraid this issue isn't actionable from our side.

user897943, 2021-03-20:
@igrr 
Thanks for responding. I emailed the vendor letting them know your response, hopefully they'll get in touch and this issue can be resolved. I'm having this issue on a daily basis. I have to restart my server every day for a particular task and every time it goes offline it renders the board offline until a hard reset. As it's a security system it's simply not acceptable whichever side is responsible. Now I understand the ESP board is generic however if the bug is applicable to the board then Espressif is responsible to fix it. From what I've been told it's an issue with the ESP-IDF so Espressif would have to deal with it.

chegewara, 2021-03-20:
@user897943 You are using language that is insulting people from espressif, even if your issue is completely not related to this one, and honestly i dont expect you will apology.

Here is link that shows the part of issue from OP is caused by user code (wrong certificate)
https://github.com/ARMmbed/mbedtls/issues/139

Also based on my experience and very few logs, i can guessing the double exception is related to low heap/RAM. Problems with ssl + BT in logs and double exception, which may occur when crash dump is being printed, but due to lack of heap it will lead to another exception. I saw it one or two times already, not in my projects thou.

user897943, 2021-03-20:
> @user897943 You are using language that is insulting people from espressif, even if your issue is completely not related to this one, and honestly i dont expect you will apology.
> 
> Here is link that shows the part of issue from OP is caused by user code (wrong certificate)
> [ARMmbed/mbedtls#139](https://github.com/ARMmbed/mbedtls/issues/139)
> 
> Also based on my experience and very few logs, i can guessing the double exception is related to low heap/RAM. Problems with ssl + BT in logs and double exception, which may occur when crash dump is being printed, but due to lack of heap it will lead to another exception. I saw it one or two times already, not in my projects thou.

No, that's not true. Espressif is a business and they employ and pay professionals to develop and maintain these products, so what exactly you want an apology for? I made no personal attack against anybody that is a fact. My beef is with the company not managing the situation. It is a disgrace when a bug is unfixed for 2 years while customers are suffering daily problems, and because of that I am angry, it is affecting me and not just me might I remind you. So where is the apology to the customers? Apologize for complaining about broken stuff? No! But when the issue is fixed then I will most certainly say thank you for that and appreciate it.

chegewara, 2021-03-20:
> It is a disgrace when a bug is unfixed for 2 years while customers are suffering daily problems

Is your bug related to this one? NO. Do you suffer due to not solved bug for 2 years? I doubt.
Did you read issue template at all? I dont think so
```
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)
```

Your complain is like calling to doctor and saying, hey doctor, i am sick, heal me.

From your logs i can say you are having issue with TCP, not the double exception
`(27390) TRANS_TCP: tcp_poll_read select error 113, errno = Software caused connection abort, fd = 52`
Then how anyone can help you with your issue if we not even know which esp-idf version you are working with?
You are just trying to hijack someones else issue and spam it with useless info.

KaeLL, 2021-03-20:
@chegewara He may have a point though. Now I think I understand what @user897943 is talking about.
The issue with DoubleException is the fact that __it completely bricks the ESP32 to a state that it can't reset itself even with panic restart enabled__. That's the problem, that the board needs a manual power cycle, which is a huge deal given the implications.
@chegewara you're also missing the fact that DoubleException is the exception thrown against the first exception, which is thrown by the CPU and/or IDF, making it hardly related to any app code. The first exception is indeed related to the app, and generally is our fault, but the second one, that happens while handling of the first one isn't, which is why filling up the bug report form won't be of much help. But I could be wrong, it could help, and I would've submitted all info had I had this issue in a non-NDA protected codebase.
The fact of the matter is this issue of DoubleException bricking the ESP32 and it not resetting itself is in fact lurking within IDF and it will continue to be so because of how rare and elusive it is. So far, I've seen only me and #6519 have that issue. The fact that it's a deal-breaking bug that it's hard to get people from Espressif to prioritize it only adds to the irony.

chegewara, 2021-03-20:
@KaeLL I am not saying this issue is less important or something, but using such language, in topic which is not an inch close to his, is not accepted in cultural community. It is hard to fix something when you dont know what is broken, but there is not much data to diagnose it. Like i said, double exception is something that may happen when there is crash in code, and after crash when core dump is generated and/or printed another exception occur due to memory shortage or memory corruption (caused by esp-idf or client code). 
https://github.com/espressif/arduino-esp32/issues/3280
https://github.com/espressif/arduino-esp32/issues/2621
https://github.com/me-no-dev/ESPAsyncWebServer/issues/519

Please check this. He instead of creating issue and filling issue template is trolling other users issues:
https://github.com/espressif/esp-idf/issues/5227#issuecomment-802347632

KaeLL, 2021-03-20:
> but using such language, in topic which is not an inch close to his, is not accepted in cultural community.
Please check this. He instead of creating issue and filling issue template is trolling other users issues:
#5227 (comment)

Tbh, I couldn't care less about it. I care only about the problem.

> due to memory shortage or memory corruption (caused by esp-idf or client code).

Do you have any sources on this? Because I've looked into DoubeException (not on IDF as I don't have the expertise enough to understand the inner-workings of the chip and the assembly), and only got what @igrr said above 1.5 years ago. Or are you just taking Ibernstone's word for it? If you do have more information regarding this exception, please share it, as I'd love to know more about it, and perhaps we might even make this issue useful.

chegewara, 2021-03-20:
I remember that me-no-dev mentioned that once on gitter too, but i never got this issue personally, i think.
If i only could help with that issue i would do it for sure, like with many other issues i do.

user897943, 2021-03-21:
> @KaeLL I am not saying this issue is less important or something, but using such language, in topic which is not an inch close to his, is not accepted in cultural community. It is hard to fix something when you dont know what is broken, but there is not much data to diagnose it. Like i said, double exception is something that may happen when there is crash in code, and after crash when core dump is generated and/or printed another exception occur due to memory shortage or memory corruption (caused by esp-idf or client code).
> [espressif/arduino-esp32#3280](https://github.com/espressif/arduino-esp32/issues/3280)
> [espressif/arduino-esp32#2621](https://github.com/espressif/arduino-esp32/issues/2621)
> [me-no-dev/ESPAsyncWebServer#519](https://github.com/me-no-dev/ESPAsyncWebServer/issues/519)
> 
> Please check this. He instead of creating issue and filling issue template is trolling other users issues:
> [#5227 (comment)](https://github.com/espressif/esp-idf/issues/5227#issuecomment-802347632)
> 
> EDIT i can find few more double exception issues, but not much info to narrow it down and usually it is user error
> 
> [espressif/arduino-esp32#2686 (comment)](https://github.com/espressif/arduino-esp32/issues/2686#issuecomment-484912909)

I understand you have good intentions at heart but what you are doing is not acceptable. There is no "rules" on the front page on what is or isn't allowed to say, so I will say it how I want. This is not your forum to decide what rules are acceptable. What is generally accepted anywhere is that as long as you do not use profanity it's ok. Until I am using profanity you can feel hurt all you want, that's your choice. I never used profanity anywhere so you're going to have to learn to toughen up and deal with it because I am not changing for you and go ahead with that thumbs down if you want, unlike you I don't care about it! I am a genuine customer who came here because a company sold me a security product that uses ESP32 and I clearly said THEY told me THIS ISSUE is causing the problem but clearly you either missed that part or ignored it. They gave me the link to this issue. So I am here to complain and rightly so. Espressif isn't a charity they are getting paid for these products and it doesn't matter how many years it has affected me, it could be a nanosecond, it's still been unresolved for 2 years and that is absolutely disgraceful. Espressif has to fix this, period. Every day I have to open my equipment room and hard-reset the ESP32 because of this issue. E-V-E-R-Y S-I-N-G-L-E D-A-Y. I am not asking you to fix it, I am asking Espressif to fix it. So unless you can fix it or if you want to come to my house at 3AM every day to push the button then please don't tell me how to speak!

igrr, 2021-03-21:
@user897943 I'm sorry I didn't understand the core of your issue report first — I understand now that the chip doesn't restart after the panic occurs, and `ELF file SHA256` is the last thing being printed. This is indeed a serious issue, although quite different than the one reported originally in this ticket by @KaeLL.

I believe this specific issue (lock-up when printing ELF file SHA256) has been fixed in commit https://github.com/espressif/esp-idf/commit/fc03161f708ae289c6e91f02296ccc91ce663498. The fix was done in release v4.2, and was backported to release/v4.1, release/v4.0, release/v3.3 i.e. all release currently maintained release branches.

user897943, 2021-03-21:
@igrr thanks for your response. I reported here because the company (Konnected.io) said this was the specific issue they're having to try to mitigate. They gave me a new firmware which didn't fix and unfortunately now it has actually made the issue worse as only today I had to pull power from the board three times. I will pass on your response to the owner of that business and I hope they can find a solution but if they can liaise with Espressif directly then that would be better. I'm just the kind of person that doesn't like to sit around and wait, if I knew anything about coding I'd fix this myself. Unfortunately I don't... So my only option now is to chase those responsible for a fix and I hope either Konnected or Espressif fixes this major issue.

igrr, 2021-03-21:
Based on the info so far, I suppose your product is the Konnekted Alarm Panel based on the ESP32. According to the author, the software [is based on the dev-esp32 branch of nodemcu](https://github.com/konnected-io/konnected-security/issues/79#issuecomment-548031172), a Lua interpreter. At the moment, dev-esp32 branch of nodemcu is built using [ESP-IDF 3.3.2](https://github.com/nodemcu/nodemcu-firmware/commits/dev-esp32/sdk). The fix I have mentioned has been merged into the 3.3 branch in commit https://github.com/espressif/esp-idf/commit/2705b76371cf78f11e78ea1d0c4b735a820315e4, and [released in v3.3.3](https://github.com/espressif/esp-idf/releases/tag/v3.3.3). It should be sufficient to update the esp-idf submodule pointer to v3.3.3 or v3.3.4 and re-build the application. Paging @heythisisnate — please let me know if this helps.

user897943, 2021-03-21:
> Based on the info so far, I suppose your product is the Konnekted Alarm Panel based on the ESP32. According to the author, the software [is based on the dev-esp32 branch of nodemcu](https://github.com/konnected-io/konnected-security/issues/79#issuecomment-548031172), a Lua interpreter. At the moment, dev-esp32 branch of nodemcu is built using [ESP-IDF 3.3.2](https://github.com/nodemcu/nodemcu-firmware/commits/dev-esp32/sdk). The fix I have mentioned has been merged into the 3.3 branch in commit [2705b76](https://github.com/espressif/esp-idf/commit/2705b76371cf78f11e78ea1d0c4b735a820315e4), and [released in v3.3.3](https://github.com/espressif/esp-idf/releases/tag/v3.3.3). It should be sufficient to update the esp-idf submodule pointer to v3.3.3 or v3.3.4 and re-build the application. Paging @heythisisnate — please let me know if this helps.

@igrr thanks, this is great! Yes it's the Konnected Alarm Panel Pro. Hopefully Nate can pick this up and see if it can be resolved. I'll send him an email now to update. I appreciate that you went above the expectations and actually researched the product and tagged the owner.

heythisisnate, 2021-03-23:
@igrr thanks for the ping and the insight. Very helpful! I hope that this fix will solve the lockup problem that some Konnected users are having.
I've bumped the nodemcu firmware to use ESP-IDF 3.3.4 and rebuilt the Konnected application. I'm communicating offline with @user897943 for feedback. 🙌 🙌

user897943, 2021-03-24:
@heythisisnate @igrr Thanks both. The issue appears to be resolved. No issues with my panel since writing the firmware that Nate sent me via email. I will continue to keep an eye and see how it performs in the next few weeks. In the meantime thanks to the helpful responses and Nate thanks for your work on this.

marcelstoer, 2021-03-24:
@igrr 🥇 absolutely fantastic customer service! You went above and beyond in your effort to get to the bottom of this. Even though this does not affect me personally I want to thank you as an ESP8266/ESP32/NodeMCU community member.

Alvin1Zhang, 2021-04-12:
Thanks for reporting, feel free to reopen.

closed 2021-04-12 by Alvin1Zhang

KaeLL, 2021-04-12:
> Also we will modify the DoubleException vector to cause an RTC watchdog reset instead of entering the panic handler. This will provide less information but will be a better safeguard against the device being locked up.

Was this implemented?