espressif/esp-idf #3811: When i work in 3.1 branch，WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu

xuhongv, 2019-07-23:
When i work in 3.1 branch in windows10 , then error message : 

`WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
`

But i checkout in master branch in windows10 , is normal to build ...

xuhongv, 2019-07-23:
[image]

xuhongv, 2019-08-19:
[image]

mohammedkw11, 2020-03-21:
hi @xuhongv , did you find the solution? I'm facing the same problem now.

igrr, 2020-04-02:
@mohammedkw11 @xuhongv Please refer to the getting started guides for the IDF version you are using. According to the screenshot, the compiler version is 8.2 (release esp32-2019r1), which is used in IDF v4.0 and later. For IDF versions 3.x, the expected compiler version is 5.2. Please check this page for setting up the correct compiler version: https://docs.espressif.com/projects/esp-idf/en/v3.3.1/get-started/windows-setup.html.

closed 2020-04-02 by igrr