espressif/esp-idf #3812: How do i connect another BLE Device by mac address

bestony, 2019-07-23:
## Environment

- Development Kit:  ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC):  v4
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:       platformio
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     macOS
- Power Supply:         USB

## Problem Description

How Do i connect another BLE Device By Device Mac Address .

Now, i know other BLE Device MAC Address ,and i want to use mac address direct connect, instead of use discovery to connect ( it may be need more power)

### Expected Behavior

have an api can conncet another device by address 

### Actual Behavior

I couldn't find the api

gengyuchao, 2019-07-23:
Their have a connection API such as "esp_err_t esp_ble_gattc_open(esp_gatt_if_t gattc_if, esp_bd_addr_t remote_bda, esp_ble_addr_type_t remote_addr_type, bool is_direct);" in esp-idf ,but I'm not sure whether  this API in the platformio too. The "remote_bda" is the mac address.

bestony, 2019-07-24:
@gengyuchao i have this api, but in platform io, i use Ardunio Code as a BLE Server and Client, I didn't have the instance of `esp_gatt_if_t`

chegewara, 2019-07-24:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEClient.h#L37
and here:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEAddress.h#L23

If you want to get `esp_gatt_if_t` you can use one of those:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEDevice.h#L27-L29

bestony, 2019-07-24:
thanks @chegewara

closed 2019-07-24 by bestony

Mohannadhaddad, 2023-10-04:
Hi @bestony 

can you help me and share a simple code that connects to BLE server MAC address instead of the UUID please?
I am struggling with the library and not able to establish the connection