espressif/esp-idf #3813: Automatic restart problem. ASSERT_PARAM(1024 0), in rwble.c at line 230.

MrHezhisheng, 2019-07-23:
I'm having trouble rebooting. Here's the print log:

ASSERT_PARAM(1024 0), in rwble.c at line 230
ets Jun 8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:720
ho 0 tail 12 room 4
load:0x40078000,len:6940
load:0x40080000,len:5812
entry 0x40080e08
eFuse Two Point: NOT supported
firmware version is:25.01.005
build time: Jul 22 2019, 17:08:32
篚�\0�廊趔?\0\0\0鬗icroPython 7adfb6c2-dirty on 2019-07-22; ESP32 module with ESP32
Type "help()" for more information.

Can you give me some advice?

gengyuchao, 2019-08-01:
Do you send packets at very small intervals? Maybe you can try to increase the time interval.

Alvin1Zhang, 2019-08-23:
@MrHezhisheng Thanks for reporting this. Would you please help share if any updates for this issue? Thanks.

Alvin1Zhang, 2019-12-19:
Thanks for reporting this. Would you please help share if any updates for this issue? Thanks. @MrHezhisheng

Alvin1Zhang, 2020-02-27:
@MrHezhisheng Thanks for reporting. Will close now due to short of feedback, feel free to reopen if more details will be provided. Thanks.

closed 2020-02-27 by Alvin1Zhang