espressif/esp-idf #3815: "example_ble_sec_gatts_demo" examples central device connect with pairing successfully for most of device but only one central device facing pairing and connection failed

sagarvegg, 2019-07-23:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

sagarvegg, 2019-07-23:
**Environment**

- Development Kit: [ESP32-Wrover-Kit]

- Kit version (for WroverKit):[v4]

- Module or chip used: [ESP32-WROVER-B]

- IDF version (run git describe --tags to find it): v4.0-dev-478-gdf61612f8-dirty

- Build System: [Make]

- Compiler version (run xtensa-esp32-elf-gcc --version to find it): (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0

- Operating System: [Windows]

- Power Supply: [external 5V]

**Problem Description**

I try to run esp-idf "example_ble_sec_gatts_demo" example with pairing/bonding work for several central devices mobile phone but only one central device refuse to connect by pairing...

**Expected Behavior**
bt_bta_dm: BTM_LE_COMPLT_EVT
BT_SMP: result state = SMP_STATE_IDLE
BT_SMP: result state = SMP_STATE_IDLE
SEC_GATTS_DEMO: key type = ESP_LE_KEY_LENC
SEC_GATTS_DEMO: key type = ESP_LE_KEY_LID
SEC_GATTS_DEMO: key type = ESP_LE_KEY_PENC
SEC_GATTS_DEMO: key type = ESP_LE_KEY_PID
SEC_GATTS_DEMO: remote BD_ADDR: dc912192a457
SEC_GATTS_DEMO: address type = 1
SEC_GATTS_DEMO: **pair status = success**
SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_BOND
SEC_GATTS_DEMO: Bonded devices number : 2
SEC_GATTS_DEMO: Bonded devices list : 2
SEC_GATTS_DEMO: dc 91 21 92 a4 57
SEC_GATTS_DEMO: 5f 2f 6d 8d ef d2

**Actual Behavior**

bt_bta_dm: BTM_LE_COMPLT_EVT
BT_SMP: result state = SMP_STATE_IDLE
BT_BTM: Device not found
SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x1f
SEC_GATTS_DEMO: remote BD_ADDR: 6ee455b2a853
SEC_GATTS_DEMO: address type = 1
SEC_GATTS_DEMO: **pair status = fail**
SEC_GATTS_DEMO: fail reason = 0x66
SEC_GATTS_DEMO: Bonded devices number : 2
SEC_GATTS_DEMO: Bonded devices list : 2
SEC_GATTS_DEMO: dc 91 21 92 a4 57
SEC_GATTS_DEMO: 5f 2f 6d 8d ef d2
SEC_GATTS_DEMO: advertising start success

**Steps to repropduce**

hare different between successfully pairing device and connected with fail to pair and connect device

[diff_file.zip](https://github.com/espressif/esp-idf/files/3421905/diff_file.zip)

 **Debug Logs**

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
[... 488 lines cut ...]

E (68940) BT_BTM: Device not found

I (68940) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (68950) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
I (68970) SEC_GATTS_DEMO: advertising start success
D (71610) BT_L2CAP: l2c_ble_link_adjust_allocation  num_hipri: 0  num_lowpri: 1  low_quota:
10  round_robin_quota: 0  qq: 10
D (71610) BT_L2CAP: l2c_ble_link_adjust_allocation LCB 0   Priority: 0  XmitQuota: 10
D (71620) BT_L2CAP:         SentNotAcked: 0  RRUnacked: 0
D (71630) BT_L2CAP: l2cu_allocate_ccb: cid 0x0000
D (71630) BT_L2CAP: l2c_link_adjust_chnl_allocation
D (71640) BT_L2CAP: CID:0x0042 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Quota:200
D (71670) BT_L2CAP: l2cu_process_fixed_chnl_resp
D (71670) BT_SMP: SMDBG l2c smp_connect_callback

D (71670) BT_L2CAP: L2CAP - rcv_cid CID: 0x0004

I (71670) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: dc912192a457
D (71680) BT_L2CAP: l2c_link_check_send_pkts
D (71680) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (71690) BT_L2CAP: l2c_link_send_to_lower
D (71700) BT_L2CAP: TotalWin=9,Hndl=0x0,Quota=10,Unack=1,RRQuota=0,RRUnack=0
I (71700) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (71710) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT

sagarvegg, 2019-07-23:
I used "nRF Connect APP" for connect ble "example_ble_sec_gatts_demo" with all idf version but all have same issue only one device face problem to connect with pairing...
And all other device connect without any issue; so what should be problem for failing to connect

gengyuchao, 2019-07-23:
How many devices did you try to connect? As far as I know, there are a maximum number of BLE connections.

sagarvegg, 2019-07-23:
@gengyuchao I know that maximum 9 number of device able to connect with esp32 multi server but that was no issue I connect one by one device connect and disconnect that means at a single instance of time only single device try to connect with pairing...

I try to connect several device one by one but only single device have face issue of ble paring failure...
[failure device: LG W30 mobile --> nRF connect app]

gengyuchao, 2019-07-23:
Is the same device always in question? If so, it may be a problem with the device itself. And as far as I know, the maximum number of connections for ESP32 Bluetooth is 7.

sagarvegg, 2019-07-24:
> Is the same device always in question? If so, it may be a problem with the device itself.
same device means what you actually tell me that all device os or ble version same?
if yes than four device with 4.0 Bluetooth version and  two device with 4.2 Bluetooth version 

and issue with only one device connection which have Bluetooth version 4.2

chegewara, 2019-07-24:
I have issue with my samsung S9+ smartphone. 
Some time ago i was testing esp32 with this smartphone, by running BLE server in nRF connect and i could connect to it, most the time.

Few weeks ago i updated android to v9.0 and since then i cant connect esp32 to samsung server(not even single time), but i still can connect from smartphone to esp32.
When i am trying to connect esp32 to smartphone i am getting GATT error 0x85.

sagarvegg, 2019-07-24:
@chegewara I have oppo f11 pro device and which have already android version 9.0 and Bluetooth version 4.2; every time I connect to this device successfully without any issue than how you say that problem related to android version?

chegewara, 2019-07-24:
Im not saying its related to android version, it is related to device or device driver. It was working on my samsung before, but after update something has been changed in it and is no longer connecting.

sagarvegg, 2019-07-24:
Okay, @chegewara than how to again connect ble device for samsung S9+ smartphone? Issue solved or not for that device and if solved than how?

chegewara, 2019-07-24:
It is not solved for S9+, im no longer using it to test esp32 and instead using other esp32, which also from time to time has the same problems with connectivity (error 0x85).
Im not saying your problem is not an issue, just wanted to share the info im having similar issues with connectivity, even without pairing, which may or may not be related to peer device not only esp32.

Alvin1Zhang, 2020-04-26:
@sagarvegg Thanks for reporting. Sorry we don't have enough details to move forward, and we have tried but failed to reproduce the issue. Feel free to reopen the issue. Thanks.

closed 2020-04-26 by Alvin1Zhang