espressif/esp-idf #3816: [BLE Mesh] How to turn a node into a provisioner ?

Quiscale, 2019-07-23:
## Environment

- Development Kit : [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-LyraT]
- Kit version : [v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROVER]
- IDF version : v4.0-dev-1191-g138c941fa
- Build System:         [Make]
- Compiler version: 8.2.0
- Operating System:     [Windows]
- Power Supply:         [USB|external 5V]

## Problem Description

I am currently trying to use the BLE Mesh API to create a game, and I will have until 16 devices to connect, each device has one button and one RGB led. I want the devices to start as node(unprovisioned) and if push the button of one, it will become a provisioner and provision the 15 devices left.
So the problems is :
- When I want to use both mode in the same program, a node can start properly, but not a provisioner. And I still don't know how to do that after some days of research.

### Expected Behavior

```
I (0) cpu_start: App cpu up.
I (467) heap_init: Initializing. RAM available for dynamic allocation:
I (473) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (479) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (485) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (492) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (498) heap_init: At 3FFCBF48 len 000140B8 (80 KiB): DRAM
I (504) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (510) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (517) heap_init: At 40092124 len 0000DEDC (55 KiB): IRAM
I (523) cpu_start: Pro cpu start user code
I (541) spi_flash: detected chip: generic
I (542) spi_flash: flash io: dio
I (542) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (550) ble_mesh_provisioner: Initializing...
I (590) BTDM_INIT: BT controller compile version [4d3bde7]
I (590) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (690) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (970) ble_mesh_provisioner: ESP_BLE_MESH_PROVISIONER_SET_DEV_UUID_MATCH_COMP_EVT, err_code 0
I (1060) ble_mesh_provisioner: ESP_BLE_MESH_PROVISIONER_PROV_ENABLE_COMP_EVT, err_code 0
I (1060) ble_mesh_provisioner: ESP_BLE_MESH_PROVISIONER_ADD_LOCAL_APP_KEY_COMP_EVT, err_code 0
I (1070) ble_mesh_provisioner: ESP_BLE_MESH_PROVISIONER_BIND_APP_KEY_TO_MODEL_COMP_EVT, err_code 0
I (1080) ble_mesh_provisioner: BLE Mesh Provisioner initialized
```

### Actual Behavior

```
I (0) cpu_start: App cpu up.
I (478) heap_init: Initializing. RAM available for dynamic allocation:
I (485) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (491) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (497) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (503) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (509) heap_init: At 3FFCC518 len 00013AE8 (78 KiB): DRAM
I (516) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (522) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (528) heap_init: At 40092124 len 0000DEDC (55 KiB): IRAM
I (534) cpu_start: Pro cpu start user code
I (553) spi_flash: detected chip: generic
I (553) spi_flash: flash io: dio
I (553) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (562) ble_mesh_provisioner: Initializing...
I (602) BTDM_INIT: BT controller compile version [4d3bde7]
I (602) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (702) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (972) ble_mesh_provisioner: ESP_BLE_MESH_PROVISIONER_SET_DEV_UUID_MATCH_COMP_EVT, err_code 0
ASSERTION FAIL [prov && prov->uuid] @ C:/msys32/home/cwx-eng/esp/esp-idf/components/bt/esp_ble_mesh/mesh_core/proxy.c:1387:
        bt_mesh_proxy_init, Device UUID is not initializedE (5562) task_wdt: Task watchdog got triggered. The following tasks did not
reset the watchdog in time:
E (5562) task_wdt:  - IDLE0 (CPU 0)
E (5562) task_wdt: Tasks currently running:    
E (5562) task_wdt: CPU 0: BTC_TASK
E (5562) task_wdt: CPU 1: IDLE1
E (10561) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10561) task_wdt:  - IDLE0 (CPU 0)
E (10561) task_wdt: Tasks currently running:
E (10561) task_wdt: CPU 0: BTC_TASK
E (10561) task_wdt: CPU 1: IDLE1
```
The program is still running after the error, but only reproduce the watchdog reset

### Steps to repropduce

1. Copy the ble_mesh_provisioner example, in $IDF_PATH\examples\bluetooth\esp_ble_mesh
2. run "make menuconfig" -> "Component config" -> "ESP BLE Mesh Support" -> Check "Support for BLE Mesh Node"
3. run "make -j4 flash monitor"
4. wait

### Question

- How can I turn a node(unprovisioned) into a provisioner after using esp_ble_mesh_init once ? Do I have to restart the ESP ?

WCCWCC, 2019-07-24:
Hi Quiscale，
     Note:    bt_mesh_proxy_init, Device UUID is not initialized
     
     The cause of the error is that the device's uuid is not initialized.

     You need to initialize the uuid variable in `esp_ble_mesh_prov_t` .

     You can refer to the `ble_mesh_node` demo.

Quiscale, 2019-07-24:
Hi WCCWCC,

Yes I found my error not a long time ago, I don't know why but to solve my problem I did as you said it :
That was my code :
```
static esp_ble_mesh_prov_t mesh_provision_prov = {
		.prov_uuid           = MESH_uuid,
		.prov_unicast_addr   = 0x0001,
		.prov_start_address  = 0x0005,
		.prov_attention      = 0x00,
		.prov_algorithm      = 0x00,
		.prov_pub_key_oob    = 0x00,
		.prov_static_oob_val = NULL,
		.prov_static_oob_len = 0x00,
		.flags               = 0x00,
		.iv_index            = 0x00,
};
```
And now It looks like :
```
static esp_ble_mesh_prov_t mesh_provision_prov = {
		.uuid = MESH_uuid,
		.output_size = 0,
		.output_actions = 0,

		.prov_uuid           = MESH_uuid,
		.prov_unicast_addr   = 0x0001,
		.prov_start_address  = 0x0005,
		.prov_attention      = 0x00,
		.prov_algorithm      = 0x00,
		.prov_pub_key_oob    = 0x00,
		.prov_static_oob_val = NULL,
		.prov_static_oob_len = 0x00,
		.flags               = 0x00,
		.iv_index            = 0x00,
};
```
And it works, so I need to register twice the uuid

closed 2019-07-24 by Quiscale