espressif/esp-idf #3817: Windows自带的命令窗口（命令提示符）搭建ESP32开发环境遇到的问题？？？

KuDianWanJia, 2019-07-23:
我在通过Windows自带的命令窗口（命令提示符）搭建ESP32开发环境时，在执行 `idf.py menuconfig`时出现以下错误：
`Checking Python dependencies...
Python requirements from D:\ESP32Windows\esp\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running ninja in directory d:\esp32windows\esp\myproject\blink\build
Executing "ninja menuconfig"...
[0/1] cmd.exe /C "cd /D D:\ESP32Windows\esp\M...:/ESP32Windows/esp/MyProject/blink/sdkconfig"
Loading defaults file D:/ESP32Windows/esp/MyProject/blink/sdkconfig.defaults...
D:/ESP32Windows/esp/esp-idf/components/bt/Kconfig:386: undefined variable "$IDF_PATH/components/bt/host/bluedroid/Kconfig.in"
FAILED: CMakeFiles/menuconfig
cmd.exe /C "cd /D D:\ESP32Windows\esp\MyProject\blink\build && C:\Python27\python.exe D:/ESP32Windows/esp/esp-idf/tools/kconfig_new/confgen.py --kconfig D:/ESP32Windows/esp/esp-idf/Kconfig --config D:/ESP32Windows/esp/MyProject/blink/sdkconfig --defaults D:/ESP32Windows/esp/MyProject/blink/sdkconfig.defaults --env-file D:/ESP32Windows/esp/MyProject/blink/build/config.env --env IDF_TARGET=esp32 --output config D:/ESP32Windows/esp/MyProject/blink/sdkconfig && "C:\Program Files\CMake\bin\cmake.exe" -E env "COMPONENT_KCONFIGS=D:/ESP32Windows/esp/esp-idf/components/app_trace/Kconfig D:/ESP32Windows/esp/esp-idf/components/bt/Kconfig D:/ESP32Windows/esp/esp-idf/components/driver/Kconfig D:/ESP32Windows/esp/esp-idf/components/efuse/Kconfig D:/ESP32Windows/esp/esp-idf/components/esp-tls/Kconfig D:/ESP32Windows/esp/esp-idf/components/esp32/Kconfig D:/ESP32Windows/esp/esp-idf/components/esp_adc_cal/Kconfig D:/ESP32Windows/esp/esp-idf/components/esp_common/Kconfig D:/ESP32Windows/esp/esp-idf/components/esp_eth/Kconfig D:/ESP32Windows/esp/esp-idf/components/esp_event/Kconfig D:/ESP32Windows/esp/esp-idf/components/esp_gdbstub/Kconfig D:/ESP32Windows/esp/esp-idf/components/esp_http_client/Kconfig D:/ESP32Windows/esp/esp-idf/components/esp_http_server/Kconfig D:/ESP32Windows/esp/esp-idf/components/esp_https_ota/Kconfig D:/ESP32Windows/esp/esp-idf/components/esp_https_server/Kconfig D:/ESP32Windows/esp/esp-idf/components/esp_wifi/Kconfig D:/ESP32Windows/esp/esp-idf/components/espcoredump/Kconfig D:/ESP32Windows/esp/esp-idf/components/fatfs/Kconfig D:/ESP32Windows/esp/esp-idf/components/freemodbus/Kconfig D:/ESP32Windows/esp/esp-idf/components/freertos/Kconfig D:/ESP32Windows/esp/esp-idf/components/heap/Kconfig D:/ESP32Windows/esp/esp-idf/components/libsodium/Kconfig D:/ESP32Windows/esp/esp-idf/components/log/Kconfig D:/ESP32Windows/esp/esp-idf/components/lwip/Kconfig D:/ESP32Windows/esp/esp-idf/components/mbedtls/Kconfig D:/ESP32Windows/esp/esp-idf/components/mdns/Kconfig D:/ESP32Windows/esp/esp-idf/components/mqtt/Kconfig D:/ESP32Windows/esp/esp-idf/components/newlib/Kconfig D:/ESP32Windows/esp/esp-idf/components/nvs_flash/Kconfig D:/ESP32Windows/esp/esp-idf/components/openssl/Kconfig D:/ESP32Windows/esp/esp-idf/components/pthread/Kconfig D:/ESP32Windows/esp/esp-idf/components/spi_flash/Kconfig D:/ESP32Windows/esp/esp-idf/components/spiffs/Kconfig D:/ESP32Windows/esp/esp-idf/components/tcpip_adapter/Kconfig D:/ESP32Windows/esp/esp-idf/components/unity/Kconfig D:/ESP32Windows/esp/esp-idf/components/vfs/Kconfig D:/ESP32Windows/esp/esp-idf/components/wear_levelling/Kconfig D:/ESP32Windows/esp/esp-idf/components/wifi_provisioning/Kconfig D:/ESP32Windows/esp/esp-idf/components/wpa_supplicant/Kconfig" "COMPONENT_KCONFIGS_PROJBUILD=D:/ESP32Windows/esp/esp-idf/components/app_update/Kconfig.projbuild D:/ESP32Windows/esp/esp-idf/components/bootloader/Kconfig.projbuild D:/ESP32Windows/esp/esp-idf/components/esptool_py/Kconfig.projbuild D:/ESP32Windows/esp/esp-idf/components/partition_table/Kconfig.projbuild D:/ESP32Windows/esp/MyProject/blink/main/Kconfig.projbuild" IDF_CMAKE=y KCONFIG_CONFIG=D:/ESP32Windows/esp/MyProject/blink/sdkconfig IDF_TARGET=esp32 "D:/ESP32Windows/ESP-IDF Tools/mconf-idf/mconf-idf.exe" D:/ESP32Windows/esp/esp-idf/Kconfig && C:\Python27\python.exe D:/ESP32Windows/esp/esp-idf/tools/kconfig_new/confgen.py --kconfig D:/ESP32Windows/esp/esp-idf/Kconfig --config D:/ESP32Windows/esp/MyProject/blink/sdkconfig --defaults D:/ESP32Windows/esp/MyProject/blink/sdkconfig.defaults --env-file D:/ESP32Windows/esp/MyProject/blink/build/config.env --env IDF_TARGET=esp32 --output config D:/ESP32Windows/esp/MyProject/blink/sdkconfig"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
`
问题中提示 `未定义变量："$IDF_PATH/components/bt/host/bluedroid/Kconfig.in"` ，我不知道该怎么解决。

而我跳过 `idf.py menuconfig` 步骤，直接运行 `idf.py build` ，结果编译成功了，并且之后也成功将程序flash到ESP32开发板里。

我希望找到解决方法，非常感谢！！！

Alvin1Zhang, 2019-07-24:
@KuDianWanJia Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

simonliu009, 2019-07-25:
建议切换idf版本后先运行 `git submodule update --init --recursive`试试看

Alvin1Zhang, 2019-08-01:
@KuDianWanJia Have you got any chance to try the approach suggested by @simonliu009 ? Would you help share if any updates for the issue? Thanks.

ShenTengTu, 2019-08-07:
> 建议切换idf版本后先运行 `git submodule update --init --recursive`试试看

I switch idf version to V3.3-beta3 (because MicroPython need) then `idf.py menuconfig` . I get error as below:

```
D:\Espressif\repository\esp\hello_world>idf.py menuconfig
Checking Python dependencies...
Python requirements from D:\Espressif\repository\esp\esp-idf\requirements.txt are satisfied.
Running ninja in directory D:\Espressif\repository\esp\hello_world\build
Executing "ninja menuconfig"...
[0/1] Re-running CMake...
-- Building for target esp32
CMake Error at D:/Espressif/repository/esp/hello_world/main/CMakeLists.txt:1 (idf_component_register):
  Unknown CMake command "idf_component_register".
Call Stack (most recent call first):
  D:/Espressif/repository/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:107 (include)
  D:/Espressif/repository/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:217 (expand_component_requirements)

CMake Error at D:/Espressif/repository/esp/esp-idf/CMakeLists.txt:39 (message):
  Failed to expand component requirements

-- Configuring incomplete, errors occurred!
See also "D:/Espressif/repository/esp/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "D:/Espressif/repository/esp/hello_world/build/CMakeFiles/CMakeError.log".
FAILED: build.ninja
D:\CMake\bin\cmake.exe -SD:\Espressif\repository\esp\hello_world -BD:\Espressif\repository\esp\hello_world\build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1
```
Which version is suggest to use?

---

It works if I replace `$IDF_PATH` to absolute path manually, in 3 lines of `esp-idf/components/bt/Kconfig` as below.
```
D:/Espressif/repository/esp/esp-idf/components/bt/Kconfig:386: undefined variable "$IDF_PATH/components/bt/host/bluedroid/Kconfig.in"

D:/Espressif/repository/esp/esp-idf/components/bt/Kconfig:391: undefined variable "$IDF_PATH/components/bt/host/nimble/Kconfig.in"

D:/Espressif/repository/esp/esp-idf/components/bt/Kconfig:404: undefined variable "$IDF_PATH/components/bt/esp_ble_mesh/Kconfig.in"
```
(If replace `$IDF_PATH` to `%IDF_PATH%`, it doesn't work .)

georgik, 2021-04-13:
@KuDianWanJia @ShenTengTu thank you for reporting the issue and we're sorry for the late response.

I see that mentioned IDF versions are already out of date. I made tests with the latest version of 3.3.5 and master. In both cases, all mentioned commands worked without a problem.
Please use our latest ESP-IDF Tools Windows Installer to deploy a fresh version of ESP-IDF.
You can download it from https://dl.espressif.com/dl/esp-idf/

Feel free to reopen the issue in case of error.

closed 2021-04-16 by Alvin1Zhang