espressif/esp-idf #3821: fatal error: wpa2/utils/base64.h: No such file or directory

manish5897, 2019-07-24:
## Environment
- Module or chip used:  [ESP32-WROOM-32]
- IDF version : v4.0-dev-1287-gd7e659df2
- Build System:         [Make]
- Compiler version : 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
The library wpa2/utils/base64.h is not found in the esp-idf. May be the library is depreciate. It shows compilation error. 

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "wpa2/utils/base64.h"

void app_main()
{
    
}

```

## Debug Logs

```
/home/manish/parkzap/esp32_led/main/WebSocket_Task.cpp:20:31: fatal error: wpa2/utils/base64.h: No such file or directory
compilation terminated.
/home/manish/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'WebSocket_Task.o' failed
make[1]: *** [WebSocket_Task.o] Error 1
/home/manish/esp/esp-idf/make/project.mk:584: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2

projectgus, 2019-07-24:
Hi @manish5897 ,

This header is an internal header of the WPA Supplicant component and it was available in previous IDF versions, but we've made it private in IDF V4.0 as it may be changed or removed in the future.

The recommended base64 API to use is the mbedTLS one (include as `#include "mbedtls/base64.h"`:
https://github.com/espressif/mbedtls/blob/development/include/mbedtls/base64.h#L36

closed 2019-07-24 by projectgus

manish5897, 2019-07-24:
## Problem Description
Add library mbedtls/base64.h, afterthat "base64_encode" is not found in this header. 

### Code to reproduce this issue

```cpp
#include "wpa2/utils/base64.h"

void app_main()
{

//hex to base64
p_buf = (char *)base64_encode((unsigned char *)p_SHA1_result, SHA1_RES_L, (size_t *)&i);
    
}

```

## Debug Logs

```
 base64_encode' was not declared in this scope
              SHA1_RES_L, (size_t *)&i);
                                      ^

projectgus, 2019-07-24:
@manish5897 Sorry, I didn't explain fully. This is a different API that also implements base64 encoding. If you read the header file in the previous link then you'll see the function name and parameters - they are similar but not 100% the same.

JulienDevillers, 2020-01-13:
Hello,
With a similar issue, I changed to #include "mbedtls/base64.h". It works well if the module is in "main" directory but if the module is in a component, the compiler raises "no such file or directory"...

What should I do?

pedros89, 2021-09-30:
Hello, I was following a tutorial. 
[https://www.youtube.com/watch?v=jm45imFsZTw](https://www.youtube.com/watch?v=jm45imFsZTw)
I have two computers with Arduino ESP32 core one with intalled v1.0.6 and the library works and the other  v2.0.0 where it does not work