espressif/esp-idf #3822: [1287] Memory corruption initializing WiFi, esp_wifi_init()

DrSegatron, 2019-07-24:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v4.0-dev-1287-gd7e659df2
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

Calling **esp_wifi_init()** with default cfg corrupts global uninitialized memory area.

### Expected Behavior

Not overwriting the values of my global variables.

### Actual Behavior

Zero-filling part of global uninitialized RAM area. (At least 24 bytes of my variables)

### Steps to repropduce

Call **esp_wifi_init()** with a structure filled by **WIFI_INIT_CONFIG_DEFAULT()**.

### Code to reproduce this issue

```cpp
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT(); //initialize structure with default values
memset(&ddmp_connection_status[6],0xff,24); //fill variable with canary values
ESP_LOG_BUFFER_HEXDUMP("6",&variable[6],24,ESP_LOG_ERROR); //all 0xff
esp_wifi_init(&cfg);
ESP_LOG_BUFFER_HEXDUMP("6",&variable[6],24,ESP_LOG_ERROR); //all 0x00
```

> CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=10
> CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=32
> # CONFIG_ESP32_WIFI_STATIC_TX_BUFFER is not set
> CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER=y
> CONFIG_ESP32_WIFI_TX_BUFFER_TYPE=1
> CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=32

## Debug Logs

Calling  **esp_wifi_init()**:
```
I (7162) wifi: wifi driver task: 3ffdfcb4, prio:23, stack:3584, core=0
I (7162) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (7172) efuse: coding scheme 0
D (7172) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (7182) efuse: coding scheme 0
D (7182) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (7192) efuse: coding scheme 0
D (7192) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (7202) efuse: coding scheme 0
D (7202) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (7212) efuse: coding scheme 0
D (7212) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (7222) efuse: coding scheme 0
D (7222) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (7232) efuse: coding scheme 0
D (7232) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
I (7242) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (7252) efuse: coding scheme 0
D (7252) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (7262) efuse: coding scheme 0
D (7262) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (7272) efuse: coding scheme 0
D (7272) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (7282) efuse: coding scheme 0
D (7282) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (7292) efuse: coding scheme 0
D (7292) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (7302) efuse: coding scheme 0
D (7302) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (7312) efuse: coding scheme 0
D (7312) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
I (7322) wifi: wifi firmware version: 3c39e82
I (7332) wifi: config NVS flash: disabled
I (7332) wifi: config nano formating: disabled
I (7332) wifi: Init dynamic tx buffer num: 32
I (7342) wifi: Init data frame dynamic rx buffer num: 32
I (7342) wifi: Init management frame dynamic rx buffer num: 32
I (7352) wifi: Init management short buffer num: 32
I (7352) wifi: Init static rx buffer size: 1600
I (7362) wifi: Init static rx buffer num: 10
I (7362) wifi: Init dynamic rx buffer num: 32
```
Memory corruption is detected here.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3426094/sdkconfig.txt)

liuzfesp, 2019-07-24:
Hi @DrSegatron, I just can't replicate your issue with $IDF_PATH/examples/wifi/getting_started/station and same sdkconfig (except the partition table configuration). Could you show me your demo code that can reproduce this issue?

DrSegatron, 2019-07-24:
You mean a complete source? It is part of a kinda large project, but I can try to whittle it down to something that I can publish, tomorrow.

liuzfesp, 2019-07-24:
No, I mean the source that can replicate the issue because I just can't replicate the issue with IDF examples.

DrSegatron, 2019-07-25:
Closing this issue as it seems to be corruption the other way around. In moving from Make to CMake, a compilation flag shared between the main component and another was only applied to the main component and not the other, making the component allocate a [1] space array, while being used as a [7] space array. So I was using the TX/RX buffer space location, which of course got overwritten.

Moral of the story: Put your target_compile_options() in each component needing them.

closed 2019-07-25 by DrSegatron