espressif/esp-idf #3823: ESP32 RMT Tx

astrorafael, 2019-07-24:
## Environment

- Development Kit:      ESP32-DevKitC clone
- Kit version (for WroverKit/PicoKit/DevKitC):
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-1287-gd7e659df2
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         USB

## Problem Description

RMT Tx driver allows Tx looping when the items are copied to the RMT RAM. And this mostly works except when the items plus End of Tx marker occupies exactly one 64 item block (or 2, 3 blocks, ...). After al 63 non zero items are transmitted, the RMT stops forever. It seems like a boundary issue when looping.

### Steps to reproduce

1. Attach an logic analyzer probe to GPIO 18 
2. compile and run the given source code.
3. See how after some waveforms, the GPIO pin stops oscilating.
4. Delete just one ```{{{ 32000, 1, 32000, 0 }}}``` item from the array
5. Compile & run again
6. See how the waveform is now continuosly generated 

### Code to reproduce this issue

The [following Gist](https://gist.github.com/astrorafael/5247a8ac1af414a39b905980e7267fb8) reproduces the problem. 

## Debug Logs

N/A

## Other items if possible

N/A

koobest, 2020-01-15:
Hi, @astrorafael 

Can you try the master branch?
thanks !!

Alvin1Zhang, 2020-01-19:
@astrorafael Thanks for reporting and very sorry for the turnaround, would you please help try the master branch and help share if any updates for the issue? Thanks.

astrorafael, 2020-01-21:
Sorry, I'm quite busy at work now. Will try your master branch change next week.

Alvin1Zhang, 2020-03-30:
@astrorafael Thanks for reporting the issue, did you get time to have a test? Thanks.

Alvin1Zhang, 2020-08-21:
@astrorafael Thanks for reporting the issue, did you get time to have a test? Would you please help share if any updates for the issue? Thanks.

astrorafael, 2020-08-27:
Hi Alvin1Zhang,
I've just come back from holidays but I still have a couple of days off, which I'll use to test this issue.

 It has been over a year since I opened the issue so I have to catch up a bit first. My environment is still not up to date, so I'll try to reproduce the problem with it. Then, I'll update the environment and test again. I'll come back as soon as I have passed the tests.

Alvin1Zhang, 2020-08-27:
@astrorafael Thanks for the updates, look forward to the testing results.

astrorafael, 2020-08-27:
Hi again
I confirm that the program supplied stopped sending pulses in my old environment.
I can confirm that the program supplied sends an infinite pulse train on GPIO#18 after environment update

* IDF version (run `git describe --tags` to find it): v4.3-dev-907-g6c17e3a64
* Compiler version (run `xtensa-esp32-elf-gcc --version` to find it):  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0

So the issue is now fixed. Sorry for the delay

Alvin1Zhang, 2020-08-28:
Thanks for sharing the updates.

closed 2020-08-28 by Alvin1Zhang