espressif/esp-idf #3824: RMT Tx Noticeable jitter

astrorafael, 2019-07-24:
-
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): 
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-1287-gd7e659df2
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
  xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

When the RMT Tx driver looping is enabled it introduces a noticeable delay between passes, much longer than the documented delay in the reference technical manual:

> **Note:** When enabling the continuous transmission mode by setting 
> RMT_REG_TX_CONTI_MODE, the transmitter will transmit the data on the channel 
> continuously, that is, from the first byte to the last one, then from the first to the last again, and
> so on. In this mode, there will be an idle level lasting one clk_div cycle between N and N+1 
> transmissions.

### Expected Behavior

When using ref clock 80 Mhz and 250 as a prescaler, according to the documentation, the delay should be around 250/80000000 secs (negligible).

### Actual Behavior

The shown delay is several orders of magnitude longer (205 ms in the example below), probably due to a long ISR.

### Steps to repropduce

1. Attach an logic analyzer probe to GPIO 18
2. compile and run the given source code.
3. Capture during a couple of minutes.

See how the timing is rock solid except when the driver loops and a 205ms delay is introduced in the high level.

### Code to reproduce this issue

The [following Gist](https://gist.github.com/astrorafael/c7c0fb5ec9fe98443b7ec786411f96c6) generates a 1 Hz square waveform by repeating items over 2 64-item block to minimize jitter when the driver loops. However, the driver introduces a noticeable 205 ms delay
between passes.

[image]

## Debug Logs
N/A

## Other items if possible
N/A

koobest, 2019-07-30:
Hi, @astrorafael
Seems another hardware issue. The first item was repeated twice.  Can you change the first item to ` {{{32767/2,1,32767/2,1}}}`?

```
rmt_item32_t items[] = {
    {{{32767/2,1,32767/2,1}}},
    .....
    {{{32767,0,28932,0}}}, 
    {{{0,0, 0, 0}}}
};
```
thanks !!

koobest, 2019-07-31:
Hi, @astrorafael
The cause has been found, when {0,0,0,0}(End mark) is sent, a tx_end interrupt will be generated. In the ISR, the conf1.mem_rd_rst register will be reset. This causes the first data to be sent twice. you can disable all the interrupt to solve your issue.
```
    fgen_init();
    rmt_clr_intr_enable_mask(~0);
    ESP_ERROR_CHECK(rmt_tx_start(RMT_CHANNEL_0, true));
```

thanks !!

negativekelvin, 2019-07-31:
`rmt_clr_intr_enable_mask(~0);`

That seems like the nuclear option, you probably only want to mask it on that channel with `rmt_set_tx_intr_en(channel, false)`.  However the driver probably shouldn't enable it in the first place if tx_conti_mode aka loop_en is set
https://github.com/espressif/esp-idf/blob/6fe853a2c73437f74c0e6e79f9b15db68b231d32/components/driver/rmt.c#L782

astrorafael, 2019-07-31:
Hi,

Thank you very much for the tip. As suggested by @negativekelvin  I've just added `rmt_set_tx_intr_en(channel, false)` and now the timing is rock solid all over the loop.

Now my fgen_init looks like:

```C
static void fgen_init()
{
    size_t N = sizeof(items) / sizeof(items[0]);;

    rmt_config_t config = {
        // Common config
        .channel              = RMT_CHANNEL_0,
        .rmt_mode             = RMT_MODE_TX,
        .gpio_num             = 18,
        .mem_block_num        = 1 + (N-1) / 64,
        .clk_div              = 250,
        // Tx only config
        .tx_config.loop_en    = true,
        .tx_config.carrier_en = false
    };

    ESP_ERROR_CHECK(rmt_config(&config));
    ESP_ERROR_CHECK(rmt_driver_install(config.channel, NO_RX_BUFFER, DEFAULT_ALLOC_FLAGS));
    ESP_ERROR_CHECK(rmt_fill_tx_items(config.channel, items, N, 0));
    ESP_ERROR_CHECK(rmt_set_tx_intr_en(config.channel, false));
}
```
I was wondering if this will be considered a s a bug and fixed in the driver or I'll have to carry this workaround as a standard procedure.

closed 2019-08-06 by astrorafael