espressif/esp-idf #3825: fatal error: eth_phy/phy_lan8720.h: No such file or directory

manish5897, 2019-07-24:
## Environment
- Module or chip used:  [ESP32-WROOM-32]
- IDF version : v4.0-dev-1287-gd7e659df2
- Build System:         [Make]
- Compiler version : 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
The library eth_phy/phy_lan8720.h is not found in the esp-idf. May be the library is depreciate. It shows compilation error. 

### Code to reproduce this issue

```cpp
#include "eth_phy/phy_lan8720.h"

void app_main()
{
    
}

```

## Debug Logs

```
/home/manish/esp32_led/main/ethernet.cpp:26:33: fatal error: eth_phy/phy_lan8720.h: No such file or directory
compilation terminated.
/home/manish/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'ethernet.o' failed
make[1]: *** [ethernet.o] Error 1
/home/manish/esp/esp-idf/make/project.mk:584: recipe for target 'component-main-build' failed

suda-morris, 2019-07-24:
@manish5897 Yes, Ethernet driver has refactored in v4.0. Now we have a new approach to install the Ethernet driver. https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/main/ethernet_example_main.c#L83

manish5897, 2019-07-24:
Let me know, the stable idf version in which #include "eth_phy/phy_lan8720.h" library is working fine.

suda-morris, 2019-07-25:
@manish5897 Thank you. If you are going to try the new `esp-eth` driver, feel free to let us know when you come into trouble. Or would you mind to close this issue since it's not a real bug in v4.0?

closed 2019-08-05 by manish5897

dvogureckiy99, 2022-10-10:
Hi, I don't understand, how can I solve this issue ?