espressif/esp-idf #3826: Download crosstool-NG and build it:

syswombat, 2019-07-24:
Hello
you have explained well so far 

Download crosstool-NG and build it:

but could you explain where to get it and install?

syswombat, 2019-07-24:
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/linux-setup-scratch.html

the Link
Follow instructions for standard setup to add the toolchain to your PATH.
are giving a error!

Alvin1Zhang, 2019-07-25:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks. @syswombat

syswombat, 2019-07-25:
•Hardware: Raspberry PI 4, 4GB RAM
•Operating System: [Linux Buster (Debain 10)]
•Power Supply: [USB|external 5V]

Basicly i wished to install the esp32 on the Arduino IDE (Board Manager) but there came a error

Tool xtensa-esp32-elf-gcc is not available for your operating system.
java.lang.RuntimeException: java.lang.Exception: Tool xtensa-esp32-elf-gcc is not available for your operating system.
 at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
 at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Tool xtensa-esp32-elf-gcc is not available for your operating system.
 at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:85)
 at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)

then i went into the Internet and went to 
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/linux-setup-scratch.html
followed the Instruction for Debian Linux

at this place 
calling up ./bootstrap && ./configure --enable-local && make install

i got a error:

checking for absolute path to patch... /usr/bin/patch
 checking for bash... /bin/bash
 checking for GNU bash >= 3.1... no
 configure: error: Required tool not found: GNU bash >= 3.1

and at this point there is a link error,
the Link
 Follow instructions for standard setup to add the toolchain to your PATH.
 are giving a error!

negativekelvin, 2019-07-25:
>Required tool not found: GNU bash >= 3.1

So install bash then

https://github.com/espressif/esp-idf/issues/589

syswombat, 2019-07-25:
@negativekelvin 

# bash --version
GNU bash, Version 5.0.3(1)-release (arm-unknown-linux-gnueabihf)

as you see 5.0 is bigger then 3.1 so it should be ok

negativekelvin, 2019-07-25:
https://bbs.archlinux.org/viewtopic.php?id=244333

syswombat, 2019-07-25:
;-) so the problem is well known, but not changed in the script.
and the information are from Februry - so i will have to find out where to change this - but bit confused

igrr, 2019-07-25:
You can apply the following patch: https://github.com/crosstool-ng/crosstool-ng/commit/b0151f5eb53c5bf70d39512a3d73e529be90bc0a

syswombat, 2019-07-27:
Hello Ivan

about my Problem
fixed as you said but then still some problems 
 
came to this part

Build the toolchain:

./ct-ng xtensa-esp32-elf
./ct-ng build
chmod -R u+w builds/xtensa-esp32-elf

:~/esp/crosstool-NG# ./ct-ng xtensa-esp32-elf
MKDIR config.gen
IN config.gen/arch.in
IN config.gen/kernel.in
IN config.gen/cc.in
IN config.gen/binutils.in
IN config.gen/libc.in
IN config.gen/debug.in
CONF config/config.in
#
# configuration written to .config
#

***********************************************************

Initially reported by: Ivan Grokhotkov <ivan@espressif.com>
URL:

Comment:
Based on xtensa-esp108-elf sample reported by Max Filippov <jcmvbkbc@gmail.com>

***********************************************************

WARNING! This sample may enable experimental features.
Please be sure to review the configuration prior
to building and using your toolchain!
Now, you have been warned!

***********************************************************

Now configured for "xtensa-esp32-elf"
root@RPi-04-IOT-conf:~/esp/crosstool-NG# ./ct-ng build
[ERROR] You must NOT be root to run crosstool-NG
[00:00] / make: *** [ct-ng:152: build] Fehler 1
=========================================================================
 
to do the installation im on raspberry pi with sudo -i
if i do not have to be root, i did a exit so without sudo.
but still the same error
 
so any adwise when the hole installation should run - smoth?
 
basicly i would like to be able to select the board on my arduino ide ;'-(
 
have a nice day
vinc

syswombat, 2019-07-29:
any help to fix it?

antmak, 2019-08-09:
https://github.com/espressif/crosstool-NG/commits/xtensa-1.22.x branch was updated, please try again @syswombat

antmak, 2019-08-09:
Ah, sorry @syswombat I've just noticed your last issue with building on RPi . 

Ok, maybe you can try an unofficial ARM build before you proceed ?
https://dl.espressif.com/dl/xtensa-esp32-elf-linux-armel-1.22.0-87-gb57bad3-5.2.0.tar.gz

Also you can take a look on a ci config https://github.com/espressif/crosstool-NG/blob/xtensa-1.22.x/.gitlab-ci.yml for build tips, maybe it give you some clues

Alvin1Zhang, 2019-10-11:
@syswombat Thanks for reporting, feel free to reopen if you still have the issue. Thanks.

closed 2019-10-11 by Alvin1Zhang