espressif/esp-idf #3827: unwanted 'as' interference

rajaganapathy007, 2019-07-25:
Hi everyone I'm Ganapathy, This is my first post in Github

The problem I'm facing right is difficult to fix and didnt find any answer in google either

Here what I did
- Ubuntu 18.06 in virtualbox minimal OS
- cloned esp-idf framework into system 
   - $IDF_PATH = /home/rajaganapathy/esp/esp-idf
- downloaded xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-amd64.tar.gz and extracted 
   - export PATH=/home/rajaganapathy/share/xtensa-esp32-elf/bin:$PATH
- also downloaded necessary files as well everything as per the guide in esp-idf installation documents

make output :
Toolchain path: /home/rajaganapathy/share/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from /home/rajaganapathy/esp/esp-idf/requirements.txt are satisfied.

everything is okay till this point but nothing compiles

I edited the make file project.mk in make folder and this is the result

Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/rajaganapathy/share/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp32-2019r1' --disable-__cxa_atexit --enable-cxx-flags='-fno-rtti -ffunction-sections' --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.2.0 (crosstool-NG esp32-2019r1) 
COLLECT_GCC_OPTIONS='-v' '--version'
 /home/rajaganapathy/share/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/cc1 -quiet -v -iprefix /home/rajaganapathy/share/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/ help-dummy -quiet -dumpbase help-dummy -auxbase help-dummy -version --version -o /tmp/ccaq63j6.s
GNU C17 (crosstool-NG esp32-2019r1) version 8.2.0 (xtensa-esp32-elf)
	compiled by GNU C version 4.9.2, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version isl-0.19-GMP

GGC heuristics: --param ggc-min-expand=97 --param ggc-min-heapsize=126136
COLLECT_GCC_OPTIONS='-v' '--version'
 as --traditional-format --version -o /tmp/ccmrJ6xi.o /tmp/ccaq63j6.s
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/rajaganapathy/share/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp32-2019r1' --disable-__cxa_atexit --enable-cxx-flags='-fno-rtti -ffunction-sections' --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.2.0 (crosstool-NG esp32-2019r1) 
COLLECT_GCC_OPTIONS='-v' '--version'
 /home/rajaganapathy/share/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/cc1 -quiet -v -iprefix /home/rajaganapathy/share/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/ help-dummy -quiet -dumpbase help-dummy -auxbase help-dummy -version --version -o /tmp/cc7EXh2n.s
GNU C17 (crosstool-NG esp32-2019r1) version 8.2.0 (xtensa-esp32-elf)
	compiled by GNU C version 4.9.2, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version isl-0.19-GMP

GGC heuristics: --param ggc-min-expand=97 --param ggc-min-heapsize=126136
COLLECT_GCC_OPTIONS='-v' '--version'
 as --traditional-format --version -o /tmp/cchSSPoB.o /tmp/cc7EXh2n.s
xtensa-esp32-elf-gcc: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found
compilation terminated.
Toolchain path: /home/rajaganapathy/share/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from /home/rajaganapathy/esp/esp-idf/requirements.txt are satisfied.
CC build/app_trace/app_trace_util.o
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp32-2019r1' --disable-__cxa_atexit --enable-cxx-flags='-fno-rtti -ffunction-sections' --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.2.0 (crosstool-NG esp32-2019r1) 
COLLECT_GCC_OPTIONS='-v' '-std=gnu99' '-Og' '-ggdb' '-Wno-frame-address' '-ffunction-sections' '-fdata-sections' '-fstrict-volatile-bitfields' '-mlongcalls' '-nostdlib' '-Wall' '-Werror=all' '-Wno-error=unused-function' '-Wno-error=unused-but-set-variable' '-Wno-error=unused-variable' '-Wno-error=deprecated-declarations' '-Wextra' '-Wno-unused-parameter' '-Wno-sign-compare' '-Wno-old-style-declaration' '-D' 'ESP_PLATFORM' '-D' 'IDF_VER="v4.0-dev-1287-gd7e659df2-dirty"' '-MMD' '-MP' '-D' '_GNU_SOURCE' '-D' 'GCC_NOT_5_2_0=1' '-D' 'MBEDTLS_CONFIG_FILE="mbedtls/esp_config.h"' '-D' 'HAVE_CONFIG_H' '-D' 'UNITY_INCLUDE_CONFIG_H' '-I' '/home/rajaganapathy/esp/esp-idf/components/app_trace/include' '-I' '/home/rajaganapathy/esp/esp-idf/components/app_update/include' '-I' '/home/rajaganapathy/esp/esp-idf/components/asio/asio/asio/include' '-I' '/home/rajaganapathy/esp/esp-idf/components/asio/port/include' '-I' '/home/rajaganapathy/esp/esp-idf/components/bootloader_support/include' '-I' '/home/rajaganapathy/esp/esp-idf/components/bt/include' '-I' '/home/rajaganapathy/esp/esp-idf/components/coap/port/include' '-I' '/home/rajaganapathy/esp/esp-idf/components/coap/port/include/coap' '-I' '/home/rajaganapathy/esp/esp-idf/components/coap/libcoap/include' '-I' '/home/rajaganapathy/esp/esp-idf/components/coap/libcoap/include/coap2' '-I' '/home/rajaganapathy/esp/esp-idf/components/console' '-I' '/home/rajaganapathy/esp/esp-idf/components/driver/include' '-I' '/home/rajaga [... cut ...]
CC := $(call dequote,$(CONFIG_SDK_TOOLPREFIX))gcc
CXX := $(call dequote,$(CONFIG_SDK_TOOLPREFIX))c++
LD := $(call dequote,$(CONFIG_SDK_TOOLPREFIX))ld
AR := $(call dequote,$(CONFIG_SDK_TOOLPREFIX))ar
OBJCOPY := $(call dequote,$(CONFIG_SDK_TOOLPREFIX))objcopy
OBJDUMP := $(call dequote,$(CONFIG_SDK_TOOLPREFIX))objdump
SIZE := $(call dequote,$(CONFIG_SDK_TOOLPREFIX))size
export CC CXX LD AR OBJCOPY OBJDUMP SIZE

none of them have 'as' command I didnt what to do I tried almost a week and didnt able to solve it   
Please Help me, Thank You

igrr, 2019-07-26:
> everything is okay till this point but nothing compiles
I edited the make file project.mk in make folder and this is the result

I would suggest going back to the unmodified version of project.mk and try to troubleshoot at that stage. Does the issue ("nothing compiles") happen with ESP-IDF examples, such as examples/get-started/blink?

rajaganapathy007, 2019-07-26:
Thank you replying

Yes, even examples/get-started/blink , examples/get-started/hello_world not compiling even before modifying the project.mk i.e I didn't able to successfully compile not even a single app yet.
Previously I worked with esp8266 it was no problem even it present some issues at first but I was able to fix it myself , but this is beyond me, 
I dont understand where 'as' get invoked it should be 'xtensa-esp32-elf-as' but it simply invokes 'as' alone

rajaganapathy007, 2019-07-26:
even i tried 'git pull origin master' command inside esp-idf folder but no difference

igrr, 2019-07-26:
Can you please run `make V=1` in one of the examples (blink) with unmodified ESP-IDF version, and attach the output here?

rajaganapathy007, 2019-07-26:
I deleted esp-idf folder along with everything and installed with this command 

git clone --recursive https://github.com/espressif/esp-idf.git 

but no difference , same result as before, This is the result 

including /home/rajaganapathy/esp/esp-idf/components/app_update/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/bootloader/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/efuse/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/esp32/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/esp_wifi/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/mbedtls/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/nghttp/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/partition_table/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/spiffs/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/ulp/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/unity/Makefile.projbuild...
Toolchain path: /home/rajaganapathy/share/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
python /home/rajaganapathy/esp/esp-idf/tools/check_python_dependencies.py
Python requirements from /home/rajaganapathy/esp/esp-idf/requirements.txt are satisfied.
PROJECT_PATH= COMPONENT_DIRS= make -C /home/rajaganapathy/esp/esp-idf/components/bootloader/subproject V=1 BUILD_DIR_BASE=/home/rajaganapathy/esp/esp-idf/examples/get-started/blink/build/bootloader TEST_COMPONENTS= TESTS_ALL= EXCLUDE_COMPONENTS= /home/rajaganapathy/esp/esp-idf/examples/get-started/blink/build/bootloader/bootloader.bin
make[1]: Entering directory '/home/rajaganapathy/esp/esp-idf/components/bootloader/subproject'
including /home/rajaganapathy/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/bootloader/subproject/main/Makefile.projbuild...
including /home/rajaganapathy/esp/esp-idf/components/efuse/Makefile.projbuild...
make -C /home/rajaganapathy/esp/esp-idf/examples/get-started/blink/build/bootloader/bootloader_support -f /home/rajaganapathy/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/home/rajaganapathy/esp/esp-idf/components/bootloader_support/component.mk COMPONENT_NAME=bootloader_support build
make[2]: Entering directory '/home/rajaganapathy/esp/esp-idf/examples/get-started/blink/build/bootloader/bootloader_support'
xtensa-esp32-elf-gcc -std=gnu99 -Og -ggdb -Wno-frame-address -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -I /home/rajaganapathy/esp/esp-idf/components/esp32/include -I /home/rajaganapathy/esp/esp-idf/components/esp_rom/include -I /home/rajaganapathy/esp/esp-idf/components/esp_common/include -I /home/rajaganapathy/esp/esp-idf/components/xtensa/include -I /home/rajaganapathy/esp/esp-idf/components/xtensa/esp32/include -D BOOTLOADER_BUILD=1 -I/home/rajaganapathy/esp/esp-idf/examples/get-started/blink/build/bootloader/../include -DESP_PLATFORM -D IDF_VER=\"v4.0-dev-1342-g84982ae2e\" -MMD -MP -DESP_PLATFORM -D IDF_VER=\"v4.0-dev-1342-g84982ae2e\" -MMD -MP   -D_GNU_SOURCE -DGCC_NOT_5_2_0=1 -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -DUNITY_INCLUDE_CONFIG_H  -D_GNU_SOURCE -DGCC_NOT_5_2_0=1 -I /home/rajaganapathy/esp/esp-idf/components/bootloader_support/include -I /home/rajaganapathy/esp/esp-idf/components/bootloader_support/include_bootloader -I /home/rajaganapathy/esp/esp-idf/components/log/include -I /home/rajaganapathy/esp/esp-idf/components/spi_flash/include -I /home/rajaganapathy/esp/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc -I /home/rajaganapathy/esp/esp-idf/components/soc/esp32/include -I /home/rajaganapathy/esp/esp-idf/components/soc/include -I /home/rajaganapathy/esp/esp-idf/components/bootloader/subproject/main/include -I /home/rajaganapathy/esp/esp-idf/components/efuse/esp32/include -I /home/rajaganapathy/esp/esp-idf/components/efuse/include -I /home/rajaganapathy/esp/esp-idf/examples/get-started/blink/build/bootloader/include  -I src -c /home/rajaganapathy/esp/esp-idf/components/bootloader_support/src/bootloader_flash.c -o src/bootloader_flash.o
as: unrecognized option '--longcalls'
/home/rajaganapathy/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'src/bootloader_flash.o' failed
make[2]: *** [src/bootloader_flash.o] Error 1
make[2]: Leaving directory '/home/rajaganapathy/esp/esp-idf/examples/get-started/blink/build/bootloader/bootloader_support'
/home/rajaganapathy/esp/esp-idf/make/project.mk:584: recipe for target 'component-bootloader_support-build' failed
make[1]: *** [component-bootloader_support-build] Error 2
make[1]: Leaving directory '/home/rajaganapathy/esp/esp-idf/components/bootloader/subproject'
/home/rajaganapathy/esp/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/home/rajaganapathy/esp/esp-idf/examples/get-started/blink/build/bootloader/bootloader.bin' failed
make: *** [/home/rajaganapathy/esp/esp-idf/examples/get-started/blink/build/bootloader/bootloader.bin] Error 2

rajaganapathy007, 2019-08-12:
I fixed it myself, Thank you for the time

Well is the issue was I'm using Virtualbox installed xtensa-esp32-elf in shared folder which is ntfs filesystem, so i reinstalled in home directory or any directory with native ext4 filesystem, it works perfectly fine

I found this, when to tried to build from scratch, i got an error for stat command result unknown then i searched for that error i found that it  won't compile when compiled in non Unix filesystem like ntfs, so i tried to install in native filesystem it works fine, both the standard tar file and compiled cross compiler works without any issue

Hope this information help somebody

closed 2019-08-12 by rajaganapathy007